espressif/esp-idf #8876: ESP32S3的SPI2_HOST使用spi_master外设时，用"spi_bus_add_device"添加大于三个从机时报错

ou2356, 2022-05-02:
## Environment
- Module or chip used:  [ESP32S3-WROOM-1]
- IDF version (run ``git describe --tags`` to find it):
    // Release v4.4
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclipse + IDF插件)]
- Power Supply:         [USB]

## Problem Description

ESP32S3的SPI外设的SPI2在[官方文档](https://docs.espressif.com/projects/esp-idf/zh_CN/release-v4.4/esp32s3/api-reference/peripherals/spi_master.html)中说到最大支持6个设备，但是当我添加第4个设备时就出现了错误。

## Expected behavior
4个SPI设备可以同时正常添加且使用。

## Actual behavior
只能同时添加3个SPI设备（当我在4个SPI设备的添加中注释掉任何一个SPI设备的添加都可以正常运行）

### Code to reproduce this issue

```cpp
    esp_err_t ret;
    spi_bus_config_t SPI2_buscfg={
        .sclk_io_num=SPI_PIN_CLK,
        .mosi_io_num=SPI_PIN_MOSI,
        .miso_io_num=SPI_PIN_MISO,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1,
//        .max_transfer_sz=8,
    };

    gpio_set_pull_mode(SPI_PIN_CLK, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(SPI_PIN_MOSI, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(SPI_PIN_MISO, GPIO_PULLUP_ONLY);
    
    //Initialize the SPI bus
    ret=spi_bus_initialize(SPI2_HOST, &SPI2_buscfg, SPI_DMA_CH_AUTO);
    ESP_ERROR_CHECK(ret);

    spi_device_interface_config_t devcfg={
    	.address_bits = 8,
        .clock_speed_hz=BMM150_SPI_SPEED,            	//Clock out at 10 MHz
        .mode=BMM150_SPI_MODE,                         //SPI mode (CPOL, CPHA)
        .spics_io_num=BMM150_PIN_CS,            //CS pin
        .queue_size=1,                          //We want to be able to queue 1 transactions at a time
        .flags = /*SPI_DEVICE_3WIRE | */SPI_DEVICE_HALFDUPLEX,
        .cs_ena_pretrans = 1,
    };
    //Attach the BMM150 to the SPI bus
    ret=spi_bus_add_device(SPI2_HOST, &devcfg, &spi);
    ESP_ERROR_CHECK(ret);

    spi_device_interface_config_t devcfg={
    	.address_bits = 8,
        .clock_speed_hz=BMP390_SPI_SPEED,            	//Clock out at 10 MHz
        .mode=BMP390_SPI_MODE,                         //SPI mode (CPOL, CPHA)
        .spics_io_num=BMP390_PIN_CS,            //CS pin
        .queue_size=1,                          //We want to be able to queue 1 transactions at a time
        .flags = /*SPI_DEVICE_3WIRE | */SPI_DEVICE_HALFDUPLEX,
        .cs_ena_pretrans = 1,
    };
    //Attach the BMP390 to the SPI bus
    ret=spi_bus_add_device(SPI2_HOST, &devcfg, &spi);
    ESP_ERROR_CHECK(ret);

    spi_device_interface_config_t devcfg_acc={
    	.address_bits = 8,
        .clock_speed_hz=BMI088_SPI_SPEED,            	//Clock out at 10 MHz
        .mode=BMI088_SPI_MODE,                         //SPI mode (CPOL, CPHA)
        .spics_io_num=BMI088_A_PIN_CS,            //CS pin
        .queue_size=1,                          //We want to be able to queue 1 transactions at a time
        .flags = /*SPI_DEVICE_3WIRE | */SPI_DEVICE_HALFDUPLEX,
        .cs_ena_pretrans = 1,
    };
    //Attach the BMI088 to the SPI bus
    ret=spi_bus_add_device(SPI2_HOST, &devcfg_acc, &spi_acc);
    ESP_ERROR_CHECK(ret);

    spi_device_interface_config_t devcfg_gyr={
    	.address_bits = 8,
        .clock_speed_hz=BMI088_SPI_SPEED,            	//Clock out at 10 MHz
        .mode=BMI088_SPI_MODE,                         //SPI mode (CPOL, CPHA)
        .spics_io_num=BMI088_G_PIN_CS,            //CS pin
        .queue_size=1,                          //We want to be able to queue 1 transactions at a time
        .flags = /*SPI_DEVICE_3WIRE | */SPI_DEVICE_HALFDUPLEX,
        .cs_ena_pretrans = 1,
    };
    //Attach the BMI088 to the SPI bus
    ret=spi_bus_add_device(SPI2_HOST, &devcfg_gyr, &spi_gyr);
    ESP_ERROR_CHECK(ret); //在这里返回错误信息
```
## Debug Logs

```
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4037ac08
file: "../components/BMI088/UserBmi088.c" line 74
func: mbmi088_user_config_spi_bus
expression: ret
```

## Other items

然后我试图查找产生这个错误的原因：
1、触发"ESP_ERR_NO_MEM"错误的位置为：https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/components/driver/spi_master.c#L338-L340
2、间接原因位置：
https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/components/driver/spi_bus_lock.c#L558-L581
https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/components/driver/spi_common.c#L812-L816
3、我推测的根本原因位置：https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/components/soc/esp32s3/include/soc/soc_caps.h#L248
（这里的宏定义恒为3，与[官方文档](https://docs.espressif.com/projects/esp-idf/zh_CN/release-v4.4/esp32s3/api-reference/peripherals/spi_master.html)中说到最大支持6个设备不符合）

##后续的自我尝试

当我推测到这个根本原因位置后，把https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/components/soc/esp32s3/include/soc/soc_caps.h#L248
更改成
```cpp
#if CONFIG_IDF_TARGET_ESP32
#define SOC_SPI_PERIPH_CS_NUM(i)            3
#else
#define SOC_SPI_PERIPH_CS_NUM(i)            ((i == SPI2_HOST)? 6 : 3)
#endif
```
编译运行之后，虽然可以成功添加4个SPI设备，但是第四个添加的设备读取到的数据全是0xFF，估计就是通信失败。

Huckies, 2022-05-04:
用逻辑分析仪抓下cs线的电平看看？六个设备指的是三根cs线分别在高低电平的时候使能一个设备

ou2356, 2022-05-04:
@Huckies 
感谢你的建议，但是很遗憾我没有逻辑分析仪。
但是六个设备真的是指3根片选控制六个设备吗？我也不太确定，可能是自己看漏了官方的文档，如果你方便的话可以发一下截图或者网址吗？我非常需要确认这一点以便后续硬件设计做出合理的更改。
我的推测是ESP32的SPI2是最大支持3个设备的，但是后来出的ESP32S2和S3等都是支持6个的，我只是怀疑是不是官方忘记更改成6个设备了，但感觉也不太可能，一切都只是我的猜测。

ou2356, 2022-05-04:
好消息是我找到一个折中的解决方法：
```cpp
    gpio_set_pull_mode(BMP390_PIN_CS, GPIO_PULLUP_ONLY);
    gpio_set_direction(BMP390_PIN_CS, GPIO_MODE_OUTPUT);
    gpio_set_level(BMP390_PIN_CS, 1);

    esp_err_t ret;
    spi_device_interface_config_t devcfg={
    	.address_bits = 8,
        .clock_speed_hz=BMP390_SPI_SPEED,            	//Clock out at 10 MHz
        .mode=BMP390_SPI_MODE,                         //SPI mode (CPOL, CPHA)
        .spics_io_num=/*BMP390_PIN_CS*/-1,            //禁用CS pin
        .queue_size=1,                          //We want to be able to queue 1 transactions at a time
        .flags = /*SPI_DEVICE_3WIRE | */SPI_DEVICE_HALFDUPLEX,
        .pre_cb = bmp390_pre_cb, //把CS置低
        .post_cb = bmp390_post_cb, //把CS置高
    };
    //Attach the BMP390 to the SPI bus
    ret=spi_bus_add_device(BMP390_SPI_HOST, &devcfg, &spi);
    ESP_ERROR_CHECK(ret);
```
就是上面的配置，我把它理解为用软件片选代替硬件片选，就是把CS脚设置为-1（即禁用CS脚），然后把通信前调用的回调函数设置为把CS置低，通信后调用的回调函数设置为把CS置高，这样也能用。

wanckl, 2022-09-09:
@ou2356 你好，确实是官方忘记更改CS对应引脚信号的问题，很抱歉给你造成了困扰。目前S2,S3,C2,C3都有这个问题，并将在这个Issue关闭后得到解决，届时将可以使用6个引脚添加6个设备来使用。同时，感谢你的反馈。

ou2356, 2022-09-10:
> @ou2356 你好，确实是官方忘记更改CS对应引脚信号的问题，很抱歉给你造成了困扰。目前S2,S3,C2,C3都有这个问题，并将在这个Issue关闭后得到解决，届时将可以使用6个引脚添加6个设备来使用。同时，感谢你的反馈。

好的，加油！

closed 2022-09-15 by espressif-bot

wanckl, 2022-09-19:
@ou2356 这个问题在master上已经修复，你可以切换到master分支进行测试。如果你使用release版本，则还需要等待内部测试完成发布下一个小版本的时候才能使用。感谢！

jinsonli, 2023-07-19:
你好， 我也遇到了类似问题。  
芯片： ESP32S3
环境： ESP-IDF4.4版本 release/v4.4 拉了官方最新的V4.4
场景：需要使用SPI2外挂 4个 从机设备，能够操作两个设备没问题（CS0/CS1），第三个设备初始化时， 发现CS2根本就没有拉低，底层代码，CS有正常初始化，也有正常添加从机设备。
https://github.com/espressif/esp-idf/issues/8876
以上issue中描述的问题，是无法添加第四个设备，我修改后可以添加第四个设备，不过在进行第3个第4个设备通讯时，CS无法拉低，
所以我怀疑这个问题还没有真正关闭。

wanckl, 2023-07-20:
@jinsonli 可以提供一下你使用的commit id吗，   
此外，GPSPI2才可以挂6个设备，GPSPI3只能3个。CS不需要额外手动初始化，通信成功但数据不对，也可能是从机没有发出数据，你可以交换你的几个设备试试，如果你有逻辑分析仪就更好了

uta-guy, 2024-04-20:
为了简化操作在ARDUINO 2.3.2内安装ESP32 V2.0.15测试运行SPI2外接6个SPI设备（需要6个CS端口）失败。 只能支持4个。查看ARDUINO下载的2.0.15库找到ESP32S3对应的soc_caps.h仍然显示限制数3. 查看最新版ESP-IDF V5.2.1中同一文件soc_caps.h确认已经修改为6，拷贝过去覆盖2.0.15库同文件，烧录无效。看来还有其他文件需要修改。 已经定制了上百片ESP32S3产品PCB，如短期内不能解决此问题只能全部报废换方案了。因为之前用的方案是METRO M4，主控是很老的SAMD21， 人家支持6个SPI毫无问题。