espressif/esp-idf #8879: esp32s3 IDF-2048 multicore usb_osglue review

chmorgan, 2022-05-02:
I'm trying to configure the console to work on the usb cdc port provided by the eep-box (and I'm guessing many other boards and dev boards).

Found the console_usb example and integrated the code in (it was pretty quick) and ran into this:

```
/* The two functions below need to be revisited in the multicore case TODO ESP32-S3 IDF-2048*/
_Static_assert(SOC_CPU_CORES_NUM == 1, "usb_osglue_*_int is not multicore capable");
```

https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/components/esp_system/port/soc/esp32s3/usb_console.c#L69

Using v4.4 but the line above is from master (figured I'd see if it was fixed in master).

Anyone looking at this? I'd like to get the console on this port vs. having to use a serial/usb converter etc to interact with the system.

igrr, 2022-05-02:
Hi @chmorgan,
We have a draft patch which fixes this, but we haven't got the USB_OTG CDC feature on S3 fully working yet (with reboot-into-download mode). There is also another issue open on Github which tracks making USB_OTG CDC work on S3: https://github.com/espressif/esp-idf/issues/8738#issuecomment-1091815661. If you don't mind, I'd close this one and keep the earlier one open.

There is also another issue about USB console over USB_SERIAL_JTAG peripheral: https://github.com/espressif/esp-idf/issues/8839. Since ESP-BOX uses USB_SERIAL_JTAG peripheral by default, I think that one is more relevant for your use case.

chmorgan, 2022-05-02:
> S3: #8738 (comment)

Yes please feel free to close.

Any timeline? We are looking to design in esp32-s3 into a product and it’s
always more comfortable with fewer open tickets. I do understand the chip
is relatively new.

I don’t quite understand the usb jtag vs usb cdc details except that I know
I’d like the console on the same single USB port that I get print output on
today with idf.py monitor and also program through.

If this means specifically support for console on usb/jtag which is what it
sounds like, then I’m a big fan!

Thank you for the ticket links.

Chris

igrr, 2022-05-02:
@chmorgan I think https://github.com/espressif/esp-idf/issues/8839 (for USB_SERIAL_JTAG console) is the simpler issue of the two. I think we can have a solution by next week.
https://github.com/espressif/esp-idf/issues/8738 (for USB_CDC) is a bit more complex, I can't give timeline for it yet.

> I don’t quite understand the usb jtag vs usb cdc details except that I know I’d like the console on the same single USB port that I get print output on today with idf.py monitor and also program through.

ESP32-S3 chip has two USB-related peripherals:
* USB_OTG peripheral (which is same as the one in ESP32-S2) — a programmable USB device and host controller. It can support various USB classes and requires a software driver to implement each USB class.
* USB_SERIAL_JTAG peripheral (new, also introduced in ESP32-C3) — a fixed-function USB composite device which implements two independent functions: CDC and USB-JTAG adapter. It can not support, for instance, USB mass storage or HID. It can be used for JTAG debugging and for serial flashing/monitoring.

You can find more information on the two in the ESP32-S3 technical reference manual.

By default, ESP32-S3 built-in USB PHY is connected to USB_SERIAL_JTAG peripheral. I think that's how ESP-BOX is set up, as well.

From usage perspective, if your device doesn't need to implement any USB features (such as MSC, HID and so on), then I would suggest using USB_SERIAL_JTAG (which is the default). If your device needs to support USB MSC, HID or other classes, then you would have to use USB_OTG.

Take note also of limitations of console features on both of them: [usb_serial_jtag](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-serial-jtag-console.html#limitations), [usb_otg](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-otg-console.html#limitations).

chmorgan, 2022-05-02:
Thank you Ivan!

igrr, 2022-05-16:
Hi @chmorgan, the "basic" console example now has support for usb_serial_jtag on ESP32-S3. This should let you have a USB console on ESP-BOX. Please see commit https://github.com/espressif/esp-idf/commit/3390d2a2d18c289b0b1e1cf121bf6670099641db for the updates.

As the issue title refers to USB_OTG CDC support on S3, I'll keep the issue open until that part is also implemented.

chmorgan, 2022-05-17:
@igrr this is great news. Thank you for putting this support in place. I think this is sufficient for our use but of course it will be good to see as the S3 support continues to build out. It still seems like a pretty new chip.

SinanAkkoyun, 2022-08-01:
Hi, thanks for all the great resources you people provided!!! <3

I am still getting the ```
master/esp-idf/components/esp_system/port/soc/esp32s3/usb_console.c:69:1: error: static assertion failed: "usb_osglue_*_int is not multicore capable"
   69 | _Static_assert(SOC_CPU_CORES_NUM == 1, "usb_osglue_*_int is not multicore capable");
      | ^~~~~~~~~~~~~~
[230/861] Performing configure step for 'bootloader'
```
error when trying to compile for the S3, is there any _approximate_ timeline at which the serial console will work on the S3 OTG? :)

igrr, 2022-08-01:
@SinanAkkoyun Actually this is now in progress, I hope that in one-two weeks this feature will be available in `master` branch.

SinanAkkoyun, 2022-08-01:
Cool, thanks so much!

udem1234, 2022-11-18:
> @SinanAkkoyun Actually this is now in progress, I hope that in one-two weeks this feature will be available in `master` branch.

Hi @SinanAkkoyun !  Any update on this ?

dchichkov, 2022-11-26:
I've started hitting this assert(SOC_CPU_CORES_NUM == 1, "usb_osglue_*_int is not multicore capable"), is there a commit or workaround available?

udem1234, 2022-12-21:
> @SinanAkkoyun Actually this is now in progress, I hope that in one-two weeks this feature will be available in `master` branch.

@igrr : Is this feature/issue resolving still in progress ?

Thank you ! :-)

kraftwerk-zone, 2023-01-14:
any update on this?

closed 2023-04-18 by espressif-bot

rodmaz, 2023-11-14:
Is this fixed in v4.4.x? We upgraded to v4.4.6 and still see this problem.

igrr, 2023-11-14:
@rodmaz Support for system console via USB_OTG on ESP32-S3 was never present in v4.4.x releases, so I don't think this is a regression. Please check that you haven't accidentally set the console output channel to USB_OTG in menuconfig.

rodmaz, 2023-11-14:
@igrr We are actually trying to use USB_OTG in v4.4.x. It seems we need to upgrade to v5.x then?

igrr, 2023-11-14:
Do you intend to use USB_OTG only for the console, or for some other functionality? If also for some other USB classes than CDC, which ones? If you are using it only for the console, have you considered using the USB_SERIAL_JTAG-based console instead?

rodmaz, 2023-11-14:
No, we want to use debugger as well with OpenOCD.

igrr, 2023-11-14:
In this case you should use console via USB_SERIAL_JTAG — this is the peripheral which provides both CDC console and JTAG interface over USB: https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/api-guides/usb-serial-jtag-console.html

(USB_OTG is a programmable USB peripheral which can be used to implement features such as MSC, CDC, HID and so on, in software. It can't be used to debug ESP32-S3 over JTAG.)