espressif/esp-idf #8880: Removing security from ESP Provisioning example doesn't work

sch0bert, 2022-05-02:
## Environment

- Development Kit: ESP-WROOM-32 generic dev kit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.4.1
- Build System: idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:   Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm trying to disable the security that by default is enables in the ESP Provisioning example. After doing it, causes that the APP (Android) is not capable to pair the device. What I'm trying to do is to avoid getting the OS notification asking to pair and share the contact list of my phone. So in the code I changed:

[image]

And in the menuconfig I decided to have only configured as Peripheral role, just accept 1 concurrent connection, also to save only 1 bonded device across reboots and disabled "Security manager legacy pairing" "Security manager secure connections (4.2)":

[image]

By doing this, the app is able to list the device but when I select it, is not capable to connect. I get the next errors:

[image]
[image]

I enabled the debug logs in the code and this is what I got:

[Fail_log.txt](https://github.com/espressif/esp-idf/files/8602582/Fail_log.txt)

### Expected Behavior

The App should be able to pair the device without the need of a secure connection.

### Actual Behavior

Failing whenever I try to pair the device

### Steps to reproduce

1. Change in main.c the security level to 0
2. Disable through menuconfig the NimBLE's configuration based on the description of the issue
3. Compile the ESP Provisioning example code
4. Flash it
5. Download the ESP Provisioning Android app
6. Open the app and try to pair the device (with or without QR scanner)

sch0bert, 2022-05-03:
@jack0c could you help me redirect this case to someone? I would like to know if this is a valid case

rahult-github, 2022-05-04:
Hi @sch0bert ,

 Since wifi credentials are shared over BLE during provisioning, the existing code recommends to have link encrypted. This leads to pairing and the pop up is observed.  For your use case, please disable the option CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION  ( It is present at Component Config -> Wi - Fi provisioning Manager ) .  

1. Make sure CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION is disabled
2. Compile image, erase_flash and run monitor
3. Attempt to provision.

At step 3, you should not observe the pairing pop up.

Thanks,
 Rahul

sch0bert, 2022-05-04:
Hello @rahult-github,

Thanks for your response. I checked my menuconfig and I only have the next options:

[image]

Am I need to enable something else in order to get the option or should I simply hardcode the value through the sdkconfig file?

I checked the documentation about CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION

[image]

I do have NimBLE enabled

rahult-github, 2022-05-04:
Hi @sch0bert ,

 The change on release/v4.4 was done after the tag v4.4.1 got created.  https://github.com/espressif/esp-idf/commit/916533ef92ce8cf50ef53fa94262cf94a5124bb5 is the commit done, that you can cherry-pick and use it to get functionality in your code. 

Thanks,
 Rahul

sch0bert, 2022-05-04:
Got it! Is there any expected date for the next stable release? Maybe something like 4.4.2 in the future where it will include this fix?

rahult-github, 2022-05-04:
Hi @sch0bert , there will be a new release with more fixes. However, the exact release date is not yet decided.

closed 2022-05-04 by sch0bert