espressif/esp-idf #8883: Micropython ESP-IDF V5.0  "Failed to resolve component 'tcpip_adapter'"

TGiles1998, 2022-05-02:
Hi, I have seen that this is an existing issue. Is this being fixed?

-- Check for working CXX compiler: /Users/thomasgiles/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /Users/thomasgiles/esp/esp-idf/tools/cmake/build.cmake:197 (message):
  Failed to resolve component 'tcpip_adapter'.
Call Stack (most recent call first):
  /Users/thomasgiles/esp/esp-idf/tools/cmake/build.cmake:225 (__build_resolve_and_add_req)
  /Users/thomasgiles/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  /Users/thomasgiles/esp/esp-idf/tools/cmake/project.cmake:422 (idf_build_process)
  CMakeLists.txt:58 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/thomasgiles/XX/micropython/ports/esp32/build-GENERIC/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
make: *** [all] Error 2

Many thanks,

Thomas

igrr, 2022-05-02:
I'm afraid this will need to be solved on Micropython side, as tcpip_adapter is removed from ESP-IDF in the upcoming  v5.0 release. For migration instructions, please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/tcpip-adapter.html

TGiles1998, 2022-05-02:
@igrr  Thank you very much for your reply. I will look into it on the micropython side.

closed 2022-05-03 by igrr