espressif/esp-idf #8884: Issues with ESP_CONSOLE_USB_SERIAL_JTAG if set light_sleep_enable = true

AxelLin, 2022-05-03:
## Environment

- Module or chip used: ESP32-C3-DevKitM-1
- IDF version: v5.0-dev-2586-ga82e6e63d9
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description
I have below config setting:
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_RTC_CLOCK_BBPLL_POWER_ON_WITH_USB=y

And 
esp_pm_config_esp32c3_t pm_confg has .light_sleep_enable = true

In my by test running "idf.py -p /dev/ttyACM0 monitor", I usually got below messages after running for a few minutes:

device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect....

ps. The module still working (it can still upload data to my server), but I cannot see any debug messages.

The dmesg shows bellow errors:
[240881.602296] usb 3-1.4: new full-speed USB device number 65 using xhci_hcd
[240881.625327] usb 3-1.4: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[240881.625336] usb 3-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[240881.625339] usb 3-1.4: Product: USB JTAG/serial debug unit
[240881.625342] usb 3-1.4: Manufacturer: Espressif
[240881.625344] usb 3-1.4: SerialNumber: 84:F7:03:67:84:6C
[240881.639920] cdc_acm 3-1.4:1.0: ttyACM0: USB ACM device
[241049.824729] usb 3-1: USB disconnect, device number 14
[241049.824737] usb 3-1.4: USB disconnect, device number 65
[241056.423400] usb 3-1: new full-speed USB device number 66 using xhci_hcd
[241056.573094] usb 3-1: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[241056.573103] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[241056.573107] usb 3-1: Product: USB JTAG/serial debug unit
[241056.573109] usb 3-1: Manufacturer: Espressif
[241056.573112] usb 3-1: SerialNumber: 84:F7:03:67:84:6C
[241056.576237] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
[241077.572801] usb 3-1: USB disconnect, device number 66
[241077.879018] usb 3-1: new full-speed USB device number 67 using xhci_hcd
[241078.007089] usb 3-1: device descriptor read/64, error -71
[241078.243098] usb 3-1: device descriptor read/64, error -71
[241078.478932] usb 3-1: new full-speed USB device number 68 using xhci_hcd
[241078.611041] usb 3-1: device descriptor read/64, error -71
[241078.847073] usb 3-1: device descriptor read/64, error -71
[241078.955178] usb usb3-port1: attempt power cycle
[241079.606893] usb 3-1: new full-speed USB device number 69 using xhci_hcd
[241079.607103] usb 3-1: Device not responding to setup address.
[241079.815255] usb 3-1: Device not responding to setup address.
[241080.022990] usb 3-1: device not accepting address 69, error -71
[241080.151017] usb 3-1: new full-speed USB device number 70 using xhci_hcd
[241080.151256] usb 3-1: Device not responding to setup address.
[241080.363228] usb 3-1: Device not responding to setup address.
[241080.571014] usb 3-1: device not accepting address 70, error -71
[241080.571176] usb usb3-port1: unable to enumerate USB device

Re-run "idf.py -p /dev/ttyACM0 monitor" does not work, it shows below error:

Executing action: app-flash
Serial port /dev/ttyACM0
Connecting...
/dev/ttyACM0 failed to connect: [Errno 32] Broken pipe
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.

I have to press the reset button on the module, then "idf.py -p /dev/ttyACM0 monitor" works.

### Expected Behavior

According to the doc: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html
If the application enters light-sleep (including automatic light-sleep) or software reset, etc. The USB CDC device will still work on the system.

### Actual Behavior
Hit below issue and idf.py stop working:
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect....

igrr, 2022-05-03:
This seems to be the same issue as https://github.com/espressif/esp-idf/issues/8507, i would suggest to close this one and use the older issue to keep track of this.

closed 2022-05-04 by AxelLin

AxelLin, 2022-05-09:
> This seems to be the same issue as #8507, i would suggest to close this one and use the older issue to keep track of this.

I think it's the same issue as #8507.
I do hope someone is working on the issue because it seems there is no response on #8507 so far.