espressif/esp-idf #8885: MBEDTLS_REMOVE_ARC4_CIPHERSUITES is not enabled

nopnop2002, 2022-05-03:
MBEDTLS_REMOVE_ARC4_CIPHERSUITES is not enabled

## Environment
```
$ idf.py --version
ESP-IDF v5.0-dev-2224-g97fb98a91b-dirty
```
## Problem Descriptio
I changed Component config --> mbedTLS --> Symmetric Chipers --> RC4 Strem Chiper --> Enable

But MBEDTLS_REMOVE_ARC4_CIPHERSUITES is not enabled.

[image]

```
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:403:2: error: 'MBEDTLS_TLS_DHE_PSK_WITH_RC4_128_SHA' undeclared here (not in a function); did you mean 'MBEDTLS_TLS_DHE_PSK_WITH_AES_128_CCM'?
  MBEDTLS_TLS_DHE_PSK_WITH_RC4_128_SHA,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MBEDTLS_TLS_DHE_PSK_WITH_AES_128_CCM
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:404:2: error: 'MBEDTLS_TLS_RSA_WITH_RC4_128_SHA' undeclared here (not in a function); did you mean 'MBEDTLS_TLS_RSA_WITH_AES_128_CCM'?
  MBEDTLS_TLS_RSA_WITH_RC4_128_SHA,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MBEDTLS_TLS_RSA_WITH_AES_128_CCM
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:405:2: error: 'MBEDTLS_TLS_RSA_WITH_RC4_128_MD5' undeclared here (not in a function); did you mean 'MBEDTLS_TLS_RSA_WITH_AES_128_CCM'?
  MBEDTLS_TLS_RSA_WITH_RC4_128_MD5,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MBEDTLS_TLS_RSA_WITH_AES_128_CCM
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:406:2: error: 'MBEDTLS_TLS_RSA_PSK_WITH_RC4_128_SHA' undeclared here (not in a function); did you mean 'MBEDTLS_TLS_RSA_PSK_WITH_AES_128_CBC_SHA'?
  MBEDTLS_TLS_RSA_PSK_WITH_RC4_128_SHA,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MBEDTLS_TLS_RSA_PSK_WITH_AES_128_CBC_SHA
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:407:2: error: 'MBEDTLS_TLS_PSK_WITH_RC4_128_SHA' undeclared here (not in a function); did you mean 'MBEDTLS_TLS_PSK_WITH_AES_128_CCM'?
  MBEDTLS_TLS_PSK_WITH_RC4_128_SHA,
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MBEDTLS_TLS_PSK_WITH_AES_128_CCM
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

From components/mbedtls/port/include/mbedtls/esp_config.h

```
 * MBEDTLS_REMOVE_ARC4_CIPHERSUITES
 * This flag removes the ciphersuites based on RC4 from the default list as
 * returned by mbedtls_ssl_list_ciphersuites(). However, it is still possible to
 * enable (some of) them with mbedtls_ssl_conf_ciphersuites() by including them
 * explicitly.
 *
 * Uncomment this macro to remove RC4 ciphersuites by default.
 */
#ifdef CONFIG_MBEDTLS_RC4_ENABLED
#define MBEDTLS_ARC4_C
#undef MBEDTLS_REMOVE_ARC4_CIPHERSUITES
#elif defined CONFIG_MBEDTLS_RC4_ENABLED_NO_DEFAULT
#define MBEDTLS_ARC4_C
#define MBEDTLS_REMOVE_ARC4_CIPHERSUITES
#else
#undef MBEDTLS_ARC4_C
#define MBEDTLS_REMOVE_ARC4_CIPHERSUITES
#endif
```

nopnop2002, 2022-05-03:
I implemented the following as a countermeasure.
```
cd esp-idf/components/mbedtls/mbedtls/include/mbedtls
cp ssl_ciphersuites.h arc4.h
```

The build error has changed.
```
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:428:10: error: 'MBEDTLS_CIPHER_ARC4_128' undeclared (first use in this function); did you mean 'MBEDTLS_CIPHER_AES_128_KW'?
   return MBEDTLS_CIPHER_ARC4_128;
          ^~~~~~~~~~~~~~~~~~~~~~~
          MBEDTLS_CIPHER_AES_128_KW
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:428:10: note: each undeclared identifier is reported only once for each function it appears in
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c: In function 'rc4_skip':
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:769:2: error: unknown type name 'mbedtls_arc4_context'; did you mean 'mbedtls_rsa_context'?
  mbedtls_arc4_context ctx;
  ^~~~~~~~~~~~~~~~~~~~
  mbedtls_rsa_context
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:776:2: error: implicit declaration of function 'mbedtls_arc4_init'; did you mean 'mbedtls_rsa_init'? [-Werror=implicit-function-declaration]
  mbedtls_arc4_init(&ctx);
  ^~~~~~~~~~~~~~~~~
  mbedtls_rsa_init
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:777:2: error: implicit declaration of function 'mbedtls_arc4_setup'; did you mean 'mbedtls_pk_setup'? [-Werror=implicit-function-declaration]
  mbedtls_arc4_setup(&ctx, key, keylen);
  ^~~~~~~~~~~~~~~~~~
  mbedtls_pk_setup
/home/nop/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:783:14: error: implicit declaration of function 'mbedtls_arc4_crypt'; did you mean 'mbedtls_pk_encrypt'? [-Werror=implicit-function-declaration]
   if ((ret = mbedtls_arc4_crypt(&ctx, len, skip_buf_in,
              ^~~~~~~~~~~~~~~~~~
              mbedtls_pk_encrypt
cc1: some warnings being treated as errors
```

mahavirj, 2022-06-20:
@nopnop2002 

Please check about deprecated config options in mbedTLS-3.1 release at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/protocols.html#deprecated-config-options. This issue (build part) has been fixed with https://github.com/espressif/esp-idf/commit/28ac0b12fb0231bcaf60b63df5c690ec2f87cf41.

Closing the issue.

closed 2022-06-20 by mahavirj