espressif/esp-idf #8886: How to implement DHCP Server on ESP32

amagalhaes31, 2022-05-03:
**DHCP Server using ethernet**

Hi all,

I'm trying to implement dhcp server on esp32 via ethernet, is it possible? If yes, is there any example?
My device is ESP32-WROVER and I use idf v4.4.1.

Tks a lot.

david-cermak, 2022-05-03:
Hi @amagalhaes31 

Yes, it is possible. You can start with any Ethernet example in IDF, just need to set these two flags in the netif config:

```cpp
netif_cfg->base.flags = ESP_NETIF_DHCP_SERVER | ESP_NETIF_FLAG_AUTOUP;
```
after initializing the default config (just before creating the netif instance):

https://github.com/espressif/esp-idf/blob/8c8f700c1df05d8287316c78b98b940e119f900c/examples/ethernet/basic/main/ethernet_example_main.c#L95-L96

Here's the link to a more complete example: https://github.com/espressif/esp-idf/issues/4086#issuecomment-559530841

amagalhaes31, 2022-05-03:
Hi @david-cermak , I thank you.
It's working !!!

david-cermak, 2022-05-04:
Thanks for sharing the update!

Could you please close this issue then?

amagalhaes31, 2022-05-04:
TKS.

closed 2022-05-04 by amagalhaes31

alexbezu, 2024-01-22:
Is there any change to IDF v5.2? Cause I've got the following error:
```
I (358) main_task: Started on CPU0
I (368) main_task: Calling app_main()
I (398) esp_eth.netif.netif_glue: a0:a3:b3:23:43:97
I (398) esp_eth.netif.netif_glue: ethernet attached to netif
I (1898) main_task: Returned from app_main()
I (1898) esp_netif_lwip: DHCP server started on interface ETHDHCPS with IP: 192.168.1.1

assert failed: netif_add /IDF/components/lwip/lwip/src/core/netif.c:406 (netif already added)
```
the device is wt32-eth01
thx