espressif/esp-idf #8887: ESP32-C3 GPIO Interrupt is not triggered during Light Sleep

ermacv, 2022-05-03:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it): v4.4.1-91-g36823b45cc
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:        USB

## Problem Description

* FreeRTOS in tickless mode
* DFS with enabled light sleep
* GPIO input interrupt

I have a custom board with external 32 khz quartz for light sleep support, the pin that is used for interrupt handling has a pull down resistor.

So the problem is that GPIO interrupt and ISR function is not triggered with enabled light sleep but the same code works without DFS light sleep.
Also tested manually calling light sleep start function and the result is the same.

### Expected Behavior
GPIO interrupt is triggered during the light sleep

### Actual Behavior
GPIO interrupt is not triggered during the light sleep (MCU stays in a light sleep mode)

### Steps to reproduce

Use the code below

### Code to reproduce this issue

Gist: https://gist.github.com/ermacv/935185a81d61d33442413c9bc284b92b

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ *] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/8611995/sdkconfig.zip)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2022-05-03:
Hi @ermacv,
I think this is kind of expected (although not clearly documented in the docs [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html#dynamic-frequency-scaling-and-peripheral-drivers)!).. In light sleep, most chip peripherals except for the configured wake up sources are suspended. This includes the GPIO peripheral and its interrupt function.

You can try to enable GPIO wakeup from light sleep, using `esp_sleep_enable_gpio_wakeup` and `gpio_wakeup_enable` functions. This will wake up the chip from light sleep when GPIO changes to a certain level, and once the chip has woken up, GPIO interrupt can be triggered. Note that GPIO light sleep wakeup only supports "low level" and "high level" interrupt types, not edge interrupt types. Because of this you'd have to change the interrupt type in the ISR, to simulate edge interrupts.

I've made a few changes to your code, it works on an ESP32-C3 Devkit-M on my side: https://gist.github.com/igrr/5c4617b99d3cfa840cb88feeb3b58bff

(Make sure you also have `CONFIG_PM_SLP_DISABLE_GPIO` disabled)

ermacv, 2022-05-03:
Hi @igrr !
Thank you for the reply. Actually a use case for my application is to catch a GPIO interrupt from a button or external IC interrupt request pin to reduce the power consumption as much as possible but also be able to use some extra peripheral at the same time (BLE e.g.).

Just to mention here I also tried to keep all reference manual mentioned power domains active during the light sleep including the CPU and still ISR doesn't work. And I don't really understand what peripheral is placed into the sleep while all power domains are active? I guess it could be related to the fact that sleep functionality is placed outer the peripheral and CPU scope. So it is like the MCU should be in active mode to be able to access all the features of the MCU otherwise you may use only quite a limited amount of hardware functions.
And I think it should be mentioned in a documentation because for me the hardware functions should work even in a sleep mode. Maybe some examples should be also placed into the documentation. Here I compare how for example Cortex-M (STM, e.g.) based MCU will work in CPU sleep. So the core is stall but peripheral may work asynchronously from it.
Maybe I'm wrong but this is my personal expectation.
And the link you provided mainly tells us about the high level modules.

About the way it may work on esp32c3:
I also had the same suggestion: wake up on low or high state and then interrupt will be triggered (not edged). But I guess the application logic will be really complex. For example, an external IC interrupt pin uses a latched state interrupt and to clear it I need to read some I2C registers from the IC but of course I cannot use I2C from GPIO ISR and I need to disable interrupt (otherwise MCU will stuck in it), call I2C code, enable interrupt.
The code I use is quite modular and high level and I don't really want to use an extra function to enable the interrupt when the application is ready.

Currently I implemented an "ISR emulated" code that uses RTOS auto reload timer that reads GPIO state every N Ms and gives a semaphore if valid state was caught. This solution was easier for me because I just replaced ISR with timer code and it works as it was with hardware GPIO interrupt but now auto light sleep is enabled.

In any case thank you and I hope the docs will be updated with the functions available during the light sleep and maybe there will be added extra examples of how to use the peripheral in light sleep mode.

esp-lis, 2022-05-19:
In sleep mode, the CPU is in clock gating or power-down state (depending on the configuration **CONFIG_PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP**), only the RTC module works, the CPU cannot directly capture the interrupt signal and respond, you can enable the GPIO wake-up function (refer to https://github.com/espressif/esp-idf/issues/8887#issuecomment-1116457596). When the pre-configured GPIO signal state arrives, first Wake up the CPU (power up or clock gating disabled), and then capture the interrupt signal to respond.

The option **CONFIG_PM_SLP_DISABLE_GPIO** is used to disable all GPIOs (disable pin pull-up, pull-down and input and output) when the SOC is in sleep, to avoid leakage current due to voltage difference with peripheral circuits (which will increase sleep power consumption).

What I recommend is to enable **CONFIG_PM_SLP_DISABLE_GPIO**, and use the following interface when GPIO is initialized to configure the pull-up, pull-down and input and output of the application-dependent GPIO pins in the sleep state to be consistent with the wake-up state.
```c
esp_err_t gpio_sleep_set_direction(gpio_num_t gpio_num, gpio_mode_t mode);
esp_err_t gpio_sleep_set_pull_mode(gpio_num_t gpio_num, gpio_pull_mode_t pull);
```

Alvin1Zhang, 2022-07-20:
@ermacv Thanks for reporting, would you please help share if any updates for the issue? Thanks.

ermacv, 2022-07-29:
Hello @Alvin1Zhang. I guess there are no updates from my side but I would like to ask you to add more verbose information about CPU/RTC HW dependencies into the esp-idf docs because it was not obvious that RTC has more priority over CPU (kind of).
The issue is resolved and could be closed.

closed 2022-07-29 by ermacv