espressif/esp-idf #8890: ESP-IDF install.sh fails with Syntax Error

taguniversalmachine, 2022-05-03:
## Problem Description
In an effort to once and for all get IDF working in a consistent basis, I made a Docker image that installs all the prerequisites, including Python, and yet I still can't get the install script to work, I assume it's a python versioning error but nothing I do can force it to use Python3.

### Expected Behavior
install.sh esp32  should install the tools 

### Actual Behavior
 ```
> [16/24] RUN ./install.sh esp32:
#19 0.410 Detecting the Python interpreter
#19 0.410 Checking "python" ...
#19 0.412 /esp/esp-idf/tools/detect_python.sh: line 16: python: command not found
#19 0.414 Checking "python3" ...
#19 0.454 Python 3.5.2
#19 0.454 "python3" has been detected
#19 0.455 Installing ESP-IDF tools
#19 0.476   File "/esp/esp-idf/tools/idf_tools.py", line 382
#19 0.476     msg = f'Rename {path_from} to {path_to} failed'
#19 0.476                                                   ^
#19 0.476 SyntaxError: invalid syntax
------
executor failed running [/bin/sh -c ./install.sh esp32]: exit code: 1
```
### Steps to reproduce

Make a Dockerfile that installs python3, and checks out the 4.4 branch of ESP-IDF
```
FROM ubuntu
# Python
RUN apt-get install --reinstall ca-certificates
RUN apt-get install -y software-properties-common
RUN add-apt-repository ppa:deadsnakes/ppa
RUN apt-get update
RUN apt-get install -y python3
RUN alias  python=python3
RUN apt-get install -y python3-pip
RUN pip3 install --user pipenv

#ESP-IDF
RUN apt-get install -y git wget flex bison gperf python3 python3-pip python3-setuptools cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
RUN mkdir -p /esp
WORKDIR "/esp"
RUN git clone -b v4.4.1 --recursive https://github.com/espressif/esp-idf.git
WORKDIR "/esp/esp-idf"
ENV  ESP_PYTHON=python3
RUN ./install.sh esp32
```

dobairoland, 2022-05-04:
Hi @taguniversalmachine. There are already official ESP-IDF docker images available (for each release branches). The source can be found [here](https://github.com/espressif/esp-idf/blob/master/tools/docker/Dockerfile) and the documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html). You can use the source if you want to build your own version.

You can ignore this:
```
#19 0.412 /esp/esp-idf/tools/detect_python.sh: line 16: python: command not found
```
If there is no `python` then it will try to use `python3`. If you really want to make `python` to be `python3` then please don't try to alias it. It won't work. Please use the `update-alternatives` as shown in the linked sources instead.

```
#19 0.454 Python 3.5.2
#19 0.454 "python3" has been detected
#19 0.455 Installing ESP-IDF tools
#19 0.476   File "/esp/esp-idf/tools/idf_tools.py", line 382
#19 0.476     msg = f'Rename {path_from} to {path_to} failed'
```
The issue here is that Python 3.5 is available and ESP-IDF v4.4 require 3.6 or newer. You can switch to a newer version of the OS or install a suitable python in your docker. You can achieve this by adding a mirror to `apt` or install Python from sources.

taguniversalmachine, 2022-05-04:
Awesome, thanks fo r the excellent support.

Alvin1Zhang, 2022-05-05:
Reopen until the fix is on GitHub, thanks.

Alvin1Zhang, 2022-05-16:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/7b33262d26922f8c0e836c42aaac067dc32c538c.

closed 2022-05-16 by Alvin1Zhang