espressif/esp-idf #8891: LittleFS for ESP32 Problems everywhere, IPC calls. Worthless, worst than FATFS in execution.

filzek, 2022-05-03:
SDK 4.4.1

LittleFS as suggest to replace FATFS or SPIFFS, but it has a lot of problems of IPC calls, hangs and concurrent executions.

The behavior in a complex environment is worthless, crosscore and make the tasks halt forever.

If you are accessing the flash you are dead, nothing work flawless or concurrent, it just crash the entire system. 
Backtrace:0x4008FAD1:0x3FFBFCA00x40084301:0x3FFBFCC0 0x400D97CF:0x3FFEC380 0x4009AD45:0x3FFEC390 0x400827D1:0x3FFEC3B0 0x4008EAE3:0x3FFEC3D0 0x4008EDF8:0x3FFEC3F0 0x4008EEFE:0x3FFEC440 0x4014BBF9:0x3FFEC470 0x4014BC6D:0x3FFEC490 0x400FA689:0x3FFEC4C0 0x400FABAD:0x3FFEC4E0 0x400FAE2A:0x3FFEC510 0x400FAEBE:0x3FFEC580 0x400FC8E5:0x3FFEC5F0 0x400FDBAD:0x3FFEC6B0 0x400FA1B6:0x3FFEC7A0 0x400D53C5:0x3FFEC7C0 0x401BAF45:0x3FFEC7E0 0x400E1859:0x3FFEC800
0x4008fad1: esp_crosscore_isr at D:/Dropbox/esp-idf/components/esp_system/crosscore_int.c:92

0x40084301: _xt_lowint1 at D:/Dropbox/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

Maybe if there is an real Espressif version that solve all the problems, so, for now we will move back and remove all code we have introduce to make it work, so, going back to FATFS WealLeaveling.

negativekelvin, 2022-05-04:
https://github.com/joltwallet/esp_littlefs

filzek, 2022-05-04:
@negativekelvin I know it, but as there is a problem on FATFS/SPIFFS corruption on power loss, we have tried in the current running solution and its is very bad, cant work flawless in the same environment, so, its not even a close chance to use LITTLEFS currently ro replace the FATFS/SPIFFS as it creates a lot of new problems that doesnt exist before.

In the future if espressif embrace and show simultaneous usage of LITTLEFS with timers, tasks, all running simultaneous, adc readings, spi, i2c, and didnt breaking anything it will be worth to test again. So far not good to use it for now.

igrr, 2022-05-05:
Hi @filzek,

Do you think you could find narrow down the issue to a smaller code sample which you could share?

LittleFS is a fairly high level component, it doesn't directly interact with IPC, tasks or other system-level features. It could be something else in the application or in IDF that is the root cause of the behavior you see.

You have posted a backtrace, but since it's not decoded to function names, it's not helpful and doesn't let us troubleshoot your issue. Please use `idf.py monitor` which will decode backtrace automatically, or use `xtensa-esp32-elf-addr2line -pfiaC -e build/your-app.elf` to manually decode the backtrace.

One more question, does your app create any tasks with priority equal to 24 (or configMAX_PRIORITIES - 1)? This could explain some issues with blocked tasks and IPC.

filzek, 2022-05-30:
Next wednesday we will have a fully debug action on this usage and produce a full report, so to be able to reproduce everywhere.

closed 2023-04-03 by Alvin1Zhang