espressif/esp-idf #8893: Compilation error: undefined reference to `_binary_x509_crt_bundle_start'

binhla, 2022-05-04:
I am trying to build https_request example (esp-idf-v4.3.1\examples\protocols\https_request) and got the error:
```
FAILED: https_request.elf
cmd.exe /C "cd . && C:\Users\laian\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\https_request.elf.rsp  -o https_request.elf  && cd ."
c:/users/laian/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/mbedtls/libmbedtls.a(esp_crt_bundle.c.obj):(.literal.esp_crt_bundle_attach+0x0): undefined reference to `_binary_x509_crt_bundle_start'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

My Idf version is 4.3.1

I had only made a single change in the menuconfig as follow

https://github.com/espressif/esp-idf/issues/7621#issuecomment-931241064

mahavirj, 2022-05-04:
@binhla 

Problem with negative serial number based certificate in bundle was fixed and you may find details regarding that in https://github.com/espressif/esp-idf/issues/7631#issuecomment-948445140

Alternatively, for build issue that you are facing, please try applying fix similar to https://github.com/espressif/esp-idf/commit/021f545f0db38985626ecb58ccdd9d492e3782fa in `http_request` example.

binhla, 2022-05-04:
> @binhla
> 
> Problem with negative serial number based certificate in bundle was fixed and you may find details regarding that in [#7631 (comment)](https://github.com/espressif/esp-idf/issues/7631#issuecomment-948445140)
> 
> Alternatively, for build issue that you are facing, please try applying fix similar to [021f545](https://github.com/espressif/esp-idf/commit/021f545f0db38985626ecb58ccdd9d492e3782fa) in `http_request` example.

Thank you for your support!

closed 2022-05-04 by binhla