espressif/esp-idf #8894: ESP32-S3 I2C System Reset Bug

hasankarainfo, 2022-05-04:
## Environment
- Development Kit:     custom 
static gpio_num_t i2c_gpio_sda = 40;
static gpio_num_t i2c_gpio_scl = 39;

- Module or chip used:  ESP32-S3-WROOM-1-N8R2
- IDF version "cpu_start: ESP-IDF:          v5.0-dev-2586-ga82e6e63d9-dirty"
- Build System:          idf.py
- Compiler version esp-2021r2-patch3-8.4.0
- Operating System:     Windows
- environment type: ESP Command Prompt.
- Using an IDE?:  Eclipse
- Power Supply:         USB

## Problem Description
-System resets when a problem occurs even in simple faults in i2c
-The problem also occurs when I short-circuit sda and scl.

### Expected Behavior
I am trying to upgrade my project, which is designed according to esp32-wroom and works without any problems, to esp32s3. 

Device list I added dynamically in esp32 working without any problem: m24c64 + pcf8563+ scd30 + sgp40 + shtc3 + 2*lps22df + 3*mcp96rl01 + si7021.  It also passes the short-circuit test without any problems. When the short circuit is removed, it continues to communicate.

When I use esp32s3, the system continues to work as long as there is no communication problem. However, the system is reset when any problem or short circuit occurs during communication with any device. 

Doesn't matter if optimization is on or off.  Doesn't matter if psram is on or off. Doesn't matter whether freertos is always running on the first core, active or deactivated. Doesn't matter if peripheral drivers are on iram or not.  Other versions I tried, 4.3.2 | 4.4 | 4.4.1 same result.

I also tried sda and scl pins on io8 and io9 but still no difference.

The other thing I tried is the "i2c_tools" project that I created with the new idf version.  When I just change the sda and scl pins and run the same tests, the result is still the same. 

Below is the output of my test in the i2c_tools project.

Step1: at first i run i2cdetect command with healthy connection and it finds devices
 
```
entry 0x403b61d8                                                              
[0;32mI (97) cpu_start: Pro cpu up.[0m                                      
[0;32mI (97) cpu_start: Starting app cpu, entry point is 0x403753ac[0m      
[0;32mI (75) cpu_start: App cpu up.[0m                                      
[0;32mI (111) cpu_start: Pro cpu start user code[0m                         
[0;32mI (111) cpu_start: cpu freq: 160000000 Hz[0m                          
[0;32mI (111) cpu_start: Application information:[0m                        
[0;32mI (114) cpu_start: Project name:     i2c_s3[0m                        
[0;32mI (119) cpu_start: App version:      1[0m                             
[0;32mI (123) cpu_start: Compile time:     May  4 2022 11:38:38[0m          
[0;32mI (129) cpu_start: ELF file SHA256:  48dde03a7963d38d...[0m           
[0;32mI (135) cpu_start: ESP-IDF:          v5.0-dev-2586-ga82e6e63d9-dirty[0m
[0;32mI (142) heap_init: Initializing. RAM available for dynamic allocation:[0m                                                                             
[0;32mI (150) heap_init: At 3FC97EF0 len 00048110 (288 KiB): D/IRAM[0m      
[0;32mI (156) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM[0m   
[0;32mI (163) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m         
[0;32mI (169) heap_init: At 600FE028 len 00001FD8 (7 KiB): RTCRAM[0m        
[0;32mI (176) spi_flash: detected chip: generic[0m                          
[0;32mI (180) spi_flash: flash io: dio[0m                                   
[0;33mW (184) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m         
[0;32mI (197) sleep: Configure to isolate all GPIO pins in sleep state[0m   
[0;32mI (204) sleep: Enable automatic switching of GPIO sleep configuration[0m
                                                                              
[0;32mI (211) cpu_start: Starting scheduler on PRO CPU.[0m                  
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m                    
[5n                                                                          
 ==============================================================              
 |             Steps to Use i2c-tools                         |               
 |                                                            |               
 |  1. Try 'help', check all supported commands               |               
 |  2. Try 'i2cconfig' to configure your I2C bus              |               
 |  3. Try 'i2cdetect' to scan devices on the bus             |               
 |  4. Try 'i2cget' to get the content of specific register   |               
 |  5. Try 'i2cset' to set the value of specific register     |               
 |  6. Try 'i2cdump' to dump all the register (Experiment)    |               
 |                                                            |               
 ==============================================================

pe 'help' to get the list of commands.                                     
e UP/DOWN arrows to navigate through command history.                        
ess TAB when typing command name to auto-complete.                           
Your terminal application does not support escape sequences.
Line editing and history features are disabled.
Windows, try using Putty instead.
                                           
i2c-tools> i2cdetect
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
: 00 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
: 50 51 -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
: -- 61 62 -- -- -- -- -- -- -- -- -- -- -- -- -- 
: 70 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```
Step 2:  I short-circuit and run the i2cdetect command
```
i2c-tools> i2cdetect
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: UU UU Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
Core  0 register dump:          
PC      : 0x4037765f  PS      : 0x00050034  A0      : 0x40376d8d  A1      : 0x3fc95330                                                                        
A2      : 0x3fce9010  A3      : 0x00000000  A4      : 0x00000004  A5      : 0x4037c348                                                                        
A6      : 0x00060323  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fc95300                                                                        
A10     : 0x00000000  A11     : 0x3fc95338  A12     : 0x00000000  A13     : 0x00000000                                                                        
A14     : 0x3fce90d4  A15     : 0x00060021  SAR     : 0x00000012  EXCCAUSE: 0x00000005                                                                        
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff                                                                        
Core  0 was running in ISR context:                                           
EPC1    : 0x4202e267  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4037765f                                                                        Backtrace:0x4037765c:0x3fc953300x40376d8a:0x3fc95370 0x400559dd:0x3fce79f0  |<-CORRUPTED                                                                      
Core  1 register dump:                                                        
PC      : 0x42002a0e  PS      : 0x00060c34  A0      : 0x8037da08  A1      : 0x3fcf56c0                                                                        
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00000001                                                                        
A6      : 0x3fc9763c  A7      : 0x3fc9763c  A8      : 0x3fc97470  A9      : 0x3fc97434                                                                        
A10     : 0x00000000  A11     : 0x3fcf2a18  A12     : 0x8037f715  A13     : 0x3fcf14f0                                                                        
A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005                                                                        
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Backtrace:0x42002a0b:0x3fcf56c00x4037da05:0x3fcf56e0 0x4037f45d:0x3fcf5700 
ELF file SHA256: 48dde03a7963d38d
Rebooting...
x<ðøESP-ROM:esp32s3-20210327
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

o-marshmallow, 2022-05-12:
Hi @hasankarainfo ,

I saw a similar issue due to that line too in the driver, can you try the fix provided here: https://github.com/espressif/esp-idf/issues/8770#issuecomment-1119561982

dhalbert, 2022-05-13:
@o-marshmallow I am seeing the same _Guru Meditation_ error on the ESP32-S3, in these circumstances:

- **Already using your fix patch from #8770** that you referenced above and provided for me
- Testing in the same version of v4.4 I was testing with in #8770
- Using LC709203F, an I2C "fuel gauge" battery meter that does significant clock-stretching
- Reproduced on several S3 boards, including Espressif ESP32-S3-DevKitC-1-N8R8, with the ESP32-S3 WROOM-1 module.

When I use a different I2C device (MSA311), and the test program in https://github.com/dhalbert/i2c_delay, used in #8770, I don't see these problems.

**Here is a new test program**, specifically for the CL709203F (it does some setup of the device): https://github.com/dhalbert/i2c_s3_stretch
This program works fine on S2 boards, but fails on S3, and gives output like this over and over, failing cyclically:
```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375744
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1660
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f74
entry 0x403b6254
I (29) boot: ESP-IDF v4.4.1-126-g944c01eef4 2nd stage bootloader
I (29) boot: compile time 20:23:23
I (29) boot: chip revision: 0
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08918h ( 35096) map
I (104) esp_image: segment 1: paddr=00018940 vaddr=3fc91d60 size=02718h ( 10008) load
I (108) esp_image: segment 2: paddr=0001b060 vaddr=40374000 size=04fb8h ( 20408) load
I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18d98h (101784) map
I (141) esp_image: segment 4: paddr=00038dc0 vaddr=40378fb8 size=08da4h ( 36260) load
I (150) esp_image: segment 5: paddr=00041b6c vaddr=50000000 size=00010h (    16) load
I (155) boot: Loaded app from partition at offset 0x10000
I (155) boot: Disabling RNG early entropy source...
I (170) cpu_start: Pro cpu up.
I (170) cpu_start: Starting app cpu, entry point is 0x403751c0
I (148) cpu_start: App cpu up.
I (184) cpu_start: Pro cpu start user code
I (184) cpu_start: cpu freq: 160000000
I (184) cpu_start: Application information:
I (187) cpu_start: Project name:     i2c-s3-stretch
I (193) cpu_start: App version:      1
I (197) cpu_start: Compile time:     May 12 2022 20:34:25
I (203) cpu_start: ELF file SHA256:  c667398f33a2a54e...
I (209) cpu_start: ESP-IDF:          v4.4.1-126-g944c01eef4
I (215) heap_init: Initializing. RAM available for dynamic allocation:
I (223) heap_init: At 3FC94E70 len 0004B190 (300 KiB): D/IRAM
I (229) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (236) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (242) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (249) spi_flash: detected chip: gd
I (253) spi_flash: flash io: dio
W (256) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (270) sleep: Configure to isolate all GPIO pins in sleep state
I (276) sleep: Enable automatic switching of GPIO sleep configuration
I (284) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) i2c-s3-stretch: I2C initialized successfully
E (304) i2c-s3-stretch: get_cell_voltage write/read: ESP_ERR_TIMEOUT
E (1314) i2c-s3-stretch: get_cell_voltage write/read: EGuru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
[... 13 lines cut ...]
Core  1 register dump:
PC      : 0x420180ba  PS      : 0x00060334  A0      : 0x82001eb6  A1      : 0x3fcf4e50  
A2      : 0x00000000  A3      : 0x3fcf2fd4  A4      : 0x8037d025  A5      : 0x3fcf1ac0  
A6      : 0x00060023  A7      : 0x00000000  A8      : 0x820095ee  A9      : 0x3fcf4e20  
A10     : 0x00000000  A11     : 0x00000008  A12     : 0x00000019  A13     : 0x3c020129  
A14     : 0x0000000c  A15     : 0xfffffff6  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x420180b7:0x3fcf4e500x42001eb3:0x3fcf4e70 0x4037b83d:0x3fcf4e90 0x4037cd6d:0x3fcf4eb0 

ELF file SHA256: c667398f33a2a54e
```
Here is a Saleae trace: 
[ESP32-S3-I2C-failure-CL709203F.sal.zip](https://github.com/espressif/esp-idf/files/8683355/ESP32-S3-I2C-failure-CL709203F.sal.zip)
Viewable with Saleae Logic, downloadable from here: https://www.saleae.com/downloads/

Here are some screenshots:
[image]

detail of the toggling on the right above:
[image]
write just before clock stretch:
[image]

I mentioned this in https://github.com/espressif/esp-idf/issues/8741#issuecomment-1124457602, but the example above is much better.

hasankarainfo, 2022-05-14:
hi @o-marshmallow. When I updated the relevant lines, it didn't work. Same problem is continue. Below is the updated version of the i2c file.

[i2c.zip](https://github.com/espressif/esp-idf/files/8692778/i2c.zip)

o-marshmallow, 2022-05-23:
Hi guys,

Sorry for the delay of response. I had a try on my end with an ESP32-S3 and when shorting SCL and SDA I got this:
```
i2c-tools> i2cdetect
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU
: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU
```
I didn't get the panic like you did.

What I did next is to try clock stretching. I don't have any device that does clock stretching, so I used another ESP32 to simulate a device doing clock stretching:
[image]

I didn't get the toggling afterwards. I will continue investigating.

@hasankarainfo The zip you uploaded seems corrupted, I cannot extract the file inside.
Out of curiosity, how did you set up your I2C bus? Which frequency? Do you have external pull-ups? How big are they?

dhalbert, 2022-05-23:
> The zip you uploaded seems corrupted, I cannot extract the file inside.

I looked at the header a while ago, and I think it's an RAR file.

o-marshmallow, 2022-05-23:
@dhalbert Oh indeed, works better now, thanks!

hasankarainfo, 2022-05-23:
Since the github does not accept rar, I changed the extension to zip. 
I remember that the i2c_tools project was not able to use the signal line at all in its default pin numbers. 
```
static gpio_num_t i2c_gpio_sda = 40;
static gpio_num_t i2c_gpio_scl = 39;
```
can you try this ?

hasankarainfo, 2022-05-23:
Today, I tried removing idf and doing master checkout again. same result. 

However, I realized that I missed an important detail about the test. Keeping it short-circuited throughout the test, does not cause the problem to occur.
```
Step1: 
sda-scl make short circuit

Step2:
>>i2cdetect

Step3:
the following see in console
"0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f"
"00: UU UU UU UU UU"

Step4:
sda-scl remove short circuit

Step5:
Sys reset...
"00: UU UU UU UU UU UU ESP-ROM:esp32s3-20210327"

```

o-marshmallow, 2022-05-23:
@hasankarainfo That's exactly what I noticed in my case too with the commit you are using. For some reasons I can't reproduce it with the latest master

mythbuster5, 2022-05-25:
Yeah.. I saw your issue and I have merged https://github.com/espressif/esp-idf/commit/0b00831703ec351c4c3b1040b004b358e4396067 into master. Which can solve the short circuit issue on master. @hasankarainfo

dhalbert, 2022-06-05:
I adapted 0b00831 to ESP-IDF v4.4.1, but I am still seeing the same clock stretch issue mentioned above in https://github.com/espressif/esp-idf/issues/8894#issuecomment-1125564756. If you want me to open a new issue for that, I can do so.

mythbuster5, 2022-06-13:
> I adapted [0b00831](https://github.com/espressif/esp-idf/commit/0b00831703ec351c4c3b1040b004b358e4396067) to ESP-IDF v4.4.1, but I am still seeing the same clock stretch issue mentioned above in [#8894 (comment)](https://github.com/espressif/esp-idf/issues/8894#issuecomment-1125564756). If you want me to open a new issue for that, I can do so.

You can still see the timeout after the fix?

dhalbert, 2022-06-13:
> You can still see the timeout after the fix?

Yes, I still see the same thing I saw in https://github.com/espressif/esp-idf/issues/8894#issuecomment-1125564756, with SDA going high after a while, when testing with the LC709203F. Have you backported this fix to v4.4? Could you get an LC709203F to test with?

mythbuster5, 2022-06-14:
> 

The fix will go into 4.4.2. And I can't get LC709203F currently... Emm it's so odd that can't solve with that commit. The thing you can make it sure first is to check ``SOC_I2C_SUPPORT_HW_FSM_RST`` under `esp32s3/include/soc/soc_caps.h` has been disabled?

BTW, I'm curious that are you using wifi or BLE with it together?

dhalbert, 2022-06-15:
@mythbuster5 

I did some more testing. I adapted your 0b00831 commit to v4.4.1 with this patch: 
[v44-i2c-patch.txt](https://github.com/espressif/esp-idf/files/8905224/v44-i2c-patch.txt). That patch is in this fork branch: https://github.com/dhalbert/esp-idf/tree/0b00831-adapted-to-v4.4.

I rebuilt my test program (https://github.com/dhalbert/i2c_s3_stretch) against the current `master` of https://github.com/espressif/esp-idf/. As you hoped, it does seem to fix the problem. Before your patch, the test program demonstrated the bug.

I also tested my test program with https://github.com/dhalbert/esp-idf/tree/0b00831-adapted-to-v4.4. That also worked. So I think my adaptation of your patch is OK. As you mentioned, `SOC_I2C_SUPPORT_HW_FSM_RST` under `esp32s3/include/soc/soc_caps.h` has been disabled.

But unfortunately when I tested my adapted patch against CircuitPython, I still see the problem. In #8770, I was able to replicate the CircuitPython problem by adding a second task to a simple ESP-IDF program, to simulate what CircuitPython was doing with tasks. But that didn't work here. https://github.com/dhalbert/i2c_s3_stretch has some commented-out code to add the second task, and uncommenting that still does not cause the problem.

So I will work on this some more, seeing just what might be different in the CircuitPython case. If the 4.4.2 release is imminent, I could test CircuitPython against that.

dhalbert, 2022-08-23:
Update: I tested again today with the tip of `release/v4.4` (2bce0a19f), which is past 4.4.2, and am still see the problem.

AxelLin, 2022-12-23:
@dhalbert 
Check if this helps: https://github.com/espressif/esp-idf/issues/9777#issuecomment-1359582249

hasankarainfo, 2022-12-23:
i will deal with it soon

dhalbert, 2022-12-23:
@AxelLin Thank you! I will test this soon.

hasankarainfo, 2023-01-13:
Hello after a long time. Today I had the opportunity to do the relevant tests again.  

Some test conditions;

Development Kit: custom
static gpio_num_t i2c_gpio_sda = 42;
static gpio_num_t i2c_gpio_scl = 41;
static uint32_t i2c_frequency = 10000;
static i2c_port_t i2c_port = I2C_NUM_0;
Module or chip used: ESP32-S3-WROOM-1-N16R8
IDF version "ESP-IDF:          v5.0"
Compiler version esp-2022r1-11.2.0
Operating System: Windows
Using an IDE: VSCODE

[image]

As in the picture, the result when I scan:

i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- 59 -- -- -- 5d -- --
60: -- 61 -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- -- -- -- -- -- -- -- UU --
i2c-tools>

When the dat and clk pins are short-circuited, when I start the scan and remove the short-circuit when I come to the 5th address, the result is:

i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: UU UU UU UU UU UU UU -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- 59 -- -- -- 5d -- --
60: -- 61 -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- -- -- -- -- -- -- -- UU --
i2c-tools>

The result is satisfactory for me. Thanks to everyone who took an interest in the issue and contributed to the fix.

closed 2023-01-13 by hasankarainfo

BeatArnet, 2023-01-14:
Thank you, reading the battery voltage now works without a workaround.

dhalbert, 2023-11-05:
I retested in CircuitPython with an LC709203F after we upgraded to ESP-IDF v5.1, and it seems to work much better now. Not sure what has been fixed in 5.1, but that's good to see.