espressif/esp-idf #8895: tls transport StoreProhibited exception when connection closed by server

MarcGeh, 2022-05-04:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version: v3
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.4.1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Hi guys, I'm currently trying to bring up the azure-iot-middleware-freertos to connect the ESP32 to azure. I'm using tls sockets for the communication based on the esp32 example [iot-middleware-freertos-samples](https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/esp32)

Once the server is closing the connection, the firmware crashes with the unhandled exception "StoreProhibited".
I'm using esp_transport_ssl_init() to initialize the context for the transaction. 
esp_transport_connect and subsequent calls to esp_transport_read/esp_transport_write are successful as well, until the server closes the connection, then I get the exception on a call of esp_transport_read.

I've been tracing the issue back to **transport.c**, where in the function **capture_tcp_transport_error** the **err_handle** from the **esp_transport_handle_t** is used to save the error code of the transaction. However, the **err_handle** was never initialized and thus a write into a memory region referenced by a NULL pointer is attempted, which causes the problem. I'm writing this issue since it makes the impression onto me, that **esp_transport_ssl_init** does not initialize the context properly, since the **esp_transport_init** function called by the **esp_transport_ssl_init** function never initializes the underlying transport foundation which is referenced by the **capture_tcp_transport_error** function.

### Steps to reproduce

You can try to run the example from [iot-middleware-freertos-samples](https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/esp32) repository, but setting it up is quite a hassle.

## Debug Logs

```
I (10454) MQTT: Packet received. ReceivedBytes=2.
I (10454) MQTT: CONNACK session present bit not set.
I (10454) MQTT: Connection accepted.
I (10474) MQTT: Received MQTT CONNACK successfully from broker.
I (10474) MQTT: MQTT connection established with the broker.
I (10484) IOT: AzureIoTProvisioning established an MQTT connection with global.azure-devices-provisioning.net
I (14484) IOT: AzureIoTProvisioning attempting to subscribe to the MQTT topic: devices/+/messages/devicebound/#
I (14654) MQTT: Packet received. ReceivedBytes=3.
I (18964) MQTT: Packet received. ReceivedBytes=192.
I (18964) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (18964) MQTT: State record updated. New state=MQTTPublishDone.
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4022d1e3  PS      : 0x00060e30  A0      : 0x8022da42  A1      : 0x3fff0710  
0x4022d1e3: capture_tcp_transport_error at /home/marc/esp/esp-idf_4_4/components/tcp_transport/transport.c:332

A2      : 0x3fff2698  A3      : 0x00008008  A4      : 0x00000001  A5      : 0x3f41212e  
A6      : 0x00000001  A7      : 0x0000000f  A8      : 0x8022d1b4  A9      : 0x3fff06e0  
A10     : 0x00000000  A11     : 0x3287c132  A12     : 0x3287c132  A13     : 0x3fff2740  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4022d1e0:0x3fff07100x4022da3f:0x3fff0740 0x4022cfb2:0x3fff0780 0x4010851f:0x3fff07b0 0x4010c3cd:0x3fff07f0 0x4010ab5d:0x3fff0820 0x4010ac9c:0x3fff0850 0x40109574:0x3fff0880 0x40108f6e:0x3fff08b0 0x40109264:0x3fff08f0 0x401002be:0x3fff0920 0x401003f1:0x3fff0a90 0x4010041f:0x3fff0ac0 0x40287ff9:0x3fff0af0 0x400d3630:0x3fff0b10 0x40096402:0x3fff0b40 
0x4022d1e0: capture_tcp_transport_error at /home/marc/esp/esp-idf_4_4/components/tcp_transport/transport.c:332

0x4022da3f: ssl_read at /home/marc/esp/esp-idf_4_4/components/tcp_transport/transport_ssl.c:241

0x4022cfb2: esp_transport_read at /home/marc/esp/esp-idf_4_4/components/tcp_transport/transport.c:190
```

david-cermak, 2022-05-04:
Hi @MarcGeh 

Thanks for the report!
This is a bug in `tcp_transport` component, the reason why it's not initialized is probably that the project uses the component without a list (internal container for multiple transport layers). This scenario is supported and should work properly, but it's not so widely used. Internally we use the transport only in lists, for example:

https://github.com/espressif/esp-idf/blob/8c8f700c1df05d8287316c78b98b940e119f900c/components/esp_websocket_client/esp_websocket_client.c#L332-L339

Sorry for the inconvenience, will fix it soon (similar issue has been fixed recently in e794786c748, again partly initialized transport, using one layer separately)

MarcGeh, 2022-05-04:
Hi @david-cermak thanks for the fast response. I need to use websockets in the future anyways, so I might switch over right away to avoid this issue.

tmaltesen, 2022-06-22:
Is there a workaround for this issue. Alle esp32 examples seems to be based on the same initialization using TLS. I am not sure how to fix this.

tmaltesen, 2022-06-23:
Thanks a lot for the info.
I am new to this environment and are not exactly sure how to do it. Should I create a transport_list myself... can you give a little details. What is in the list, how to register it etc.

tmaltesen, 2022-06-23:
Thanks again :-)
My problem is am an not using it directly, but through the "Azure IoT middleware for freertos" layer. 
So it is deep into some code I have not deep insight into. And of course I need to get it running "yesterday" ;-(

MarcGeh, 2022-06-23:
@tmaltesen Unfortunately you will not make this work unless you are willing to make some minor changes to the [transport_tls_esp32.c](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ESPRESSIF/esp32/components/sample-azure-iot/transport_tls_esp32.c) file. If this is an option for you I can guide you through this.

tmaltesen, 2022-06-23:
This is an option, absolutely 😊

tmaltesen, 2022-06-23:
Very much appreciated if you have a workaround that I am able to implement.
The esp-idf is completely new to me but I’m experienced C/C++ embedded programmer.
But still struggling a bit to learn the build environment.

/Thomas

MarcGeh, 2022-06-23:
Change the NetworkContext definition to have an esp_transport_list_handle_t and a scheme name instead of the [esp_transport_handle_t](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/154ba9fbd280b855c6e49e6e9c9ee34c35705c0f/demos/projects/ESPRESSIF/esp32/components/sample-azure-iot/transport_tls_esp32.c#L47) so it looks like this:
```c
struct NetworkContext {
  const char* scheme;
  esp_transport_list_handle_t xTransportList;
  uint32_t ulReceiveTimeoutMs;
  uint32_t ulSendTimeoutMs;
};
```

Now in each function where pNetworkContext->xTransport is used, replace it by a local variable initialized like this:
```c
  esp_transport_handle_t xTransport = esp_transport_list_get_transport(pNetworkContext->xTransportList, pNetworkContext->scheme);
  if (xTransport == NULL) {
    ESP_LOGE(TAG, "Failed to get transport handle for scheme %s", pNetworkContext->scheme);
    return eTLSTransportInvalidParameter;
  }
```

You still have to properly initialize the network struct and hand it over to the IoT client as part of the AzureIoTTransportInterface_t. My initialization of it looks e.g. like this:
```c
  memset(&network_context_, 0, sizeof(NetworkContext_t));
  network_context_.xTransportList = esp_transport_list_init();
  network_context_.scheme = "_mqtt";
  if (network_context_.xTransportList == NULL) {
    ESP_LOGE(TAG, "Failed to create transport list");
  } else {
    esp_transport_handle_t mqtt = esp_transport_ssl_init();
    if (mqtt == NULL) {
      ESP_LOGE(TAG, "Failed to create transport handle");
    } else {
      esp_transport_list_add(network_context_.xTransportList, mqtt, network_context_.scheme);
    }
  }
```
Keep in mind that in the original example the call to esp_transport_ssl_init() was inside the TLS_Socket_Connect function. Now the initialization is handled outside of that module, so you have to take care to also tear it down once you are done with it.
For completeness, this is how your functions inside transport_tls_esp32.c should look like:
```c
/*-----------------------------------------------------------*/

TlsTransportStatus_t TLS_Socket_Connect(NetworkContext_t *pNetworkContext, const char *pHostName, uint16_t usPort,
                                        const NetworkCredentials_t *pNetworkCredentials, uint32_t ulReceiveTimeoutMs, uint32_t ulSendTimeoutMs) {
  TlsTransportStatus_t xReturnStatus = eTLSTransportSuccess;

  if ((pNetworkContext == NULL) || (pHostName == NULL) || (pNetworkCredentials == NULL)) {
    ESP_LOGE(TAG,
             "Invalid input parameter(s): Arguments cannot be NULL. pNetworkContext=%p, "
             "pHostName=%p, pNetworkCredentials=%p.",
             pNetworkContext, pHostName, pNetworkCredentials);
    return eTLSTransportInvalidParameter;
  }

  esp_transport_handle_t xTransport = esp_transport_list_get_transport(pNetworkContext->xTransportList, pNetworkContext->scheme);
  if (xTransport == NULL) {
    ESP_LOGE(TAG, "Failed to get transport handle for scheme %s", pNetworkContext->scheme);
    return eTLSTransportInvalidParameter;
  }
  pNetworkContext->ulReceiveTimeoutMs = ulReceiveTimeoutMs;
  pNetworkContext->ulSendTimeoutMs = ulSendTimeoutMs;
  if (pNetworkCredentials->ppcAlpnProtos) {
    esp_transport_ssl_set_alpn_protocol(xTransport, pNetworkCredentials->ppcAlpnProtos);
  }

  if (pNetworkCredentials->xDisableSni) {
    esp_transport_ssl_skip_common_name_check(xTransport);
  }

  if (pNetworkCredentials->pucRootCa) {
    esp_transport_ssl_set_cert_data(xTransport, pNetworkCredentials->pucRootCa, strlen(pNetworkCredentials->pucRootCa));
  }
[... 75 lines cut ...]

  if ((pNetworkContext == NULL) || (pBuffer == NULL) || (xBytesToSend == 0)) {
    ESP_LOGE(TAG,
             "Invalid input parameter(s): Arguments cannot be NULL. pNetworkContext=%p, "
             "pBuffer=%p, xBytesToSend=%d.",
             pNetworkContext, pBuffer, xBytesToSend);
    return eTLSTransportInvalidParameter;
  }

  esp_transport_handle_t xTransport = esp_transport_list_get_transport(pNetworkContext->xTransportList, pNetworkContext->scheme);
  if (xTransport == NULL) {
    ESP_LOGE(TAG, "Failed to get transport handle for scheme %s", pNetworkContext->scheme);
    return eTLSTransportInvalidParameter;
  }

  tlsStatus = esp_transport_write(xTransport, pBuffer, xBytesToSend, pNetworkContext->ulSendTimeoutMs);
  if (tlsStatus < 0) {
    ESP_LOGE(TAG, "Writing failed, errno= %d", errno);
    return ESP_FAIL;
  }

  return tlsStatus;
}
/*-----------------------------------------------------------*/
```

tmaltesen, 2022-06-23:
So great, a single clearifcations about initialization and cleanup.

The initialization code:

  memset(&network_context_, 0, sizeof(NetworkContext_t));

  network_context_.xTransportList = esp_transport_list_init();

  network_context_.scheme = "_mqtt";
     …..

Did you put that in the prvConnectToServerWithBackoffRetries or even higher into the prvAzureTask. It makes sense in the main AzureTask since the destroy of the transport list can be done when /if the task loop is exited.

/Thomas

MarcGeh, 2022-06-23:
Make it on a as high as possible layer, since it is actually encouraged to reuse the same handle for as long as possible.

tmaltesen, 2022-06-23:
Great so much thank you – it seems to work.
Now the IoT loop iteration in the demo is continuing to run.
So it disconnects and connects again without crashing.

/Thomas

closed 2023-01-13 by euripedesrocha