espressif/esp-idf #8896: Crash in tcpip_thread when using esp_vfs_select() on both socket and UART simultaneously

hadrava, 2022-05-04:
## Environment

- Development Kit:      none (board: WT32-ETH01)
- Module or chip used:  WT32-S1 (ESP32-D0WD)
- IDF version: `v5.0-dev-2586-ga82e6e63d9`
- Build System:         idf.py
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0`
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I am using `select()` (most probably `esp_vfs_select()`) to wait for data on TCP socket and UART simultaneously. The application sometimes crashes in `tcpip_thread` when there is incoming traffic on both file descriptors. The problem disappears when the task is pinned to the same core as the `tcpdip_thread`. I believe that pining should not be necessary, as it is quite limiting.

I also tried to enable config option `CONFIG_LWIP_TCPIP_CORE_LOCKING=y`. But this disables `select()`'s ability to be interrupted on UART data.

### Expected Behavior

No crashes on `select()` should happen. Especially when it is explicitly allowed by documentation of option:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-vfs-support-select

### Actual Behavior

System crashes inside `tcpip_thread`
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x400f7fd8  PS      : 0x00060b30  A0      : 0x800e4efc  A1      : 0x3ffb90f0  
0x400f7fd8: lwip_netconn_do_write at /home/had/esp-idf/components/lwip/lwip/src/api/api_msg.c:1920
```

### Steps to reproduce

1. navigate to `tcp_server` example provided with `esp-idf` (`esp-idf/examples/protocols/sockets/tcp_server`)
2. replace `tcp_server.c` with slightly modified version from: https://gist.github.com/hadrava/55d5612d44f98df76a41a8a9158ab502#file-tcp_server-c
3. adjust UART pin assignments (on line 208) for particular board
4. configure ethernet, disable WiFi and IPv6. (or use linked `sdkconfig.defaults` for WT32-ETH01 board)
5. connect to tcp server and send some data to it (for example `yes | nc 10.11.13.10 3333`)
6. send few bytes to UART2

### Code to reproduce this issue

Simplified version - this also crashes, even without `send(sock)`:
```cpp
        fd_set rfds;
        FD_ZERO(&rfds);
        FD_SET(sock, &rfds);
        FD_SET(uart_fd, &rfds);
        struct timeval timeout_tv = {
            .tv_sec = 5,
            .tv_usec = 0,
        };
        int max_fd = (sock > uart_fd) ? sock : uart_fd;
        int ret = select(max_fd + 1, &rfds, NULL, NULL, &timeout_tv);

        if (ret > 0 && FD_ISSET(sock, &rfds)) {
            len = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
        }
        if (ret > 0 && FD_ISSET(uart_fd, &rfds)) {
            len = read(uart_fd, rx_buffer, 1);
        }

```
https://gist.github.com/hadrava/55d5612d44f98df76a41a8a9158ab502#file-tcp_server-c

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x400f7fd8  PS      : 0x00060b30  A0      : 0x800e4efc  A1      : 0x3ffb90f0  
0x400f7fd8: lwip_netconn_do_write at /home/had/esp-idf/components/lwip/lwip/src/api/api_msg.c:1920

A2      : 0x3ffbf850  A3      : 0x3ffb9130  A4      : 0x00000000  A5      : 0x000005d6  
A6      : 0x3f409650  A7      : 0x400e4f68  A8      : 0x800f7fd0  A9      : 0x00000000  
0x400e4f68: tcpip_thread at /home/had/esp-idf/components/lwip/lwip/src/api/tcpip.c:134

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffb8358  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x400f7fd5:0x3ffb90f00x400e4ef9:0x3ffb9110 0x400e4f98:0x3ffb9130 0x40089cc1:0x3ffb9160 
0x400f7fd5: lwip_netconn_do_write at /home/had/esp-idf/components/lwip/lwip/src/api/api_msg.c:1919
0x400e4ef9: tcpip_thread_handle_msg at /home/had/esp-idf/components/lwip/lwip/src/api/tcpip.c:168
0x400e4f98: tcpip_thread at /home/had/esp-idf/components/lwip/lwip/src/api/tcpip.c:154
0x40089cc1: vPortTaskWrapper at /home/had/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141
```

https://gist.github.com/hadrava/55d5612d44f98df76a41a8a9158ab502#file-full-console-log-txt

atanisoft, 2022-05-05:
I've been able to reproduce this (and similar) failures for a while now, see some additional traces in https://github.com/atanisoft/ESP32CommandStation/issues/64. My usage is similar to your's but a bit more complex, I'm using a VFS implementation to wake the task early from `select()` which may or may not include sockets. I've also added VFS support to the TWAI driver but have not submitted a PR for this as the implementation is specific to my project (not using `twai_message_t` for read/write)

I've done some digging into VFS layers and I suspect there may also be a bug where VFS wakes LwIP for a non-socket `select()` being "ready" to act on.

david-cermak, 2022-05-05:
@hadrava Thank you for this clean and concise report with very detailed STR, much appreciated! I was able to recreate the issue very quickly.

The problem seems to be in using the same semaphore for waking a blocked select and for synchronizing messages (mboxes) in lwIP. This works perfectly with lwip-only sockets, since this is a thread local semaphore and, as such, is used sequentially within one thread only.

I don't see a good solution at this moment with IDF default settings: `LWIP_NETCONN_SEM_PER_THREAD=1`.
As for the `LWIP_TCPIP_CORE_LOCKING`, we will still have to implement vfs integration in this mode. (This would fix the problem, as a new, designated semaphore gets created on every `select()` call, the problem is how to access it from vfs, tho) -- Again, thanks for checking this mode, the integration with vfs was somehow forgotten.

My naive fix at this moment would be:
```patch
diff --git a/components/vfs/vfs.c b/components/vfs/vfs.c
@@ -1061,6 +1061,12 @@ int esp_vfs_select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *errorfds
     if (sel_sem.is_sem_local && sel_sem.sem) {
         vSemaphoreDelete(sel_sem.sem);
         sel_sem.sem = NULL;
+    } else if (sel_sem.sem && sel_sem.is_sem_local == false) {
+        SemaphoreHandle_t  *s = sel_sem.sem;
+        if (xSemaphoreTake(*s, 0)) {
+//            static int count = 0;
+//            printf("Taken:%d!\n", count++);
+        }
     }
     _lock_acquire(&s_fd_table_lock);
     for (int fd = 0; fd < nfds; ++fd) {
```
but will have to make sure it really works in all cases.

closed 2022-05-19 by espressif-bot