espressif/esp-idf #8898: CMake error after updating from v4.3.1 or v4.3.2 to v4.4 or v4.4.1

vdm97, 2022-05-04:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.3.1, v4.3.2, v4.4, v4.4.1
- Build System: CMake
- Compiler version: Standard version of each release
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: VS Code with Espressif IDF v1.4.0
- Power Supply: Implemented in custom board

## Problem Description

When updating from v4.3.1 or v4.3.2 to v4.4 or v4.4.1 i get an compilation error. Somehow `CONFIG_BOOTLOADER_OFFSET_IN_FLASH` does not get set to the real value, when the file `build/bootloader/build.ninja` gets generated. Therefore the file contains the following:

```
#############################################
# Custom command for esp-idf\esptool_py\CMakeFiles\bootloader_check_size

build esp-idf/esptool_py/CMakeFiles/bootloader_check_size: CUSTOM_COMMAND || _project_elf_src bootloader.elf esp-idf/bootloader_support/libbootloader_support.a esp-idf/efuse/libefuse.a esp-idf/esp_common/libesp_common.a esp-idf/esp_hw_support/libesp_hw_support.a esp-idf/esp_rom/libesp_rom.a esp-idf/esp_system/libesp_system.a esp-idf/hal/libhal.a esp-idf/log/liblog.a esp-idf/main/libmain.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/soc/libsoc.a esp-idf/spi_flash/libspi_flash.a esp-idf/xtensa/libxtensa.a gen_project_binary
  COMMAND = cmd.exe /C "cd /D C:\path\build\bootloader\esp-idf\esptool_py && c:\Users\user\esp\esp-idf-tools\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader CONFIG_BOOTLOADER_OFFSET_IN_FLASH C:/path/build/bootloader/bootloader.bin"

# =============================================================================
# Write statements declared in CMakeLists.txt:
# C:/Users/user/esp/esp-idf/CMakeLists.txt
# =============================================================================
#############################################
```

If i manually change `CONFIG_BOOTLOADER_OFFSET_IN_FLASH` to `0x1000` the compilation succeeds. I still get [this](https://github.com/espressif/esp-idf/issues/8897) error. But that error is probably unrelated so i created a seperate issue.

## Debug Logs

```
> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

-- Found Git: C:/Users/user/esp/esp-idf-tools/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/user/esp/esp-idf-tools/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/user/esp/esp-idf-tools/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/user/esp/esp-idf-tools/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/path/sdkconfig_debug
Loading defaults file C:/path/sdkconfig_debug...
-- Found PythonInterp: C:/Users/user/esp/esp-idf-tools/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7")
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "esp-app" version: 100
-- Adding linker script C:/path/build/esp-idf/esp_system/app/memory.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_system/app/esp32/sections.app.in
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.api.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.libgcc.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.newlib-data.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.syscalls.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.newlib-funcs.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/app/esp32.rom.newlib-time.app
-- Adding linker script C:/Users/user/esp/esp-idf/components/soc/esp32/app/esp32.peripherals.app
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth 
esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib 
[... 139 lines cut ...]
[90/97] Linking C static library esp-idf\spi_flash\libspi_flash.a
[91/97] Linking C static library esp-idf\micro-ecc\libmicro-ecc.a
[92/97] Linking C static library esp-idf\soc\libsoc.a
[93/97] Linking C static library esp-idf\hal\libhal.a
[94/97] Linking C static library esp-idf\main\libmain.a
[95/97] Linking C executable bootloader.elf
[96/97] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated C:/path/build/bootloader/bootloader.bin
[97/97] cmd.exe /C "cd /D C:\path\build\bootloader\esp-idf\esptool_py && c:\Users\user\esp\esp-idf-tools\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader CONFIG_BOOTLOADER_OFFSET_IN_FLASH C:/path/build/bootloader/bootloader.bin"
FAILED: esp-idf/esptool_py/CMakeFiles/bootloader_check_size
cmd.exe /C "cd /D C:\path\build\bootloader\esp-idf\esptool_py 
&& c:\Users\user\esp\esp-idf-tools\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader CONFIG_BOOTLOADER_OFFSET_IN_FLASH C:/path/build/bootloader/bootloader.bin"
Traceback (most recent call last):
  File "C:/Users/user/esp/esp-idf/components/partition_table/check_sizes.py", line 148, in <module>
    main()
  File "C:/Users/user/esp/esp-idf/components/partition_table/check_sizes.py", line 142, in main
    check_bootloader(args.offset, int(args.bootloader_offset, 0), args.bootloader_binary)
ValueError: invalid literal for int() with base 0: 'CONFIG_BOOTLOADER_OFFSET_IN_FLASH'
ninja: build stopped: subcommand failed.
[609/1020] Linking C static library esp-idf\nghttp\libnghttp.a
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

igrr, 2022-05-04:
@vdm97 does this issue happen if you build one of IDF example projects? In your project, does `idf.py fullclean` help with this issue?

igrr, 2022-05-04:
I think I understand now, it looks like your project is using a customized bootloader? If you have a custom bootloader derived from IDF 4.3.x, it will not be built correctly with IDF 4.4.x. Could you please try to run your application with the standard bootloader provided in IDF first, to see if that is the issue?
(I think it is also the reason for the other issue you have reported.)

vdm97, 2022-05-05:
@igrr Thank you for your fast response. Your finding of the custom bootloader is correct. Using `idf.py fullclean` does not help.
I am now trying to replace the custom bootloader with the standard one. Did the custom bootloader configuration change completely from v4.3 to v4.4? I cannot find [custom bootloader examples for version v4.3.2](https://github.com/espressif/esp-idf/tree/v4.3.2/examples) but there is an [example for version v4.4.1](https://github.com/espressif/esp-idf/tree/v4.4.1/examples/custom_bootloader). I inherited the project so i do not fully understand the layout but there is a `components\bootloader\` subdirectory in the project. Should i move the relevant contents to the new `bootloader_components\main\` _format_? And what would be easier: Creating a new project and moving all content or keeping the project but replacing `components\bootloader\` with `bootloader_components\main\`?

igrr, 2022-05-05:
I think the easiest way to move you bootloader customizations is as follows:

1. Compare (using some diff tool) your project's `components\bootloader` subdirectory with `%IDF_PATH%\components\bootloader`. Find the bootloader customizations which were done in your project.
2. If the customizations are small and can be done using [bootloader hooks](https://github.com/espressif/esp-idf/tree/master/examples/custom_bootloader/bootloader_hooks), this is the best option. It is much more likely that such customizations will be portable to newer versions of ESP-IDF. In this case, remove `components\bootloader`and add `bootloader_components\my_boot_hooks` as shown in that example, then port over your custom functionality.
3. If the customizations are more significant, try to do them by [overriding the `main` component of the bootloader](https://github.com/espressif/esp-idf/tree/master/examples/custom_bootloader/bootloader_override). In this case, remove `components\bootloader` and add `bootloader_components\main`, as shown in the example,  then port your custom functionality.

Both of these options are more future-proof than overriding the complete bootloader.

vdm97, 2022-05-05:
Thank you very much for your time! Option 2 worked very nice for me.

closed 2022-05-05 by vdm97