espressif/esp-idf #8899: wake up deep sleep from timer with rtc clock set at 8MHz /256

ElieSKickMaker, 2022-05-04:
Hello, 

 I'm using the [sntp example ](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sntp). When by menuconfig, I set : 
 - CONFIG_RTC_CLK_SRC = Internal 150KHz , it's working and the cpu is waked up after 10 s 
 - CONFIG_RTC_CLK_SRC = Internal 8MHz, it's not working, the cpu is never waked up. 
 
 Do I miss something ?

AxelLin, 2022-05-05:
https://github.com/espressif/esp-idf/issues/8007

AxelLin, 2022-07-14:
@ElieSKickMaker 
Which esp-idf version are you using?
The fix should be available in v4.4 and newer branches. (But the v4.3 still needs fix)
This is duplicated of https://github.com/espressif/esp-idf/issues/8007

cssword, 2022-07-29:
#9455 

The cpu waked up when use 8MHz RTC in my case. 
But the sleep time is more than 3 minutes and printed time is 10 s after the before printed time.