espressif/esp-idf #8903: Random crashing with automatic light sleep and mqtt

patrickwilliamson1, 2022-05-04:
## Environment

- Module or chip used:  ESP32-WROOM-32E
- IDF version ESP-IDF v4.4-dirty
- Build System:      idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:      Battery

## Problem Description

Application subscribes to an MQTT topic and publishes to a different topic every 60s. The timing of this is controlled by a task called sleep_manager. It uses vtaskDelay to wake up from light sleep, read uart sensor data, publish mqtt and then release the light sleep lock. 

After a 1000-10000 mqtt mesages, the program crashes, and attached core dumps are saved using core_dump to flash config setting.

Any ideas where to look to find the cause of this? seems like most threads are at /include/hal/cpu_ll.h:39 when it happens

void sleep_manager(void *arg)
{
  app_handle_t *app_handle = (app_handle_t*)arg;

  while (1)
  {
    if (power_data.light_sleep_wake_ms < (esp_timer_get_time() / 1000) || power_data.light_sleep_wake_req)
    {        
        // printf("\nhere1");
      if(power_data.light_sleep_state == LS_SLEEPING)
      {
        esp_pm_lock_acquire(handle);   
      }

      power_data.light_sleep_state = LS_AWAKE;
      power_data.light_sleep_wake_req = false;
      power_data.light_sleep_wake_ms = esp_timer_get_time() / 1000 + get_light_sleep_s()*1000;
      power_event(PWR_EVENT_LIGHT_SLEEP_WAKE, app_handle);

    }

    if(power_data.light_sleep_start_req == true)
    {
      power_data.light_sleep_start_req = false;
      if(power_data.light_sleep_state == LS_AWAKE)
      {
        power_data.light_sleep_state = LS_SLEEPING;
        esp_pm_lock_release(handle);   
      }
    }
}

### Expected Behavior

no crashing

### Actual Behavior

crash

### Steps to reproduce

unsure, happens randomly

### Code to reproduce this issue

private, cannot share but can share snippets

## Other items if possible

here are 2 core dumps from 2 separate hardwares:
[... 372 lines cut ...]
.coredump.tasks.data 0x3ffce438 0x164 RW
.coredump.tasks.data 0x3ffcea20 0x17c RW
.coredump.tasks.data 0x3ffd57a0 0x164 RW
.coredump.tasks.data 0x3ffd55d0 0x1c8 RW
.coredump.tasks.data 0x3ffd2574 0x164 RW
.coredump.tasks.data 0x3ffd23f0 0x17c RW
.coredump.tasks.data 0x3ffb228c 0x164 RW
.coredump.tasks.data 0x3ffb1ee0 0x3a4 RW
.coredump.tasks.data 0x3ffba07c 0x164 RW
.coredump.tasks.data 0x3ffb7d70 0x18c RW
.coredump.tasks.data 0x3ffceba4 0x164 RW
.coredump.tasks.data 0x3ffcf3b0 0x158 RW
.coredump.tasks.data 0x3ffd8cd4 0x164 RW
.coredump.tasks.data 0x3ffd8af0 0x1dc RW
.coredump.tasks.data 0x3ffba648 0x164 RW
.coredump.tasks.data 0x3ffba4c0 0x180 RW
.coredump.tasks.data 0x3ffdb320 0x164 RW
.coredump.tasks.data 0x3ffdb160 0x1b8 RW
.coredump.tasks.data 0x3ffbb5c4 0x164 RW
.coredump.tasks.data 0x3ffbb410 0x1ac RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8627820/sdkconfig.txt)

xiaojie76, 2022-05-07:
Hi @patrickwilliamson1,  first look at the information of board 1,
```
==================== CURRENT THREAD STACK =====================
#0 0x40091d09 in _xt_context_save () at C:/Users/p/OneDrive/Desktop/esp-idf-2/components/freertos/port/xtensa\xtensa_context.S:195
#1 0x7ffceaf0 in ?? ()
#2 0x40090c6b in esp_light_sleep_start () at C:/Users/p/OneDrive/Desktop/esp-idf-2/components/esp_hw_support/sleep_modes.c:687
```
It looks like an exception triggered near the execution of the` #2 0x40090c6b` instruction, and the exception was triggered again in subsequent panic handler, so the last seen exception cause is DoubleException. 
You can check the disassembly file to confirm what `#2 0x40090c6b` does, why the pc jumps to `#1 0x7ffceaf0` next (this is an illegal address), it may be caused by the destruction of the stack or heap.

jack0c, 2022-06-15:
@patrickwilliamson1 Suggest debug in 2 directions:
1. under voltage in some sleep progress, you can update to latest IDFv4.4 if possible. If you can help us to reproduce the issues, it will be the best.
2. flash issue, can you tell if you power down flash during the light sleep.

jack0c, 2022-07-07:
@patrickwilliamson1 any updates?

patrickwilliamson1, 2022-07-07:
This can be closed, it was a power supply issue

closed 2022-08-02 by ESP-Marius