espressif/esp-idf #8904: Nimble Master can't connect mulit slave?

novumdun, 2022-05-05:
When Nimble Master have connected one slave, the Master can't connect other device. Because ble_gap_conn_active will return "BLE_HS_EALREADY" error when ble_gap_connect is called.

IshaESP, 2022-05-10:
Hi @novumdun ,
Can you provide more information regarding this?
1. Are you using an existing application? Is yes, which one?  If not, then can you please tell us what exactly are you doing to reproduce this issue?
2. I tried to reproduce it on my end using blecent example, with some modifications to replicate this issue. 
The current code of blecent supports one connection to slave followed by one round of gatt read write subscribe procedures. 
So by calling blecent_scan() in blecent_on_subscribe() puts master in discovery mode for another connection. Which in turn helps master to establish connections with 3 slaves ( By default, maximum number of concurrent connections are 3 in esp-idf)
This setup of connection between 1 master and 3 slaves is working fine with blecent example.

novumdun, 2022-05-11:
Hi, now it works.
I used put the scan in the "BLE_GAP_EVENT_CONNECT" event code,
```c
    case BLE_GAP_EVENT_CONNECT:
        if (event->connect.status == 0) {
            scan();
        }
        break;
```
but now I put the scan in another task.
I think the scan shouldn't be called in  "BLE_GAP_EVENT_CONNECT" event.

IshaESP, 2022-05-11:
Hi @novumdun,
I tried putting scan() in the "BLE_GAP_EVENT_CONNECT" event code, it worked for me. The events posted in gap layer are in host context. So, initiating a new command from host in these events should work.

 
        case BLE_GAP_EVENT_CONNECT:
                if (event->connect.status == 0) {
                    MODLOG_DFLT(INFO, "Connection established..Going for scanning again ");
                    blecent_scan();
                }
         

For reference, I have attached a screenshot of the output  - [image]

rahult-github, 2022-05-23:
Hi @novumdun , 

 Can you please confirm if issue is working as mentioned in [comment](https://github.com/espressif/esp-idf/issues/8904#issuecomment-1123268432)  here ? 

Thanks,
 Rahul

novumdun, 2022-05-24:
Hi, it is working. Thanks!

closed 2022-05-24 by novumdun