espressif/esp-idf #8905: ESP32C3 Using UART1 to send data at high speed will be truncated

GKoSon, 2022-05-05:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [Y ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [Y ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [N ] I have searched the issue tracker for a similar issue and not found a similar issue.

UART is easy for use

I write a function like below
void  ESP_UART_Send(uint8_t *data,int datalen){
    uart_write_bytes(ECHO_UART_PORT_NUM, (const char *) data, datalen);
}
place in \examples\peripherals\uart\uart_echo

if i send a data packet 
when data packet is small ,send will.
because the log is a line.

when data packet is long,send will truncated the data
like this
tx："B10088007700000000004C00004C00004";

rx："B100880077000";
rx：“00000004C00004C00004”

how to send big data?

o-marshmallow, 2022-05-09:
Hi @GKoSon,

I am not sure to understand the issue: who is the receiver? Who prints the `rx:` lines? What is the UART baudrate? Which pin are you using for transmitting?

Have you considered the fact that both the receiver and the sender have small UART hardware FIFOs and so at some point the data will be truncated and you'll have to "merge" them in the software?

GKoSon, 2022-05-10:
I try 
WIN10PC+CH340   workwell
the question is Golang UART need  "merge" them in the software

thanks

closed 2022-05-10 by GKoSon