espressif/esp-idf #8906: pbuf_free LWIP_ASSERT failure and crash

JonathanWitthoeft, 2022-05-05:
## Environment

- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 3.3V]

## Problem Description

We are seeing occasional crashes from some of our customers. When this happens it appears to be when an `ACTION_RX_HANDLE` mdns action calls `pbuf_free(action->data.rx_handle.packet->pb);` and for some reason the `LWIP_ASSERT("pbuf_free: p->ref > 0", p->ref > 0);` fails.

## Debug Logs

```
panic_abort at esp-idf/components/esp_system/panic.c:365
esp_system_abort at esp-idf/components/esp_system/system_api.c:110
abort at esp-idf/components/newlib/abort.c:46
__assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)
pbuf_free at esp-idf/components/lwip/lwip/src/core/pbuf.c:757
 (inlined by) pbuf_free at esp-idf/components/lwip/lwip/src/core/pbuf.c:729
_mdns_execute_action at esp-idf/components/mdns/mdns.c:4017
 (inlined by) _mdns_service_task at esp-idf/components/mdns/mdns.c:4132

```

JonathanWitthoeft, 2022-05-05:
The part I left out is that this is a certified HomeKit accessory. I imagine HomeKit is at a higher layer using the mDNS library, but it depends on DNS/mDNS for bonjour device discovery.

JonathanWitthoeft, 2022-05-16:
@david-cermak is there anything I can do to help debug this issue? Here is a different crash in pbuf_free that has a longer backtrace:
```
0x40093725: panic_abort at .../components/esp_system/panic.c:379
0x40093f31: esp_system_abort at .../components/esp_system/system_api.c:112
0x40098d3d: abort at .../components/newlib/abort.c:46
0x400fa968: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)
0x4015617e: pbuf_free at .../components/lwip/lwip/src/core/pbuf.c:757
 (inlined by) pbuf_free at .../components/lwip/lwip/src/core/pbuf.c:729
0x4015d899: nd6_send_na at .../components/lwip/lwip/src/core/ipv6/nd6.c:1340
0x4015edc2: nd6_input at .../components/lwip/lwip/src/core/ipv6/nd6.c:549
0x4015fed1: icmp6_input at .../components/lwip/lwip/src/core/ipv6/icmp6.c:121
0x4015d25e: ip6_input at .../components/lwip/lwip/src/core/ipv6/ip6.c:1090
0x4016003d: ethernet_input at .../components/lwip/lwip/src/netif/ethernet.c:229
0x40150209: tcpip_thread_handle_msg at .../components/lwip/lwip/src/api/tcpip.c:180
 (inlined by) tcpip_thread at .../components/lwip/lwip/src/api/tcpip.c:154
 ```

david-cermak, 2022-05-16:
@JonathanWitthoeft As mentioned earlier, I don't have a better suggestion than trying to locate the memory corruption.
Could you please follow these guidelines https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-corruption-detection

Let's start with setting [CONFIG_HEAP_CORRUPTION_DETECTION](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-heap-corruption-detection) to [Comprehensive](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#comprehensive)
and see if some a new, more detailed info appears?

david-cermak, 2022-05-17:
@JonathanWitthoeft I would still suggest updating the configuration, it would still crash (but I expect earlier and perhaps with some additional clue where the corruption comes from) so you'd see the backtrace eventually reported by your customers.

> Hopefully it is a bug that has been fixed already.

I'm not aware of any recent fixes to related components (mdns, ethernet, lwip), but the crash might really disappear to appear somewhere else...

---

Back to my original question: 

> Do you init and deinit the mdns frequently?

Does you application deinitializes some components (not only mdns) or simply sets everything up and just run?

JonathanWitthoeft, 2022-05-24:
> Forgotten quote or a typo? That would explain the corruption... I'm afraid this won't compile clean, though.

Sorry I was attempting to make the code more generic and removed the quotes on accident. The quotes are indeed there in my code
```
	txtData[txtIdx].key   = "feature";
	txtData[txtIdx].value = "1";
```

I am going to do some further debugging of the local API. I will work on getting you some more code once I test a bit more.

david-cermak, 2022-07-12:
@JonathanWitthoeft Any update about this issue? Were you able to recreate it with a reduced project that you can share?

JonathanWitthoeft, 2022-07-12:
I ran out of time debugging this issue and unfortunately can not put more effort into it right now. A daily crash (with plenty of memory and no apparent heap corruption) is what we are experiencing which recovers without the customers knowledge. There is a small group of our customers that use this feature, and the crash/reboot has no noticeable impact.

david-cermak, 2022-07-15:
Could you please close this issue then? You can always reopen it when you have more details or time for debugging.

closed 2022-09-09 by Alvin1Zhang