espressif/esp-idf #8907: Low level library using FreeRTOS

mateusglucas, 2022-05-05:
## Environment

- Module or chip used:  ESP32-S2
- IDF version: v4.4
- Build System:         idf.py
- Compiler version: 8.4.0
- Operating System:     Windows
- Using an IDE?: Yes (Espressif IDE)

## Problem Description

I'm trying to enable UART interrupt in the bootloader_before_init hook using hal/interrupt_controller_ll.h functions.

### Actual Behavior

I can't use hal/interrupt_controller_ll.h functions on bootloader because these functions relies on xtensa component, which is dependent of FreeRTOS.

### Expected Behavior

Low level libraries should not be dependent of FreeRTOS library.

igrr, 2022-05-05:
Hi @mateusglucas,
We don't support custom interrupt-related functionality in the 2nd stage bootloader. If you really need to implement something like this, please take a look at esptool's flasher stub code, which enables UART interrupts: https://github.com/espressif/esptool/blob/6cf41b430c2bcc691dd607b9dbb9a31d4e4b6844/flasher_stub/stub_io.c#L79-L81. This code relies on functions provided by the on-chip ROM.
However, please note that we don't have any test cases for interrupt related features in the bootloader context, and can't guarantee that adding such functionality to the bootloader won't cause any issues.

mateusglucas, 2022-05-06:
Hi @igrr,

The code you indicated helped me a lot, thank you! By the way, it would be great to see more information in the documentation about implementing a custom bootloader.

closed 2022-05-06 by mateusglucas