espressif/esp-idf #8908: iOS crashes when trying to get prov-config protocomm endpoint

sch0bert, 2022-05-05:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used: ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.4.1-1-g093bb9882c
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:   Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

I'm not sure this is a IDF issue but I'm experimenting a very weird issue that started happening when I moved from Blueroid to NimBLE in my project. The issue happens only in iOS every time the app poles the wifi status through the characteristic prov-config. This is what I see in the serial:

[image]

As you can see, I'm able to read the other characteristics without any issue but when I try the "prov-config" I get the error.

Also, in the app I get also a message asking to pair even when I disabled the force encryption (from what I was told in this ticket: https://github.com/espressif/esp-idf/issues/8880):

[image]

When I accept to pair in the app, the app poles again "prov-config" and crashes.

### Expected Behavior

The iOS App should be able to read the characteristics without crashing and not ask to pair.

### Actual Behavior

The iOS app crashes and ask you to pair.

### Steps to reproduce

Is hard to ask you to reproduce it as is not possible with the WiFi Provisioning example as is not using protocomm 

Any hints you could give me on how to tackle this issue is welcome.

rahult-github, 2022-05-06:
Hi @sch0bert ,

May i request you to please share more details that will assist us to debug this further ?

1. Can you please share the crash backtrace you observe ?

2. Is it possible for you to enable debug logs and get a failure log ?

3. Since you have mentioned --> "Is hard to ask you to reproduce it as is not possible with the WiFi Provisioning example as is not using protocomm"  ,  Can't the same logical behaviour be replicated using the existing applications in IDF ?   Also, if protocomm is not used, can you please share how the bluetooth functionality / API details of your application ? any code for reference that can be shared ? 

Thanks,
 Rahul

sch0bert, 2022-05-06:
Hello @rahult-github ,

Here is the debug log:

[iOSCrashing.txt](https://github.com/espressif/esp-idf/files/8641525/iOSCrashing.txt)

If this is not enough, I can prepare some code example for more context (I can't share the original code)

rahult-github, 2022-05-08:
Hi @sch0bert  ,  

In logs we see that iOS has attempted multiple read / write of characteristics. However, at one step, the read on characteristic fails. This i assume is because the link is not encrypted. After this, in logs we see stack attempting to search our / peer security records. LE Rand command is sent from host and Host NOCP is then observed.  This log doesn't capture anything further.   

As the crash happening on iOS side, there is a possibility that this may be apple specific . Can you try changing the iOS version and check ? 

Since we do not know iOS implementation , you can try to capture over the air sniffer log to see what is happening and get clue as to why iOS is crashing. 

Thanks,
 Rahul

sch0bert, 2022-05-09:
Hello @rahult-github,

The iOS implementation makes use of the same ESPProvision.swift implementation inside the utility given in the ESP Provisioning library. The error happens in the next command:

[image]

And we ge the next error in iOS:

**Error: HandyManager.pollWiFiStatus(completionHandler:) (414) : Error while sending config data error: Optional(Error Domain=CBATTErrorDomain Code=5 "Authentication is insufficient." UserInfo={NSLocalizedDescription=Authentication is insufficient.})**

It gives me some questioning the last line of the command, should we stop encrypt and decrypt every message in the scenarios where I disable the force link encryption during characteristic read/write? If that the case, why is it working for the other characteristics?

sch0bert

sch0bert, 2022-05-13:
hello @rahult-github ,

Is it enough the information or do you need something else to solve the last question?

sch0bert

rahult-github, 2022-05-13:
Hi @sch0bert ,

For the query --> should we stop encrypt and decrypt every message in the scenarios where I disable the force link encryption during characteristic read/write  . 
     This can be your implementation specific. However, in general, solutions encrypt the link and do communication till the lifetime of the connections.

For the crash on iOS -->  From the given logs, esp32 side doesn't show any anomaly. Why iOS is crashing , i may not be the correct person to answer. Did you get a chance to check with different iOS version ? 

Thanks,
 Rahul

sch0bert, 2022-05-13:
Hello @rahult-github,

I disabled the force link encryption as it was causing the pop up request in the phone to share contact list which for my application this can raise some concerns for my end users. At the end this is not the final issue as having this option enabled/disabled always causes the iOS crash. So what you saw in the log, that you believe that this could be an issue of not having encrypted the link, is what makes me wonder that there should be something else missing from the FW perspective or in the framework as I already tested all configurations given by menuconfig. Is there anything else that you think I need to configure outside the framework configuration to control this link encryption?

I had before Blueroid and everything was working from the app and embedded perspective but the issue in the app happened when I moved to NimBLE. As is known, NimBLE brings the benefit of releasing memory resources that in my case is something I'm looking forward for some other implementations. 

Regarding the iPhones and iOS version I have tested it, here is the list, in all cases is crashing:

[image]