espressif/esp-idf #8911: Lower priority of device name in BLE advertisement

ztefanjo, 2022-05-06:
In our case, the manufacturing data that we use in advertisement is truncated when the device name is too long. We suggest to place the device name after the manufacturing data (and perhaps after other data as well) in order to better handle the case when all data can't fit in the advertisement. It is more likely that a truncation of the device name is acceptable.

xiewenxiang, 2022-05-27:
We have another way to configure the broadcast data, you can set the broadcast data freely.

```c
esp_err_t esp_ble_gap_config_adv_data_raw(uint8_t *raw_data, uint32_t raw_data_len);
```

ztefanjo, 2022-10-06:
esp_ble_gap_config_adv_data_raw works fine, thanks.

closed 2022-10-06 by ztefanjo