espressif/esp-idf #8912: Missing Header mcpwm_bdc_speed_control

patrickmagee, 2022-05-06:
- Development Kit:       FireBeetle V2ESP32-WROOM-32E
- Module or chip used:  ESP32-WROOM-32E
- IDF version: framework-espidf 3.40302.0 (4.3.2)
- Build System:        idf.py
- Compiler version: toolchain-xtensa-esp32 8.4.0+2021r2-patch3
- Operating System:  Windows
- Using an IDE?:  VSCode
- Power Supply:     USB 5V

## Problem Description

Attempting to build the example 
https://github.com/espressif/esp-idf/tree/a82e6e63d9/examples/peripherals/mcpwm/mcpwm_bdc_speed_control

I get an error 
src/main.cpp:14:10: fatal error: driver/pulse_cnt.h: No such file or directory

The code to reproduce is the same as in the GitRepo
I tried manually searching the installed libs but I cant see any instances of these files?
C:\Users\PMagee\.platformio\packages\framework-espidf\components

Debug Log:
```
> Executing task in folder ESPIDF_mcpwm_test: C:\Users\PMagee\.platformio\penv\Scripts\platformio.exe run <

Processing firebeetle32 (platform: espressif32; board: firebeetle32; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/firebeetle32.html
PLATFORM: Espressif 32 (4.2.0) > FireBeetle-ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40302.0 (4.3.2)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30300.0 (3.3.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - tool-openocd-esp32 2.1100.20220411 (11.0)
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\firebeetle32\src\main.o
src/main.cpp:13:10: fatal error: driver/gptimer.h: No such file or directory
 #include "driver/gptimer.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

## Other items if possible

Cmake and sdkconfig attached.

igrr, 2022-05-06:
Hi @patrickmagee,
It looks like you are trying to build an example from `master` branch of this repository with an older version of ESP-IDF (4.3.2).

Please check the examples from the same version of ESP-IDF as the one you are using locally, i.e. https://github.com/espressif/esp-idf/tree/release/v4.3/examples/peripherals/mcpwm/

patrickmagee, 2022-05-10:
Hi Igrr,

Yes you are correct, I'm using VScode Platformio and their latest available version is 4.3.2, it always lags behind I believe.
I got the functionality working with the example you linked.
Thanks very much for your assistance.

closed 2022-09-23 by Alvin1Zhang