espressif/esp-idf #8915: is:issue is:open 4.4版本esp32s3 ULP demo  build报错

aixier, 2022-05-07:
es.py --offset 0x8000 bootloader 0x0 D:/Espressif/frameworks/work/ulp_fsm/ulp/build/bootloader/bootloader.bin"
Bootloader binary size 0x4a50 bytes. 0x35b0 bytes (42%) free.
[7/914] Performing configure step for 'ulp_main'
FAILED: esp-idf/main/ulp_main-prefix/src/ulp_main-stamp/ulp_main-configure
esp-idf\main\CMakeFiles\ulp_main-configure-3c5e743.bat c36b60ba6a2bd561
-- The ASM compiler identification is unknown
-- Didn't find assembler
-- The C compiler identification is unknown
CMake Error at CMakeLists.txt:4 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
CMake Error at CMakeLists.txt:4 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "D:/Espressif/frameworks/work/ulp_fsm/ulp/build/esp-idf/main/ulp_main/CMakeFiles/CMakeOutput.log".
See also "D:/Espressif/frameworks/work/ulp_fsm/ulp/build/esp-idf/main/ulp_main/CMakeFiles/CMakeError.log".
Batch file failed at line 3 with errorcode 1
[14/914] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_fields.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Alvin1Zhang, 2022-05-07:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

igrr, 2022-05-11:
@aixier could you please run `python %IDF_PATH%/tools/idf_tools.py list` from IDF command prompt and show the output of that command? It looks like the ULP assembler is not installed in your environment.

sudeep-mohanty, 2022-05-16:
@aixier The error you are observing is due to the fact that the GNU assembler toolchain does not yet support `esp32s3` as a target. We would be adding support for `esp32s3` in the near future and once that is done, you should be able to build and run `ulp_fsm` examples on the `esp32s3`.

bhxxwx, 2022-05-23:
I have the same question, I have tryed IDF-V4.4.1, IDF-V5.0(master), none of then could build example project ulp_fsm/ulp and  example project ulp_riscv/gpio for ESP32S3 target,but in the doc(https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.1/esp32s3/api-guides/ulp.html), says should work. So, is it because the documentation is written, but the toolchain is not integrated yet?

sudeep-mohanty, 2022-07-18:
@aixier @bhxxwx ULP-FSM support is now added in v5.0 ([405ced6a8](https://github.com/espressif/esp-idf/commit/405ced6a839385c2b5e7db2ea44452969b69e943)) and v4.4 ([1bd2e3f9d](https://github.com/espressif/esp-idf/commit/1bd2e3f9dfd1155b076a23e5d0495adafffecbfa)). Apologies for the delay in reporting!

closed 2022-08-01 by sudeep-mohanty