espressif/esp-idf #8916: Can not use Structured efuse fields

linxizhixi, 2022-05-07:
## Environment

- Development Kit:      ESP32-C3,ver3
- Module or chip used:  ESp32-C3
- IDF version:          v4.3.2
- Build System:         idf.py
- Compiler version:     v8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?:        Vscode
- Power Supply:         USB

## Problem Description
Can not use `Structured efuse fields` in esp_efuse_custom_table.csv

### Expected Behavior
Generate source files according to the esp_efuse_custom_table.csv table

### Actual Behavior
Generation failed

### Steps to reproduce

1. Create project using example efuse from ESP-IDF Examples
2. Change efuse_table.csv table and the code from https://docs.espressif.com
[image]
3. use `python efuse_table_gen.py`
[image]

### Code to reproduce this issue

```
WR_DIS,                           EFUSE_BLK0,   0,    32,     Write protection
WR_DIS.RD_DIS,                    EFUSE_BLK0,   0,    1,      Write protection for RD_DIS
WR_DIS.FIELD_1,                   EFUSE_BLK0,   1,    1,      Write protection for FIELD_1
WR_DIS.FIELD_2,                   EFUSE_BLK0,   2,    4,      Write protection for FIELD_2 (includes B1 and B2)
WR_DIS.FIELD_2.B1,                EFUSE_BLK0,   2,    2,      Write protection for FIELD_2.B1
WR_DIS.FIELD_2.B2,                EFUSE_BLK0,   4,    2,      Write protection for FIELD_2.B2
WR_DIS.FIELD_3,                   EFUSE_BLK0,   5,    1,      Write protection for FIELD_3
WR_DIS.FIELD_3.ALIAS,             EFUSE_BLK0,   5,    1,      Write protection for FIELD_3 (just a alias for WR_DIS.FIELD_3)
WR_DIS.FIELD_4,                   EFUSE_BLK0,   7,    1,      Write protection for FIELD_4

```
## Debug Logs

```
PS D:\C_Cpp\esp_idf\efuse> cd C:\Users\zyt\esp\esp-idf-v4.3.2\components\efuse
PS C:\Users\zyt\esp\esp-idf-v4.3.2\components\efuse> python efuse_table_gen.py --idf_target esp32c3 D:\C_Cpp\esp_idf\blink_efuse\main\esp_efuse_custom_table.csv
Max number of bits in BLK 256
Parsing efuse CSV input file D:\C_Cpp\esp_idf\blink_efuse\main\esp_efuse_custom_table.csv ...
Verifying efuse table...
Field at WR_DIS.RD_DIS, EFUSE_BLK0, 0, 1 overlaps WR_DIS, EFUSE_BLK0, 0, 32
```

## Other items if possible

```
CONFIG_EFUSE_CUSTOM_TABLE=y
CONFIG_EFUSE_VIRTUAL=y
```

nikidimitrow, 2022-05-08:
Hi, maybe you try to run example from esp-idf **v4.4.1** (https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/system/efuse.html). Can you try the example from esp-idf **v4.3.2** (https://docs.espressif.com/projects/esp-idf/en/v4.3.2/esp32/api-reference/system/efuse.html#how-to-add-a-new-field)

KonstantinKondrashov, 2022-05-09:
Hi! @linxizhixi! This feature (`Structured efuse fields`) was introduced in idf v4.4. It was not backported to v4.3.

Commits that adds this feature:
4d35b1fc5b8a410b783a5bb4542637d3e0e48647
29f853633df38701c614e7cbcc9f4533d4a568f2

closed 2022-05-31 by Alvin1Zhang