espressif/esp-idf #8917: idf master版本编译esp32s3报错

aixier, 2022-05-08:
Adding linker script D:/Espressif/frameworks/esp-idf-v4.4/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32s3 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: D:/Espressif/frameworks/esp-idf-v4.4/components/bootloader D:/Espressif/frameworks/esp-idf-v4.4/components/bootloader_support D:/Espressif/frameworks/esp-idf-v4.4/components/efuse D:/Espressif/frameworks/esp-idf-v4.4/components/esp32s3 D:/Espressif/frameworks/esp-idf-v4.4/components/esp_common D:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support D:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom D:/Espressif/frameworks/esp-idf-v4.4/components/esp_system D:/Espressif/frameworks/esp-idf-v4.4/components/esptool_py D:/Espressif/frameworks/esp-idf-v4.4/components/freertos D:/Espressif/frameworks/esp-idf-v4.4/components/hal D:/Espressif/frameworks/esp-idf-v4.4/components/log D:/Espressif/frameworks/esp-idf-v4.4/components/bootloader/subproject/main D:/Espressif/frameworks/esp-idf-v4.4/components/bootloader/subproject/components/micro-ecc D:/Espressif/frameworks/esp-idf-v4.4/components/newlib D:/Espressif/frameworks/esp-idf-v4.4/components/partition_table D:/Espressif/frameworks/esp-idf-v4.4/components/soc D:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash D:/Espressif/frameworks/esp-idf-v4.4/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Espressif/frameworks/work/ulp_fsm/ulp/build/bootloader
[1/1] cmd.exe /C "cd /D D:\Espressif\frameworks\work\ulp_fsm\ulp\build\bootloader\esp-idf\esptool_py && D:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:/Espressif/frameworks/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 D:/Espressif/frameworks/work/ulp_fsm/ulp/build/bootloader/bootloader.bin"
Bootloader binary size 0x4a50 bytes. 0x35b0 bytes (42%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Alvin1Zhang, 2022-05-09:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

sudeep-mohanty, 2022-07-18:
@aixier ULP-FSM for `esp32s3` is now supported on on IDF-v5.0 . Please try out. Thanks!

closed 2022-08-01 by sudeep-mohanty