espressif/esp-idf #8918: Component "soc" files not found

Eran125, 2022-05-08:
Hi,
I add to the project folder called "Common" and under this folders the are my components folders.
One of the components is "Camera" and it use IDF component "soc", I add to the CmakeList file (locate in Camera folder) this row: "PRIV_REQUIRES soc" but I get an error: fatal error: soc/system_reg.h: No such file or directory #include "soc/system_reg.h"
This are my settings:
chip esp32s3
IDF 4.4
IDE VScode
folder tree: Project
                              |--- Common
                                     |--- Camera
                                     |--- CLI
                                     |--- RTC
                              |--- Main

nikidimitrow, 2022-05-08:
Hi [Eran125](https://github.com/Eran125),

from your code I got the impression that structure of the project is:

|-- <Project-name>/
    |-- Common/
        |-- <Camera>/
            |-- file1.h
            |-- file2.h
            |-- etc...

But in the example below you show that:

|--- Common
|--- Camera
|--- CLI
|--- RTC
|--- Main

Eran125, 2022-05-08:
This is part of the tree:
Project
----------|Common
---------------------|Camera
-------------------------------|conversions
-------------------------------|Driver
-------------------------------|sensors
---------------------|CLI
---------------------|RTC
----------|Main
The issue is from one of the files in the camera component.

nikidimitrow, 2022-05-08:
Hi [Eran125](https://github.com/Eran125),
in your main CMakeList.txt file do you have added:

include($ENV{IDF_PATH}/tools/cmake/project.cmake)?

Is it possible to provide some snipped of main CMake file without any project specific stuff, if it for commercial purpose.

Eran125, 2022-05-09:
Hi,
This is the project CMakeList.txt file:
cmake_minimum_required(VERSION 3.5)
list(APPEND EXTRA_COMPONENT_DIRS "Common/Camera" "Common/CLI" "Common/RTC")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(Proj)

nikidimitrow, 2022-05-09:
Hi [Eran125](https://github.com/Eran125),
can you try the following CMakeList.txt

```
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS
                                        "Common/Camera" 
                                         "Common/CLI"
                                  "Common/RTC"
)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(Proj)
```

Eran125, 2022-05-09:
Hi,
I change the list append to set as you wrote but I get the same results, same error.
../Common/Camera/driver/ll_cam.c:17:10: fatal error: soc/system_reg.h: No such file or directory #include "soc/system_reg.h"

nikidimitrow, 2022-05-09:
[Eran125](https://github.com/Eran125), how you build the project? With VS Code ESP-IDF Extension or with ESP-IDF 4.4 console?

Eran125, 2022-05-09:
With VS Code ESP-IDF Extension

nikidimitrow, 2022-05-09:
Can you try with ESP-IDF 4.4 console, please? If it work console, please take a look at this issues: https://github.com/espressif/esp-dev-kits/issues/41

Eran125, 2022-05-10:
I think this error fix (have other errors now to fix), the issue was the chip selected. I had to delete the "sdkconfig.old" and "sdkconfig.defualt" before I select the new target.
I have another question about the "sdkconfig" file, it auto generate every build - so how can I add new parameter?

nikidimitrow, 2022-05-11:
Hi,
for generation of sdkconfig file you can check documentation page https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#project-configuration-menu

Eran125, 2022-05-12:
Tank you, now the build success.

closed 2022-05-12 by Eran125