espressif/esp-idf #8923: RTL8201 - 10Mbps speed and PHY status

Simonefardella, 2022-05-09:
Hi guys, i've designed two boards with ESP32-WROOM and the ETH Phy chip RTL8201, i've received them today and flashed with the standard esp32 client using ESP32 Arduino.

I’m using the internal GPIO17 in order to generate the 50Mhz clock signal, here you have the schematic:

[image]

The sketch works, but i’m experiencing two main issues:

The speed of the RTL8201F-VB-CG is only working 10Megabits, why?
The RTL8201F-VB-CG becomes really hot without doing nothing and i don't know why this is happening!

Also, before the RTL8201 connects at 10M, this is the log that i see in arduino console and one of the two led flashes a couple of time.

[image]

Same results using the esp-idf eth example, nothing changes.

Maybe there are some registers that are not well written by ESP-IDF core classes?

Simonefardella, 2022-05-10:
@suda-morris, as you developed these eth class, you have some info?

kostaond, 2022-05-13:
Closed with no action since user reported the problem was caused by PCB issue.

closed 2022-05-13 by kostaond