espressif/esp-idf #8924: ADC DMA 输出错误

flair0626, 2022-05-10:
将例程 adc_dma_read 修改成只使用ADC1后返回信息不正确，没有通道信息，值也是不对的。
`#define TIMES 256

#define ADC_RESULT_BYTE 2
#define ADC_CONV_LIMIT_EN 0
#define ADC_CONV_MODE ADC_CONV_SINGLE_UNIT_1
#define ADC_OUTPUT_TYPE ADC_DIGI_OUTPUT_FORMAT_TYPE1

#if CONFIG_IDF_TARGET_ESP32S2
static uint16_t adc1_chan_mask = BIT(0) | BIT(1) | BIT(2);
static uint16_t adc2_chan_mask = 0;
static adc_channel_t channel[3] = {ADC1_CHANNEL_0, ADC1_CHANNEL_1, ADC1_CHANNEL_2};
#endif

static const char *TAG = "ADC DMA";

static void continuous_adc_init(uint16_t adc1_chan_mask, uint16_t adc2_chan_mask, adc_channel_t *channel, uint8_t channel_num)
{
    adc_digi_init_config_t adc_dma_config = {
        .max_store_buf_size = 1024,
        .conv_num_each_intr = TIMES,
        .adc1_chan_mask = adc1_chan_mask,
        .adc2_chan_mask = adc2_chan_mask,
    };
    ESP_ERROR_CHECK(adc_digi_initialize(&adc_dma_config));

    adc_digi_configuration_t dig_cfg = {
        .conv_limit_en = ADC_CONV_LIMIT_EN,
        .conv_limit_num = 250,
        .sample_freq_hz = 10 * 1000,
        .conv_mode = ADC_CONV_MODE,
        .format = ADC_OUTPUT_TYPE,
    };

    adc_digi_pattern_config_t adc_pattern[SOC_ADC_PATT_LEN_MAX] = {0};
    dig_cfg.pattern_num = channel_num;
    for (int i = 0; i < channel_num; i++)
    {
        adc_pattern[i].atten = ADC_ATTEN_DB_0;
        adc_pattern[i].channel = channel[i];
        adc_pattern[i].unit = ADC_UNIT_1;
        adc_pattern[i].bit_width = SOC_ADC_DIGI_MAX_BITWIDTH;
    }
    dig_cfg.adc_pattern = adc_pattern;
    ESP_ERROR_CHECK(adc_digi_controller_configure(&dig_cfg));
}

#if !CONFIG_IDF_TARGET_ESP32
static bool check_valid_data(const adc_digi_output_data_t *data)
{
    const unsigned int unit = data->type2.unit;
    if (unit > 2)
        return false;
    if (data->type2.channel >= SOC_ADC_CHANNEL_NUM(unit))
        return false;

    return true;
}
#endif

void app_main(void)
{
    esp_err_t ret;
    uint32_t ret_num = 0;
    uint8_t result[TIMES] = {0};
    memset(result, 0xcc, TIMES);

    continuous_adc_init(adc1_chan_mask, adc2_chan_mask, channel, sizeof(channel) / sizeof(adc_channel_t));
    adc_digi_start();

[... 130 lines cut ...]
I (292) ADC DMA: C:Unit: 1, Channel: 0, Value: 984
I (302) ADC DMA: C:Unit: 1, Channel: 0, Value: 982
I (302) ADC DMA: C:Unit: 1, Channel: 0, Value: 982
I (312) ADC DMA: C:Unit: 1, Channel: 0, Value: 982
I (312) ADC DMA: C:Unit: 1, Channel: 0, Value: 982
I (322) ADC DMA: C:Unit: 1, Channel: 0, Value: 980
I (332) ADC DMA: C:Unit: 1, Channel: 0, Value: 982
I (332) ADC DMA: C:Unit: 1, Channel: 0, Value: 980
I (342) ADC DMA: C:Unit: 1, Channel: 0, Value: 97e
I (342) ADC DMA: C:Unit: 1, Channel: 0, Value: 97e
I (352) ADC DMA: C:Unit: 1, Channel: 0, Value: 97e
I (352) ADC DMA: C:Unit: 1, Channel: 0, Value: 97e
I (362) ADC DMA: C:Unit: 1, Channel: 0, Value: 97c
I (362) ADC DMA: C:Unit: 1, Channel: 0, Value: 97c
I (372) ADC DMA: C:Unit: 1, Channel: 0, Value: 97c
I (372) ADC DMA: C:Unit: 1, Channel: 0, Value: 97a
I (382) ADC DMA: C:Unit: 1, Channel: 0, Value: 97a
I (392) ADC DMA: C:Unit: 1, Channel: 0, Value: 978
I (392) ADC DMA: C:Unit: 1, Channel: 0, Value: 976
I (402) ADC DMA: C:Unit: 1, Channel: 0, Value: 976
I (402) ADC DMA: C:Unit: 1, Channel: 0, Value: 976
I (412) ADC DMA: C:Unit: 1, Channel: 0, Value: 976
I (412) ADC DMA: C:Unit: 1, Channel: 0, Value: 974
I (422) ADC DMA: C:Unit: 1, Channel: 0, Value: 972
I (422) ADC DMA: C:Unit: 1, Channel: 0, Value: 972`

WangShuoran, 2022-05-10:
你的IDF版本是多少？
看看这篇资料？https://www.bilibili.com/video/BV1fq4y1a7o4

flair0626, 2022-05-17:
> 你的IDF版本是多少？ 看看这篇资料？https://www.bilibili.com/video/BV1fq4y1a7o4

ESP-IDF v4.4.1 应该还是IDF的bug