espressif/esp-idf #8925: 当我打开编译优化

WangShuoran, 2022-05-10:
## Environment
- Development Kit: ESP32-C3,ver3
- Module or chip used: ESp32-C3
- IDF version: v4.4.1
- Build System: idf.py
- Compiler version: v8.4.0
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Vscode
- Power Supply: USB

## Problem Description
我想要进行编译优化，但是一旦设置编译优化就有报错。

### Expected Behavior
在ESP32C3的v4.3.2是编译正确的，但我想在使用更高的IDF版本以更好的使用ESP32C3的性能。但是其中我设置的编译优化会导致我的项目编译失败：
```cmake
# 编译优化等级
# idf_build_set_property(COMPILE_OPTIONS "-O3" APPEND)
```
### Actual Behavior
```cmake
CMakeFiles\__idf_nvs_flash.dir\src\nvs_storage.cpp.obj.d -o esp-idf/nvs_flash/CMakeFiles/__idf_nvs_flash.dir/src/nvs_storage.cpp.obj -c C:/Espressif/frameworks/esp-idf-v4.4.1/components/nvs_flash/src/nvs_storage.cpp
In member function 'void nvs::Storage::fillEntryInfo(nvs::Item&, nvs_entry_info_t&)',
    inlined from 'bool nvs::Storage::nextEntry(nvs_opaque_iterator_t*)' at C:/Espressif/frameworks/esp-idf-v4.4.1/components/nvs_flash/src/nvs_storage.cpp:780:30:    
C:/Espressif/frameworks/esp-idf-v4.4.1/components/nvs_flash/src/nvs_storage.cpp:734:20: error: 'char* strncpy(char*, const char*, size_t)' output may be truncated copying 15 bytes from a string of length 15 [-Werror=stringop-truncation]
             strncpy(info.namespace_name, name.mName, sizeof(info.namespace_name) - 1);
             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors
[84/995] Building CXX object esp-idf/nvs...sh.dir/src/nvs_partition_manager.cpp.obj
ninja: build stopped: subcommand failed.
```
### Steps to reproduce
当我在项目中的CMakeList.txt加入
```cmake
# 编译优化等级
# idf_build_set_property(COMPILE_OPTIONS "-O3" APPEND)
```
将会出现上述Actual Behavior问题。

[image]

### Code to reproduce this issue
随便拿nvs相关的例程就会报错，这里使用的例程是nvs_rw_value

Huckies, 2022-05-10:
你设置-1把末尾的'\0'给截了啊，这个警告就是在提示你

不想看到的话加个``-Wno-stringop-truncation``掐了就是

WangShuoran, 2022-05-11:
@Huckies 你好，谢谢回答我的问题，报错的这个文件并不是我写的，是官方的IDF组件库中的内容。我认为你随便用一个nvs例程，加入编译优化，都会报这个错。

Huckies, 2022-05-11:
``cc1plus.exe: some warnings being treated as errors``只是一个警告而已，知道代码在干什么就可以无视

WangShuoran, 2022-05-11:
你好，感谢回复，看来我打开编译优化，会把警告视作错误处理，那么我如果忽略它的警告？
我当时试了在项目CMakeLists.txt中加入：
```cmake
idf_build_set_property(COMPILE_OPTIONS "-Wno-stringop-truncation" APPEND)
```
但是依然报错，请问我加入这条内容和`idf_build_set_property(COMPILE_OPTIONS "-O3" APPEND)`的前后顺序有关系吗？

Huckies, 2022-05-11:
请贴出完整的``CmakeLists.txt``

WangShuoran, 2022-05-11:
你好，感谢回复。我这里在你说我以前尝试过的加入`idf_build_set_property(COMPILE_OPTIONS "-Wno-stringop-truncation" APPEND)`时，我尝试把这句话提前到编译优化前面，现在编译成功了。
附成功的CMakeLists.txt：
```cmake
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(xxx)

# 忽略编译优化导致的报错
idf_build_set_property(COMPILE_OPTIONS "-Wno-stringop-truncation" APPEND)

# 编译优化等级
idf_build_set_property(COMPILE_OPTIONS "-O3" APPEND)

```
谢谢您，给我的学习之路上了一课。没有您的帮助，我的效率会非常慢，祝您每天都有美好的心情！

closed 2022-05-11 by WangShuoran