espressif/esp-idf #8926: 新的PCB无法下载程序

K-G-1, 2022-05-10:
## Environment

- Development Kit:      自制PCB
- Module or chip used:  [ESP32C3]
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:   CMake      [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: 使用VScode进行环境搭建
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
自己制作了一块PCB，焊接完成之后无法下载，使用`monitor`指令查看，串口有信息输出，显示是USB_BOOT，但是使用`flash`指令无法下载，使用的是`power_save`例程，请问如何才能切换到串口Boot，并正常下载？

[image]

以下是部分原理图截屏
[image]

[image]

[image]

### Expected Behavior
正常下载
### Actual Behavior
无法下载

## Debug Logs

以下是正常复位之后的log输出
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c0dc
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

```
以下是下载模式的log输出
```
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xfffff�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x0 (USB_BOOT)
wait usb download
```

Huckies, 2022-05-12:
把vs code状态栏上的``UART``点开换成``JTAG``试试？

Huckies, 2022-05-12:
你开flash加密了吗？

K-G-1, 2022-05-12:
> 你开flash加密了吗？

谢谢帮助，这个问题解决了，是PCB上GPIO8需要设置默认上拉，才可以使用串口下载

closed 2022-05-12 by K-G-1