espressif/esp-idf #8927: IDF compile error

mike-2020, 2022-05-10:
When the following option is selected:

[ ] Disable messages in ESP_RETURN_ON_* and ESP_EXIT_ON_* macros

Got compile error below:

_[ 57%] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborencoder_close_container_checked.c.obj
[ 57%] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
~/SDKs/esp-idf-v4.4/components/esp_adc_cal/esp_adc_cal_esp32.c:86:19: error: 'LOG_TAG' defined but not used [-Werror=unused-const-variable=]
 const static char LOG_TAG[] = "ADC_CALI";
                   ^~~~~~~
[ 57%] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborencoder.c.obj
[ 57%] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/esp_camera.c.obj
cc1: some warnings being treated as errors
make[2]: *** [esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/build.make:63: esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/esp_adc_cal_esp32.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:5107: esp-idf/esp_adc_cal/CMakeFiles/__idf_esp_adc_cal.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs...._

Alvin1Zhang, 2022-05-10:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

FurballTheGreat, 2022-10-25:
I can confirm that this is still an issue. Recent changes to the v4.4 branches have made [ ] Disable messages in ESP_RETURN_ON_* and ESP_EXIT_ON_* macros cause the error described.