espressif/esp-idf #8930: Changing IDF version downwards ends in invalid platform

ddomnik, 2022-05-10:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version v4
- Module or chip used: ESP32-WROOM-32D
- IDF version v5.0-dev-2586-ga82e6e63d -> v4.4
- Build System: CMake|idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes - Universal Online Installer 2.15
- Power Supply: USB

## Problem Description

After facing build problems with version v5.0-dev-2586-ga82e6e63d #8929, I chose to switch back to v4.4 via the Espressif -> Download and Configure ESP-IDF -> Use an existing ESPD-IDF ... dialog. IDF_PATH gets configured.
After pressing "Install Tools" in the ESP-IDF Tools installing dialog the error below occours.

Deleting the .meta folder in a workspace did not resolve the issue.
Running the Online Installer again over the existing installation did not resolve the issue.

### Expected Behavior

Moving a version backwards should not break the selected platform.

### Actual Behavior

Installing tools ends in macos-arm64 despite beeing on a Windows machine.

### Steps to reproduce

1. Install IDF 4.4 via Espressif IDE 2.4.
2. Install IDF 5.0 over v4.4 via Online Installer 2.15
3. Chose framework 4.4 via Espressif -> Download and Configure ESP-IDF
4. Install tools

// If possible, attach a picture of your setup/wiring here.

## Debug Logs

```
Installing tools...
This can take a while. Please be patient.
Executing C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py install all
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1824, in main
    action_func(args)
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1415, in action_install
    tools_info = load_tools_info()
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 945, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 963, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 875, in from_json
    raise RuntimeError('invalid platform %s for tool %s version %s' %
RuntimeError: invalid platform macos-arm64 for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0

Executing C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py install-python-env
Python 3.8.7
pip 20.3.3 from C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip (python 3.8)

Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1824, in main
    action_func(args)
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1540, in action_install_python_env
    wheels_dir = get_wheels_dir()
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1472, in get_wheels_dir
    tools_info = load_tools_info()
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 945, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 963, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 875, in from_json
    raise RuntimeError('invalid platform %s for tool %s version %s' %
RuntimeError: invalid platform macos-arm64 for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0

Executing C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1824, in main
    action_func(args)
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 1174, in action_export
    tools_info = load_tools_info()
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 945, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 963, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py", line 875, in from_json
    raise RuntimeError('invalid platform %s for tool %s version %s' %
RuntimeError: invalid platform macos-arm64 for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

```

dobairoland, 2022-05-10:
`macos-arm64` support was added for v4.4 only later and the change is not yet on Github. I think this issue would go away after
6ae22d9f3b2f03d970b118d8b3f42739828baeb7 arrives here.

dobairoland, 2022-05-10:
Oh, it seems it is already here. I'm not sure which version is accessible via Espressif IDE 2.4.

cc @kolipakakondal PTAL. I don't know if there is a workaround for this.

kolipakakondal, 2022-05-10:
Hi @dobairoland Seems to be something going wrong here!

> C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_tools.py install all

The above command is trying to get `macos-arm64` toolchains when the user is on a windows machine. Do you think tools json is corrupted? I need to check further on this.

kolipakakondal, 2022-05-10:
Hi @ddomnik 

Could you verify IDF_PATH and see to which version of IDF it's pointing and change the path to IDF 4.4. Then you could  invoke Install Tools from eclipse and see if that helps

dobairoland, 2022-05-10:
> The above command is trying to get macos-arm64 toolchains when the user is on a windows machine. Do you think tools json is corrupted?

I think an older `idf_tools.py` is used with a newer `tools.json` file -  it doesn't know `macos-arm64`. I don't think it actually tries to install a MAC tool on Windows. It just cannot parse the JSON file.

kolipakakondal, 2022-05-10:
Ah, got it. Maybe IDE is overriding esp-idf directory. I will check it.

alirana01, 2022-09-12:
@kolipakakondal I tried investigating it seems like when you install over a directory already containing the Espressif IDE and launch it again it still points to the old version until you switch to a new workspace

closed 2023-02-01 by dobairoland