espressif/esp-idf #8931: Extraneous SDA transition during i2c stop sequence

mbratch, 2022-05-10:
## Environment

- Development Kit:   ESP32-PICO-Kit|
- Kit version 4
- Module or chip used:  ESP32-PICO-D4
- IDF version: v4.4
- Build System:   idf.py
- Compiler version: 8.4.0
- Operating System:  Windows 11
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (Eclipse)
- Power Supply:   External 5V

## Problem Description

I am running ESP-IDF v4.4.1 and using i2c_master_write_to_device to write a byte to an I2C device.

I'm seeing an odd pulse on the SDA line, I think from "master" (the ESP32) between the final ACK of the write and the stop sequence. The annotated trace below:

[Scope trace](https://ibb.co/X49T0nb)

I would not expect SDA to go high after the last ACK. After the ACK, SDA should be held low into the stop sequence. Otherwise, it looks like it might be attempting a Repeated Start, but "aborts" it before the next clock pulse. I'm not seeing any functional failures in my device, so it doesn't seem to be causing any harm. I was just surprised to see it.

### Code to reproduce this issue

```cpp
esp_err_t rc = i2c_master_write_to_device(_i2cPort, address, &value, 1, pdMS_TO_TICKS(10));
```

mbratch, 2022-05-10:
@o-marshmallow  thanks for confirming the observation. I'm aware of the I2C spec indicating that SDA can change while SCL is low, so that's fine. I don't know if there's a time limit to when the stop must be issued, but this particular signal is very short (1uS roughly) so we shouldn't be in any peril there.

Really, I brought up this issue as an anomaly (I hadn't seen it in other I2C traces) and wanted to make sure it was something expected of the ESP32. I agree it does seem harmless if it always remains within the time constraints of the I2C spec.

o-marshmallow, 2022-05-10:
@mbratch From what I know, there is no time limit for issuing a STOP condition after an (N)ACK, it's a bit tricky to test, but in theory, the SCL line can be kept low for as long as you want. In fact, even a device could do this with clock stretching, le's say for example ot latch the writes changes internally.

I just tested at a higher frequency to be sure, and the artefact is still here but for much less time (360ns), so I guess we can say it's an expected and controlled artefact :smile:

mbratch, 2022-05-10:
@o-marshmallow thanks, again, for confirming. I'll close this since it's not really an issue.

closed 2022-05-10 by mbratch

o-marshmallow, 2022-05-11:
@mbratch You are welcome