espressif/esp-idf #8932: UARTx get free space in tx_buffer_size (uart_driver_install)

imwhocodes, 2022-05-10:
New version of [arduino-esp32](https://github.com/espressif/arduino-esp32) add function to set `tx_buffer_size` that is used by [uart_driver_install](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv419uart_driver_install11uart_port_tiiiP13QueueHandle_ti)

This add the ability to add arbitrary buffer that is then copied on _TxFiFo_ of the UART when free

The problem now become that inside esp-idf there is no function to get free space left on the TX Buffer allocated internally by `uart_driver_install(...)` 

Currently Arduino implementation check UART registers get free space left in TxFiFo, but of course the value stored in the register keep count only of the space free in TxFiFo and not the extra RAM allocated buffer by `uart_driver_install(...)` 

Is possible to have a function similar to [uart_get_buffered_data_len](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv426uart_get_buffered_data_len11uart_port_tP6size_t) that return data pending for write? or space left maybe?

Or there is any other way to extract that information from esp-idf?

Thanks,
Luca

imwhocodes, 2022-05-17:
Hi

Any hints on how to get that informations in other ways?

Thanks,
Luca

AxelLin, 2022-05-18:
https://github.com/espressif/esp-idf/issues/3078

ginkgm, 2022-05-18:
Hi @imwhocodes ,

Good idea, we will add this.

imwhocodes, 2022-06-06:
Hi

Is there any ETA for this feature?
I thought you already have the `used buffer qty` somewhere internally to the API, is it only matter to expose a `get function` to the user side?

Thanks,
Luca

closed 2022-07-07 by espressif-bot