espressif/esp-idf #8933: Some way to persist BLE bonding information for re-connection after power-off

Huckies, 2022-05-10:
Hi there, I've tried both the Bluedroid and the NimBLE host stack implementing a 'client' role device.

I watched through the examples and found them all scan public advertising every time the module is booted. What I want is more like how a computer or a smartphone treat with a HOGP mouse: there are two modes: A & B. In mode A, the module scans all the advertising around and user may pick one device to connect. While in mode B the module discovers only the expected advertising and ignores the others.

It's known to me that the Bluedroid stores bonding information in ``bt_config.conf`` and reads it back during initializing, but it seems to be shut down in ESP-IDF, can we have it back or is there some similar way to achieve the same purpose? Thanks in advance.

IshaESP, 2023-01-11:
Hi @Huckies ,
Bluedroid has bt_config,conf file. It uses a file system. In ESP-IDF we store the bonding information in NVS partition. 

So application can do below:
1. Read nvs on reboot. 
2. Attempt to connect to the devices read from nvs.
3. In nvs, we store only BD address and the LTK(Long Term Key - generated and stored when device is bonded for the first time).

Huckies, 2023-01-11:
@IshaESP Thank you for the advice.

closed 2023-01-11 by Huckies