espressif/esp-idf #8936: Task stack analysis during -fstack-usage and runtime differs by >1000 bytes

michael-sayapin, 2022-05-11:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         idf.py (ninja)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux
- Power Supply:         external 5V

## Problem Description

Hi,

I'm trying to build some kind of static analyzer for required task stack sizes. We are using ESP-MESH and Bluetooth Classic (SPP), and currently counting every byte. 

The tasks we run are pretty complicated and experimenting with "give it a lot and tune using uxTaskGetStackHighWaterMark" is not very productive since finding the worst callpath in running firmware is non-trivial.

So I've added stack and call-analyzing options to GCC:

```
idf_build_set_property(COMPILE_OPTIONS "-fstack-usage" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-fdump-ipa-cgraph" APPEND)
```

With this, plus a little tool based on https://github.com/sharkfox/stack-usage I can generate "worst case" stack profiles for every task. (Of course, the call stack does not support indirect calls or ROM functions.)

However, I cannot explain the behavior when adding a single task that does nothing:

```cpp
static void test_task(void *param) {
  int stack = *(int *)param;
  for (;;) {
    int watermark = uxTaskGetStackHighWaterMark(NULL);
    printf("test task free stack memory: %d/%d (occupied %d)", watermark,
           stack, stack - watermark);
    vTaskDelay(pdMS_TO_TICKS(1000));
  }
}

...

void app_main() {
  int stack = 0;
  stack = 8000;
  xTaskCreate(test_task, "test_task", stack, &stack, 6, NULL);
  stack = 6000;
  xTaskCreate(test_task, "test_task", stack, &stack, 6, NULL);
  stack = 5000;
  xTaskCreate(test_task, "test_task", stack, &stack, 6, NULL);
  stack = 3000;
  xTaskCreate(test_task, "test_task", stack, &stack, 6, NULL);
  stack = 2000;
  xTaskCreate(test_task, "test_task", stack, &stack, 6, NULL);
}
```

The first print I get is more or less correct (I guess it is the watermark BEFORE calling printf):

```
test task free stack memory: 7636/8000 (occupied 364)
test task free stack memory: 5700/6000 (occupied 300)
test task free stack memory: 4708/5000 (occupied 292)
test task free stack memory: 2700/3000 (occupied 300)
test task free stack memory: 1700/2000 (occupied 300)
```

However the second and subsequent prints are consuming much more stack:

```
test task free stack memory: 6964/8000 (occupied 1036)
test task free stack memory: 4820/6000 (occupied 1180)
test task free stack memory: 3940/5000 (occupied 1060)
test task free stack memory: 1964/3000 (occupied 1036)
test task free stack memory: 964/2000 (occupied 1036)
```

The .su files excerpt for the relevant methods:

```
main.c:33:13:test_task  48  static
main.c:62:6:app_main  64  static
log.c:187:6:esp_log_writev  48  static
log.c:204:6:esp_log_write 80  static
log_freertos.c:30:6:esp_log_impl_lock 32  static
log_freertos.c:41:6:esp_log_impl_lock_timeout 32  static
log_freertos.c:52:6:esp_log_impl_unlock 32  static
log_freertos.c:60:7:esp_log_system_timestamp  96  static
log_freertos.c:100:10:esp_log_timestamp 32  static
```

I've tried to override `log` component and use https://github.com/mickjc750/prnf (with `-Wno-error=format` and `-Wno-error=format-security`) only to find stack requirements even higher.

I understand that tasks store their TCBs and so on, but it's unlikely to occupy almost 1KB.

What am I missing?..

### Expected Behavior

Tasks stacks occupy a predictable amount of memory.

### Actual Behavior

Over 1KB is missing.

negativekelvin, 2022-05-11:
3 arg vprintf consuming ~700bytes is not surprising

igrr, 2022-05-11:
@michael-sayapin could you please explain this part of output of your tool?
```
main.c:62:6:app_main  64  static
log.c:187:6:esp_log_writev  48  static
log.c:204:6:esp_log_write 80  static
```
The code snippet you show has `printf` but this output implies that ESP_LOG is somehow involved? Did you run the tool for a code snippet with ESP_LOG and change the snippet to `printf` later?

In any case, `printf` (or `vprintf`, as @negativekelvin points out) is the likely culprit for you analysis. It is challenging to accurately find its stack usage for two reasons:

1. Newlib is compiled into a static library and distributed as part of the toolchain, so your tool doesn't "see" its source code.
2. Execution of vprintf involves numerous indirect calls via function pointers. So even if the source code was available to your analyzer you would still need to give it a lot of hints to understand which functions are likely called by certain function pointers.

If the purpose of your test was to check the stack usage of a simple task, move printf/logging to the main task, and pass the measured stack size value from the worker task via a variable or a queue.

michael-sayapin, 2022-05-11:
Hi @igrr ,

This is the output of `-fstack-usage` GCC compile option, I think it estimates stack usage given all the static declarations at compile time. Then you can use `-fdump-ipa-cgraph` and build full call graph (for ESP-IDF it's a huge 2GB JSON file) and recurse to find out "worst" call paths to estimate maximum stack usage.

Like you said, `vprintf` is called indirectly, so it's impossible to consider it with this tool. I've also tried changing `ESP_LOGx` call to `printf` call, it didn't seem to increase the estimated stack, but probably because it also goes into indirection when calling actual functions from ROM.

What's interesting is that I tried replacing it with https://github.com/mickjc750/prnf and ended up with even worse results (watermark ~1360). I also tried calling `prnf` directly from the `test_task` to figure out the stack usage, which will not involve any more indirect calls. It estimated `test_task` at 240 bytes, but the watermark was, again, higher.

I will try passing the watermark using queue to isolate this, most likely you are right. I guess the question would be then, how would one approach estimating vprintf's stack requirements?

michael-sayapin, 2022-05-11:
@igrr You were right. I tried isolating printf into another task. The producer task was estimated by static analysis to take 512 bytes. That was exactly what it took :party: The printf-ing task estimated at 384 bytes and took 1152.

igrr, 2022-05-11:
Good to know the estimate now agrees with the reality!

> I guess the question would be then, how would one approach estimating vprintf's stack requirements?

There's no simple and reliable solution I can think of, unfortunately (and largely the reason why we don't have stack size analysis in IDF out of the box.)

The closest practical solution I may suggest is that you annotate some functions with their indirect calls destinations or known measured stack usage.
For example, somewhere in your stack analysis tool you can keep the list:

```
# calls via a function pointer
esp_log_write: vprintf
...

# measured stack usage
vprintf: 1000
esp_wifi_start: 200
...
```

That is obviously error prone, though — it's easy to miss some particular call via a function pointer. But you might still get some success with this approach, at the cost of some manual work.

michael-sayapin, 2022-05-12:
@igrr , might you be interested in compiling closed-source and / or static libs with `-fstack-usage -fdump-ipa-cgraph` and publishing the resulting .su and .cgraph files? That will allow more precise stack calculation. I can also share a tool to calculate cumulative stacks in case .cgraph files expose too much.

igrr, 2022-05-12:
The second of the two problems I mentioned is still valid though, even if all the code is compiled with stack usage and call graph information.

For example, newlib will use function pointers in FILE structure to determine the internal 'read' and 'write' functions to execute for this file. I am not familiar with the code of the closed source part, but I do know that some function pointer tables are used there as well, to dispatch calls to OS functions and to the ROM functions.

michael-sayapin, 2022-05-12:
It makes much smaller work for "hardcoded estimated call table" approach though. Those calls can be measured using the watermark. Right now we're talking about tens of thousands of calls to libs in https://github.com/espressif/esp32-wifi-lib/tree/master/esp32 - if they had .su and .cpath files together, we would need a few hundred, perhaps.

michael-sayapin, 2022-05-13:
@igrr We've been able to instrument the stack-usage tool with additional call graph (just a little YML file enumerating additional callers and callees), so we are able to estimate worst case stack down to the static lib calls. If you can provide .su and .cgraph files for precompiled libs, this should allow for a pretty accurate estimation. I hope this will benefit the community, since right now everybody just "guesstimate" it and keep increasing the stacks if stack overflow is detected, sometimes in production... :-)

PS: I've tried to bypass by objdump-ing the libs, but xtensa objdump hides `sp` so I'm not sure which register is the stack pointer.

michael-sayapin, 2022-05-13:
Another update.

I've been able to pull call stacks from `objdump -d` for precompiled libs. With Xtensa assembly it's always the first instruction that looks like this:

```
000000dc <esp_mesh_set_config>:
  dc:	008136        	entry	a1, 64
```

Meaning `esp_mesh_set_config` needs 64 bytes on stack.

Call graphs are a bit more tricky. I dumped all sections via `readelf -a`, and I can see some calls:

```
Relocation section '.rela.text.esp_mesh_set_config' at offset 0x11f4c contains 212 entries:
 Offset     Info    Type                Sym. Value  Symbol's Name + Addend
00000064  00012a01 R_XTENSA_32            00000000   g_is_standalone_sta + 0
00000060  00008301 R_XTENSA_32            00000000   .data.s_mesh_ie_crypto_plain_key + 0
... cut for brevity ...
00000068  00015c01 R_XTENSA_32            0000002c   esp_mesh_set_router + 0
... cut for brevity ...
0000006c  00017201 R_XTENSA_32            00000000   esp_mesh_channel_enable_jp + 0
... etc
```

I think I can construct call paths from this, but it's quite a bit of work.

SoucheSouche, 2023-02-24:
Closing this issue due to low priority and time of last update.
Feel free to reopen if necessary.

closed 2023-02-24 by SoucheSouche