espressif/esp-idf #8941: Repeated reboot observed on ESP32 while implementing pppos_client example with SIM800L

JimitD, 2022-05-13:
----------------------------- Delete below -----------------------------

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    crosstool-NG esp-2021r2-patch3 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes (Visual Studio Code)]
- Power Supply:         [USB|external 5V]

## Problem Description

We have connected ESP32-WROOM-32 Dev Kit to SIM800L module with active SIM card connection

Menu-config: Modem Device: SIM800
TXD Pin: 17, RXD Pin: 16, 
Set Access Point Name: www

Now, we are trying to use built in example of PPPOS_CLIENT to establish PPPOS connection with SIM800L Modem. 

As soon as we run the Application, it rebooting multiple time with following error code being thrown 

### Expected Behavior

It should have initialized modem and create PPPOS connection. 

### Actual Behavior

Throwing error 

### Steps to reproduce

1. Connect ESP32-WROOM-32 Dev Kit to SIM800L Modem Module with 
2. Try to run example project PPPOS_CLIENT
3.  Define Menu-config as described avove
4. Build the project and run 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
 while (1) {
        modem_dce_t *dce = NULL;
        /* create dce object */
#if CONFIG_EXAMPLE_MODEM_DEVICE_SIM800
        dce = sim800_init(dte);
#elif CONFIG_EXAMPLE_MODEM_DEVICE_BG96
        dce = bg96_init(dte);
#elif CONFIG_EXAMPLE_MODEM_DEVICE_SIM7600
        dce = sim7600_init(dte);
#else
#error "Unsupported DCE"
#endif
        assert(dce != NULL);

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
E (10459) esp-modem: esp_modem_dte_send_cmd(256): process command timeout
E (10459) dce_service: esp_modem_dce_sync(225): send command failed
E (10459) sim800: sim800_init(158): sync failed

assert failed: app_main pppos_client_main.c:270 (dce != NULL)

Backtrace:0x40081a5a:0x3ffb68200x40086111:0x3ffb6840 0x4008b8e1:0x3ffb6860 0x400d75bd:0x3ffb6980 0x40134b91:0x3ffb6ac0 0x4008928d:0x3ffb6ae0
0x40081a5a: panic_abort at C:/esp_env/esp/esp-idf/components/esp_system/panic.c:402

0x40086111: esp_system_abort at C:/esp_env/esp/esp-idf/components/esp_system/esp_system.c:128

0x4008b8e1: __assert_func at C:/esp_env/esp/esp-idf/components/newlib/assert.c:85

0x400d75bd: app_main at D:\Working\ESPProject\pppos_client\build/../main/pppos_client_main.c:270 (discriminator 1)

0x40134b91: main_task at C:/esp_env/esp/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x4008928d: vPortTaskWrapper at C:/esp_env/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/8687252/sdkconfig.zip)

david-cermak, 2022-05-17:
Hi @JimitD 

This line
```
E (10459) sim800: sim800_init(158): sync failed
```
says that the synchronization between your device and the ESP32 failed. This is the initial procedure, where the ESP32 sends `AT` command over UART and expects `OK`. I'm afraid there's not much to look at, other than checking the wires and making sure the command arrives to the device and it responds. 
Additionally, you can check:
* connecting your modem directly to your PC and send `AT\r\n` in some terminal App (e.g. Putty)
* checking the commands and replies with a logic analyzer, making sure that the Tx/Rx pins are not used for any other connection
* running some console application on ESP32, e.g. https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console to send commands and check replies

JimitD, 2022-05-18:
Hello @david-cermak, 

Thank you for the reply & suggestions. 

The actual problem was physical connectivity of SIM800 Modem to ESP32. The AT command was failing due to my ESP32 was powered from some other source and Modem powered through other source. 

I have connected common ground to both the Module and now it is working fine. 

Thanks again.

closed 2022-05-18 by JimitD