espressif/esp-idf #8942: How BLE Mesh loosing provisioning data after power reset

gsapinto, 2022-05-13:
Hi there! I'm using a Sensor Client - Sensor Server configuration, but making the provisioning through through the nRF Mesh Android App, I've noticed that if there is a power reset (for example when starting a monitor on the IDF or simply pulling the power plug), all of the provisioning data is lost. I've checked a similar Issue and the recommendation was to enable an option on the menuconfig:

`Component config -> BLE Mesh Support -> Store BLE Mesh Node configuration persistently`

I've done this, but the problem persists, curious thing is the app seems to retain the App Key and Provisioning but if I monitor the device it has the provisioning open:
`SENSOR_SERVER: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0` 
And it's not running my programm until I re-provision the node. 
I'm attaching my sdkconfig file, does anyone know what I'm missing? 

Thank you and best regards
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8689109/sdkconfig.txt)

InfiniteYuan, 2022-06-07:
```
CONFIG_BLE_MESH_SPECIFIC_PARTITION=y
CONFIG_BLE_MESH_PARTITION_NAME="ble_mesh"
```

Did the project's partition table has "ble_mesh" nvs data partition?

jtmullen, 2023-06-11:
@gsapinto I have the same issue where the esp seems to have the data but doesn't run the application code until being re-provisioned. Did you find a resolution to this?

arjndr, 2024-02-05:
having the same issue, any updates or workarounds?

kunj2308, 2025-01-01:
if anyone solved above issue please , share the project of file for the same , in addition if anyone have tutorial about that configuration please share that also, thanks ..

forx157, 2025-01-02:
Hi, 
May I ask if you can share the info-level logs from your devices with me?

kunj2308, 2025-01-02:
> Hi, May I ask if you can share the info-level logs from your devices with me?

Hello ,thanks for concern but my issue is solved by enabling this option in  configuration menu in ESP-IDF, which shown in attached image 

[image]

Alvin1Zhang, 2025-03-31:
Thanks for reporting, feel free to reopen if the issue still happens.

closed 2025-03-31 by Alvin1Zhang

forx157, 2025-04-08:
Hi， @wangtong5023 
You can clear the information in flash with the command `idf.py erase-flash -p port`.
Or use `esp_ble_mesh_deinit` with `erase_flash` in the deinit param set to true.
Once cleared it is ready to be provisioning again.

wangtong5023, 2025-04-08:
> Hi， [@wangtong5023](https://github.com/wangtong5023) You can clear the information in flash with the command `idf.py erase-flash -p port`. Or use `esp_ble_mesh_deinit` with `erase_flash` in the deinit param set to true. Once cleared it is ready to be provisioning again.

Thanks for help. I'm trying, but the boards not work. If I reset client board, they don't connect auto.

wangtong5023, 2025-04-08:
Could you tell me which example program you used? 
I want to keep BLE Mesh from loosing provisioning data after power reset.

forx157, 2025-04-08:
@wangtong5023 

Can you share the sdkconfig and log (Mesh log info level) you are using?

wangtong5023, 2025-04-08:
I used idf V5.3.2, and BLE Mesh examples are vendor client and vendor server.
The log is blow: 

[image]

[image]

When I reset the client board after the network is set up, I need to reset the server boards at the same time to mesh again. How can I reset the client after power off and auto provision ?

wangtong5023, 2025-04-14:
@forx157 Thanks for your help. I've found the problem. I use serial port software, I don't know why it always resets ESP32, I thought there was something wrong with my program.