espressif/esp-idf #8943: Eclipse ESP-IDF on Mac: serial monitor is not connecting

Paveldeg, 2022-05-13:
## Environment

- Development Kit:      ESP32-PICO-DeVKitM-2
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Module or chip used:  ESP32-PICO-MINI-02
- IDF version (run ``git describe --tags`` to find it):
    // v4.4.1-dirty
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     macOS
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes Eclipse
- Power Supply:       USB

## Problem Description
When ESP32 board is connected to a USB port ESP-IDF serial monitor in Eclipse IDE cannot connect to it, shows empty window 

### Expected Behavior
When ESP32 board is connected to a USB port ESP-IDF serial monitor in Eclipse IDE should connect to it, and display the board window 

### Actual Behavior
When ESP32 board is connected to a USB port ESP-IDF serial monitor in Eclipse IDE cannot connect to it, shows empty window 

### Steps to reproduce

1. connect ESp32 board to USb port
2. Click terminal icon and open dialog, select USB port, click OK
3. Notice that the terminal view says "connecting" but never does, no output displayed
4. Notice that using "screen" command from Mac's terminal app connecting to same USB port works fine and shows ESP32 board output

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

kolipakakondal, 2022-05-16:
Hi @Paveldeg Is it on Apple M1(arm based)? There was an issue reported earlier and it was fixed with https://github.com/espressif/idf-eclipse-plugin/commit/7328171cb49df45520679aaa8a663f6a7e783a34. Fix will be available with the v2.5.0 release

If it's not Apple M1, could you share the error log to check further? 
https://github.com/espressif/idf-eclipse-plugin#error-log

Paveldeg, 2022-05-16:
Hi,
Yes this is M1. Thanks for the update.

kolipakakondal, 2022-05-17:
Hi @Paveldeg Thanks for your reply.

https://github.com/espressif/idf-eclipse-plugin/suites/6426977112/artifacts/234930711
Here is the master build update site which you could install to avoid this error.

How to install it? Please check [this](https://github.com/espressif/idf-eclipse-plugin#installing-idf-eclipse-plugin-from-local-archive)

closed 2022-05-17 by Paveldeg