espressif/esp-idf #8944: Unknown CMake command "fatfs_create_spiflash_image"

MadDogMayCry0, 2022-05-13:
esp-idf 1.4.0 @ vscode extension

when i put it to CMakeList.txt
```spiffs_create_partition_image(storage ../spiffs_image FLASH_IN_PROJECT)```

```
CMake Error at main/CMakeLists.txt:2 
(fatfs_create_spiflash_image) 
Unknown CMake command "fatfs_create_spiflash_image".
```
that's it

igrr, 2022-05-13:
@MadDogMayCry0 the feature to generate FatFS partitions is new, it has been introduced on the `master` branch only and isn't present in any of the released versions yet.
You have probably met the issue because you have tried to compile an example provided on the `master` branch with and older version of esp-idf.

MadDogMayCry0, 2022-05-13:
> @MadDogMayCry0 the feature to generate FatFS partitions is new, it has been introduced on the `master` branch only and isn't present in any of the released versions yet. You have probably met the [...]

That's what I thought :)

closed 2022-05-13 by MadDogMayCry0