espressif/esp-idf #8945: Nimble BLE cannot send notification when multiple clients are connected

listout, 2022-05-13:
## Environment

- Development Kit: N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm using a slightly modified version of the Nimble SPP example code. The primary changes I've made are adding a send gap event callback function if a second device connects (checking connection handle) using the function `ble_gap_set_event_cb`. So it looks something like this: 

```c
	    if (connection_handle == 1)
		    ble_gap_set_event_cb(connection_handle, ble_test_server_gap_event, NULL);
```

Now when I try to send a notification to the second device or connection handle 1, I'm getting a guru meditation error. I can correctly send notification to the other device or connection handle 0. I've noticed this issue while I try to send notification to two separate value handle, too. I've looked through the IDF documentation for Nimble but couldn't find much about handling multiple clients.

### Expected Behavior

Able to notify to chosen device using the connection handle. 

### Actual Behavior

Core panics when trying to do so.

### Steps to reproduce

1. Connect two clients or centrals to the SPP peripheral device
2. Try notifying to the second device

### Code to reproduce this issue

Code was long, so attached [gist](https://gist.github.com/listout/45065afdcb8ceb1b13decb6bffbf16de).

## Debug Logs

```
vhci.c 134
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008a241  PS      : 0x00060035  A0      : 0x8008a428  A1      : 0x3ffbe6e0  
0x4008a241: vhci_recv_end at vhci.c:?

A2      : 0x00000002  A3      : 0x3ffb557c  A4      : 0x40017620  A5      : 0x3ffb8140  
A6      : 0x00000003  A7      : 0x00060723  A8      : 0x8008a241  A9      : 0x3ffbe650  
A10     : 0x0000000b  A11     : 0x0000000b  A12     : 0x0000000c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x400d2c4b  EPC2    : 0x00000000  EPC3    : 0x4008a241  EPC4    : 0x4008a241
0x400d2c4b: uart_hal_write_txfifo at /home/listout/esp/esp-idf/components/hal/uart_hal_iram.c:35

0x4008a241: vhci_recv_end at vhci.c:?

0x4008a241: vhci_recv_end at vhci.c:?

Backtrace:0x4008a23e:0x3ffbe6e00x4008a425:0x3ffbe700 0x400e7ba8:0x3ffbe720 0x4008a43c:0x3ffbe740 0x40082d79:0x3ffbe760 0x4008363e:0x3ffbc2f0 0x4008a1c9:0x3ffbc310 0x400f84c8:0x3ffbc330 0x400e0d68:0x3ffbc360 0x400e06be:0x3ffbc380 0x400d7639:0x3ffbc4a0 0x400ddde5:0x3ffbc4c0 0x400dde5a:0x3ffbc4f0 0x400dc76f:0x3ffbc510 0x400da03e:0x3ffbc540 0x400d9e85:0x3ffbc570 0x400d9215:0x3ffbc5a0 0x400d6a25:0x3ffbc5c0 0x40124589:0x3ffbc5e0 0x400939ed:0x3ffbc600 
0x4008a23e: vhci_recv_end at vhci.c:?

0x4008a425: r_vhci_isr at ??:?

0x400e7ba8: r_eif_isr at ??:?

0x4008a43c: r_eif_isr_wrapper at intc.c:?

0x40082d79: _xt_lowint1 at /home/listout/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4008363e: cause_sw_intr_to_core_wrapper at /home/listout/esp/esp-idf/components/bt/controller/esp32/bt.c:1033

0x4008a1c9: vhci_set_interrupt at vhci.c:?

0x400f84c8: API_vhci_host_send_packet at ??:?

0x400e0d68: esp_vhci_host_send_packet at /home/listout/esp/esp-idf/components/bt/controller/esp32/bt.c:1381

0x400e06be: ble_hci_trans_hs_acl_tx at /home/listout/esp/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:136

0x400d7639: ble_hs_tx_data at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:714

0x400ddde5: ble_hs_hci_acl_tx_now at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:519

0x400dde5a: ble_hs_hci_acl_tx at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:577

0x400dc76f: ble_l2cap_tx at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap.c:452

0x400da03e: ble_att_tx at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_cmd.c:75

0x400d9e85: ble_att_clt_tx_notify at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:902

0x400d9215: ble_gattc_notify_custom at /home/listout/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4178

0x400d6a25: app_main at /home/listout/Documents/work/idf_projects/ble_server/build/../main/main.c:555

0x40124589: main_task at /home/listout/esp/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x400939ed: vPortTaskWrapper at /home/listout/esp/esp-idf/components/freertos/port/xtensa/port.c:131

Core  1 register dump:
PC      : 0x4012364e  PS      : 0x00060835  A0      : 0x800d27ca  A1      : 0x3ffbd460  
0x4012364e: esp_pm_impl_waiti at /home/listout/esp/esp-idf/components/esp_pm/pm_impl.c:839

A2      : 0x00000000  A3      : 0x00000028  A4      : 0x80093ca1  A5      : 0x3ffba520  
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x801017a2  A9      : 0x3ffbd430  
A10     : 0x00000000  A11     : 0x010f9d15  A12     : 0x0000001a  A13     : 0x00000030  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x4012364b:0x3ffbd4600x400d27c7:0x3ffbd480 0x40092469:0x3ffbd4a0 0x400939ed:0x3ffbd4c0 
0x4012364b: cpu_ll_waiti at /home/listout/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /home/listout/esp/esp-idf/components/esp_pm/pm_impl.c:837

0x400d27c7: esp_vApplicationIdleHook at /home/listout/esp/esp-idf/components/esp_system/freertos_hooks.c:63

0x40092469: prvIdleTask at /home/listout/esp/esp-idf/components/freertos/tasks.c:3973 (discriminator 1)

0x400939ed: vPortTaskWrapper at /home/listout/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

listout, 2022-05-16:
I even tried using another read_val_handle and having separate UUID's for the two clients then using subscribe event to set flags. Then based upon the flags I tried sending notification. Same issue. When either of the clients are connected, I can send notifications perfectly fine but can't send notification when both the clients are connected. Only the fist client gets notified and after that I get the same backtrace error.

listout, 2022-05-16:
Please, does anyone know anything about this issue? Am I doing something wrong or is this by design?

listout, 2022-05-17:
@IshaESP okay. Thanks for clarifying.

closed 2022-05-17 by listout

IshaESP, 2022-05-17:
Hi
Yes, I have edited my previous comment. It supports connection but not sending notifications.

listout, 2022-05-17:
Got it.

ascillato, 2023-05-25:
This is working on latest version