espressif/esp-idf #8946: Attaching interrupt to GPIO38 to wake from deep sleep does nothing

DustinWatts, 2022-05-13:
## Environment

- Development Kit: ESP32-S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): V1
- Module or chip used:  ESP32-S3-WROOM-1-N8R8
- IDF version (run ``git describe --tags`` to find it):
    // v4.4-1-qb8050b365e
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // crosstool-NG esp-2021r2-patch3 (8.4.0
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes. Arduino IDE
- Power Supply: USB

## Problem Description

Attaching an interrupt to GPIO38 does not wake the ESP32 from deep sleep. Although the documentation says it should:

```
gpio_num: GPIO number used as wakeup source. 
Only GPIOs which are have RTC functionality can be used: 0,2,4,12-15,25-27,32-39. 
```
Source: https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s3/api-reference/system/sleep_modes.html#_CPPv428esp_sleep_enable_ext0_wakeup10gpio_num_ti

The function `esp_sleep_is_valid_wakeup_gpio()` also confirms that this does not work, because it returns FALSE. 

Used GPIO1 as a test, and that works although the docs says it shouldn't. 

### Expected Behavior

Pulling GPIO38 to GND wakes up the ESP32S3 from deep sleep.

### Actual Behavior

The ESP32S3 does not wake up.

### Steps to reproduce

1. Upload a sketch with the code below
2. Wait for the ESP to go to sleep
3. Try to wake it up by pulling GPIO38 to ground.

### Code to reproduce this issue

In Adruino:
```cpp
pinMode(GPIO_NUM_38, INPUT_PULLUP);

esp_sleep_enable_ext0_wakeup(GPIO_NUM_38, 0);
esp_deep_sleep_start();
```

## Other items if possible
Using a simple sketch to check for valid wakeup GPIOs only pins 1 trough 21 are valid:

```
Wake up should work on GPIO_NUM_1: 1
Wake up should work on GPIO_NUM_2: 1
Wake up should work on GPIO_NUM_3: 1
Wake up should work on GPIO_NUM_4: 1
Wake up should work on GPIO_NUM_5: 1
Wake up should work on GPIO_NUM_6: 1
Wake up should work on GPIO_NUM_7: 1
Wake up should work on GPIO_NUM_8: 1
Wake up should work on GPIO_NUM_9: 1
Wake up should work on GPIO_NUM_10: 1
Wake up should work on GPIO_NUM_11: 1
Wake up should work on GPIO_NUM_12: 1
Wake up should work on GPIO_NUM_13: 1
Wake up should work on GPIO_NUM_14: 1
Wake up should work on GPIO_NUM_15: 1
Wake up should work on GPIO_NUM_16: 1
Wake up should work on GPIO_NUM_17: 1
Wake up should work on GPIO_NUM_18: 1
Wake up should work on GPIO_NUM_19: 1
Wake up should work on GPIO_NUM_20: 1
Wake up should work on GPIO_NUM_21: 1
Wake up should work on GPIO_NUM_26: 0
```

From 26 on `esp_sleep_is_valid_wakeup_gpio()` returns FALSE.

DustinWatts, 2022-05-14:
Update:

Using a simple sketch to check for valid wakeup GPIOs only pins 1 trough 21 are valid:
```
Wake up should work on GPIO_NUM_1: 1
Wake up should work on GPIO_NUM_2: 1
Wake up should work on GPIO_NUM_3: 1
Wake up should work on GPIO_NUM_4: 1
Wake up should work on GPIO_NUM_5: 1
Wake up should work on GPIO_NUM_6: 1
Wake up should work on GPIO_NUM_7: 1
Wake up should work on GPIO_NUM_8: 1
Wake up should work on GPIO_NUM_9: 1
Wake up should work on GPIO_NUM_10: 1
Wake up should work on GPIO_NUM_11: 1
Wake up should work on GPIO_NUM_12: 1
Wake up should work on GPIO_NUM_13: 1
Wake up should work on GPIO_NUM_14: 1
Wake up should work on GPIO_NUM_15: 1
Wake up should work on GPIO_NUM_16: 1
Wake up should work on GPIO_NUM_17: 1
Wake up should work on GPIO_NUM_18: 1
Wake up should work on GPIO_NUM_19: 1
Wake up should work on GPIO_NUM_20: 1
Wake up should work on GPIO_NUM_21: 1
Wake up should work on GPIO_NUM_26: 0
```
From 26 on `esp_sleep_is_valid_wakeup_gpio()` returns FALSE.

igrr, 2022-05-14:
I'm afraid the function description was written for the esp32 chip, we forgot to update it for the esp32-s3. In this case the result of esp_sleep_is_valid_wakeup_gpio is correct: on esp32-s3 GPIOs above 26 are not RTC IOs, and can't be used for wakeup from deep sleep.

We'll update the function description to fix this inconsistency!

DustinWatts, 2022-05-14:
Thanks for the update!

closed 2022-05-14 by DustinWatts