espressif/esp-idf #8948: Unified Provisioning via HTTP REST server?

twlawrenceko, 2022-05-14:
Hello,

We would like to know if we set up a REST server on ESP, what is the mechanism built inside ESPSoftAPProvisioning app that we can migrate into REST server? So user can visit webpage created by ESP and enter WiFi credentials for provisoning.

Thank you.

laukik-hase, 2022-06-06:
Hello, @twlawrenceko!

WiFi Provisioning in ESP-IDF uses a custom protocol based on `protobuf` ([Protocol Buffers](https://github.com/protocolbuffers/protobuf)) for secure session establishment and sharing of credentials. You can read about the implementation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html).

If security is not an issue in your case, please check out [tonyp7/esp32-wifi-manager](https://github.com/tonyp7/esp32-wifi-manager) as a reference.

mahavirj, 2022-06-20:
Hope reply in earlier comment is helpful here. Please use esp32.com forum for further questions.

closed 2022-06-20 by mahavirj