espressif/esp-idf #8949: WebSocket client example is missing

vinothkrstatiq, 2022-05-14:
earlier i saw a WebSocket client example in the protocols folder. Now it is not available cloud you please update it again .Also check it is working then upload it

david-cermak, 2022-05-17:
Hi @vinothkrstatiq 

[esp_websocket_client](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client) has been moved to an external repository, the example is located here https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client/examples

Please check the documentation on [Removed or deprecated components](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/migration-guides/removed-components.html#removed-or-deprecated-components)

vinothkrstatiq, 2022-05-17:
Hi @david-cermak  ,
Thanks for the answer .can you also update me that  given example tested and  working from your end 
https://github.com/espressif/esp-protocols/blob/master/components/esp_websocket_client/examples/main/websocket_example.c

david-cermak, 2022-05-17:
Yes, this example is tested and works as expected, not only on my end, but also in `esp-protocols`'s CI infrastructure:
https://github.com/espressif/esp-protocols/runs/6272732011?check_suite_focus=true#step:7:104

vinothkrstatiq, 2022-05-17:
Hi @david-cermak 

Yes, it is working for me. wonderful 
I am grateful for your help and support 
cheers.....

filzek, 2022-05-18:
> Yes, this example is tested and works as expected, not only on my end, but also in `esp-protocols`'s CI infrastructure: https://github.com/espressif/esp-protocols/runs/6272732011?check_suite_focus=true#step:7:104

David, do you think that a full hole punch p2p example could be created by espressif running for esp32 X app X cloud usage, this should be extreme important to next generation non cloud demand devices...

AxelLin, 2022-05-18:
@david-cermak 

A question regarding component version in idf_component.yml. (Sorry, this is off-topic)

What is the menaing of "~" and "^" before the version?
Are they the same meaning? means >= or == ?
e.g.
component: "~1.0.0"
component: "^1.0.0"

david-cermak, 2022-05-18:
> full hole punch p2p example could be created by espressif 

@filzek  Yes, please create a new feature request, this sounds interesting and could be helpful for more users.

---

> What is the menaing of "~" and "^" before the version?

@AxelLin This notation is used to restrict the component to a specific *MAJOR* (`^`) or `MINOR` (`~`) version, for example:
* `~1.2.0` means `>=1.2.0,<1.3.0` (or `~1.2`)
* `^1.0.0` means `>=1.0.0,<2.0.0` (or `^1.0` or `^1`)

closed 2022-05-23 by Alvin1Zhang