espressif/esp-idf #8951: ESP32 USB Host CDC-ACM driver for CH34x

AristarchosOfSamos, 2022-05-14:
Hi,

Seen CDC USB host drivers for FTDI and SiLabs chips in new S3/S2/C3 ESP32 variants.

Although these do work, it seems that the CH34x USB serial chip brand is quite widespread and in use by really almost anything nowadays in many devices and gadgets.
So, it could be a good idea if such a driver could be implemented.
Eg. now we cannot use one of the abundant arduino clones as a device attached on a ESP32-S3, having this driver though it could open lots of prospects.

Is there any hint if this has been done already ?

Thanks in advance,

tore-espressif, 2022-05-17:
Hi @AristarchosOfSamos , thank you for your interest in esp-idf USB drivers.

Indeed, CH340 needs a custom driver. The RX and TX paths work, but functions for setting of linecoding, baudrate etc need to be implemented. I'll keep you updated.

Just a side note to your post: ESP32C3 does not contain USB On-the-Go peripheral (only USB-serial-JTAG)

AristarchosOfSamos, 2022-05-17:
Hi,
Thanks for considering this request, a CH34x driver is definitively required nowadays!
Waiting to read any news on it, got some ESP32-S3-Devkits here to test that.
Regards,

tore-espressif, 2022-05-20:
Update: we are waiting for CH340 manufacturer to provide complete documentation

sramrajkar, 2022-07-03:
Hi,
Sorry to bug on this topic, but even I am using ESP32-S2 (or S3 in future) for an ELM327 USB OBD adaptor and it uses CH340T. Currently I would also need this feature to set line parameters as the baud rate of the OBD IC and the default baud of CH340 dont match.
Any luck on this front? I dont mind applying any patches myself. But please note I am super new to USB.
Also any idea when would CH34x be supported in the IDF?
Thank you

tore-espressif, 2022-07-07:
Hello @sramrajkar @AristarchosOfSamos 

I uploaded an early version of CH340 driver into my esp-idf fork, feel free to have a look: https://github.com/tore-espressif/esp-idf/tree/feature/usb_host/cdc_ch340/examples/peripherals/usb/host/cdc/cdc_acm_vcp

It covers only the basics: RX/TX, setting of baudrate, linecoding and DTR/RTS control.

_Note:_ Please note that there are two commits in the linked branch:
1. CDC-ACM driver bug fix https://github.com/tore-espressif/esp-idf/commit/438234ad702fcd8ddc9781dcb2782f15a014d235
2. CH340 driver https://github.com/tore-espressif/esp-idf/commit/d7f3702c610c1689f500ad130cfc06d619638a27

Please let me know whether this solution works for you

AristarchosOfSamos, 2022-07-08:
Hi  @tore-espressif , first test looks ok.
Got it compiled and after flashed on a esp32-s3-devkitc it can open an attached CH34x on its USB port {an arduino nano clone) viaa the provided test/example.

Output
`I (277) sleep: Configure to isolate all GPIO pins in sleep state
I (283) sleep: Enable automatic switching of GPIO sleep configuration
I (291) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (311) VCP example: Installing USB Host
I (341) VCP example: Installing CDC-ACM driver
I (341) VCP example: Opening CH340 device
I (741) VCP example: Setting up line coding
`

I'll run some tests and see how the comms are, I'll report back with any remarks.

Either way, thanks for your effort.

Aristarchos

AristarchosOfSamos, 2022-07-08:
Ok @tore-espressif 
Done more tests, speed, disconnection/reconnection, full duplex and it looks just fine!
For me it is ready to be commited on main esp-idf repo in v5.0 branch (I guess).

Thanks for an excellent work.
Aristarchos

alexandru98765, 2022-07-09:
Hi, i got some problems when trying to build this to an esp32-S2. In menuconfig i set the esp system settings - channel for console output to usb cdc and the example configuration to ch340. Can someone help me with this?

```
[98/100] Linking C executable bootloader.elf
[99/100] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated C:/Espressif/frameworks/esp-idf-v4.4/examples/peripherals/usb/host/cdc/cdc_acm_vcp/build/bootloader/bootloader.bin
[100/100] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4\examples\peripherals\usb\host\cdc\cdc_acm_vcp\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Espressif/frameworks/esp-idf-v4.4/examples/peripherals/usb/host/cdc/cdc_acm_vcp/build/bootloader/bootloader.bin"
Bootloader binary size 0x51c0 bytes. 0x1e40 bytes (27%) free.
[1028/1037] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/ch34x_usb.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/ch34x_usb.cpp.obj

ccache C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I../../../../../../../components/newlib/platform_include -I../../../../../../../components/freertos/include -I../../../../../../../components/freertos/include/esp_additions/freertos -I../../../../../../../components/freertos/port/xtensa/include -I../../../../../../../components/freertos/include/esp_additions -I../../../../../../../components/esp_hw_support/include -I../../../../../../../components/esp_hw_support/include/soc -I../../../../../../../components/esp_hw_support/include/soc/esp32s2 -I../../../../../../../components/esp_hw_support/port/esp32s2/. -I../../../../../../../components/esp_hw_support/port/esp32s2/private_include -I../../../../../../../components/heap/include -I../../../../../../../components/log/include -I../../../../../../../components/lwip/include/apps -I../../../../../../../components/lwip/include/apps/sntp -I../../../../../../../components/lwip/lwip/src/include -I../../../../../../../components/lwip/port/esp32/include -I../../../../../../../components/lwip/port/esp32/include/arch -I../../../../../../../components/soc/include -I../../../../../../../components/soc/esp32s2/. -I../../../../../../../components/soc/esp32s2/include -I../../../../../../../components/hal/esp32s2/include -I../../../../../../../components/hal/include -I../../../../../../../components/hal/platform_port/include -I../../../../../../../components/esp_rom/include -I../../../../../../../components/esp_rom/include/esp32s2 -I../../../../../../../components/esp_rom/esp32s2 -I../../../../../../../components/esp_common/include -I../../../../../../../components/esp_system/include -I../../../../../../../components/esp_system/port/soc -I../../../../../../../components/esp_system/port/public_compat -I../../../../../../../components/xtensa/include -I../../../../../../../components/xtensa/esp32s2/include -I../../../../../../../components/driver/include -I../../../../../../../components/driver/esp32s2/include -I../../../../../../../components/esp_pm/include -I../../../../../../../components/esp_ringbuf/include -I../../../../../../../components/efuse/include -I../../../../../../../components/efuse/esp32s2/include -I../../../../../../../components/vfs/include -I../../../../../../../components/esp_wifi/include -I../../../../../../../components/esp_event/include -I../../../../../../../components/esp_netif/include -I../../../../../../../components/esp_eth/include -I../../../../../../../components/tcpip_adapter/include -I../../../../../../../components/esp_phy/include -I../../../../../../../components/esp_phy/esp32s2/include -I../../../../../../../components/esp_ipc/include -I../../../../../../../components/app_trace/include -I../../../../../../../components/esp_timer/include -I../../../../../../../components/mbedtls/port/include -I../../../../../../../components/mbedtls/mbedtls/include -I../../../../../../../components/mbedtls/esp_crt_bundle/include -I../../../../../../../components/app_update/include -I../../../../../../../components/spi_flash/include -I../../../../../../../components/bootloader_support/include -I../../../../../../../components/nvs_flash/include -I../../../../../../../components/pthread/include -I../../../../../../../components/esp_gdbstub/include -I../../../../../../../components/esp_gdbstub/xtensa -I../../../../../../../components/esp_gdbstub/esp32s2 -I../../../../../../../components/espcoredump/include -I../../../../../../../components/espcoredump/include/port/xtensa -I../../../../../../../components/wpa_supplicant/include -I../../../../../../../components/wpa_supplicant/port/include -I../../../../../../../components/wpa_supplicant/esp_supplicant/include -I../../../../../../../components/ieee802154/include -I../../../../../../../components/console -I../../../../../../../components/asio/asio/asio/include -I../../../../../../../components/asio/port/include -I../../../../../../../components/cbor/port/include -I../../../../../../../components/unity/include -I../../../../../../../components/unity/unity/src -I../../../../../../../components/cmock/CMock/src -I../../../../../../../components/coap/port/include -I../../../../../../../components/coap/lib [... cut ...]
         esp_err_t err_rc_ = (x);                                                                \
                              ^
../main/cp210x_usb.cpp:80:18: error: 'class esp_usb::CP210x' has no member named 'send_custom_request'
     return this->send_custom_request(CP210X_WRITE_REQ, CP210X_CMD_SET_BREAK, 0, this->intf, 0, NULL);
                  ^~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Espressif\frameworks\esp-idf-v4.4\examples\peripherals\usb\host\cdc\cdc_acm_vcp>
```

AristarchosOfSamos, 2022-07-09:
@tore-espressif 
Hope you will soon get this commited in the main repo.
Thanks again.
Aristarchos

closed 2022-07-09 by AristarchosOfSamos

tore-espressif, 2022-07-11:
@AristarchosOfSamos I'm happy it works for you.

@alexandru98765 It seems that your esp-idf is out-dated. We added support for custom commands only in April (so it isn't included in any official release yet). Please update esp-idf and try again. If you keep having problems, please open another issue, thanks!

sramrajkar, 2022-08-04:
@tore-espressif Thank you for this feature addition. It works very well for me.

THEDSP, 2022-08-30:
Hello I got the following error 
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\cdc_acm_vpc.o:(.literal.app_main+0x54): undefined reference to `cdc_acm_host_install' 
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\cdc_acm_vpc.o:(.literal.app_main+0x60): undefined reference to `cdc_acm_host_close'   
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\cdc_acm_vpc.o: in function `app_main':
C:\Users\Dvimaya\Documents\PlatformIO\Projects\ESP32ch34/src/cdc_acm_vpc.cpp:79: undefined reference to `cdc_acm_host_install'       
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\cdc_acm_vpc.o: in function `CdcAcmDevice::close()':
C:/Users/Dvimaya/.platformio/packages/framework-espidf/components/usb/include/usb/cdc_acm_host.h:298: undefined reference to `cdc_acm_host_close'
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\ch34.o:(.literal._ZN7esp_usb5CH34xC2EtPK28cdc_acm_host_device_config_th+0x8): undefined reference to `cdc_acm_host_open_vendor_specific'
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\ch34.o:(.literal._ZN7esp_usb5CH34x15line_coding_setEP21cdc_acm_line_coding_t+0x24): undefined reference to `cdc_acm_host_send_custom_request'
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\ch34.o: in function `CdcAcmDevice::open_vendor_specific(unsigned short, unsigned short, unsigned char, cdc_acm_host_device_config_t const*)':
C:/Users/Dvimaya/.platformio/packages/framework-espidf/components/usb/include/usb/cdc_acm_host.h:293: undefined reference to `cdc_acm_host_open_vendor_specific'
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\ch34.o: in function `CdcAcmDevice::close()':
C:/Users/Dvimaya/.platformio/packages/framework-espidf/components/usb/include/usb/cdc_acm_host.h:298: undefined reference to `cdc_acm_host_close'
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\src\ch34.o: in function `CdcAcmDevice::send_custom_request(unsigned char, unsigned char, unsigned short, unsigned short, unsigned short, unsigned char*)':
C:/Users/Dvimaya/.platformio/packages/framework-espidf/components/usb/include/usb/cdc_acm_host.h:327: undefined reference to `cdc_acm_host_send_custom_request'
c:/users/dvimaya/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/Dvimaya/.platformio/packages/framework-espidf/components/usb/include/usb/cdc_acm_host.h:327: undefined reference to `cdc_acm_host_send_custom_request'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1

THEDSP, 2022-08-30:
> @tore-espressif Thank you for this feature addition. It works very well for me.

can you help me to successful compilations ?

sramrajkar, 2022-08-31:
Hey,
Did you include the patched component in your build system? What IDF
version are you using?

THEDSP, 2022-09-01:
> Hey, Did you include the patched component in your build system? What IDF version are you using?
> […](#)

i used 4.4 master in main branch of idf

sramrajkar, 2022-09-01:
Where did you put the CDC code?
And did you modify make files?
Also I am building this with 5.x dev branch of idf.

THEDSP, 2022-09-01:
i found linux driver for ch341 for epp i2c mode so how can we port them in to the esp.
https://github.com/zoobab/ch341-parport/tree/master/CH341PAR_LINUX
where to start any hint ?

sramrajkar, 2022-09-04:
I am just guessing at this point. If you know what custom commands are
needed for I2C to work, then this patch from expressif shows how to send
those commands over the USB interface. You may have to replicated that for
the I2C commands yourself after enumeration is successful.

THEDSP, 2022-09-04:
Which patch

THEDSP, 2022-09-05:
> I am just guessing at this point. If you know what custom commands are needed for I2C to work, then this patch from expressif shows how to send those commands over the USB interface. You may have to replicated that for the I2C commands yourself after enumeration is successful.
> […](#)

hello @sramrajkar 
which patch do you talking about would you tell me the bare mettle api for usb communication i am confused ?

tore-espressif, 2022-09-05:
To everyone seeing this in the future: Please open new issues for new problems, if no other issue describes the problem you see, rather than replying here, we've locked this not because we don't care about issues but because we do care about them, and tracking a number of discussions in a long thread makes it hard for us to address them all.

THEDSP, 2022-09-05:
sorry sir @tore-espressif 
hello @sramrajkar  we can continued our diction in #9652

a1exwang, 2022-11-18:
> Hello @sramrajkar @AristarchosOfSamos
> 
> I uploaded an early version of CH340 driver into my esp-idf fork, feel free to have a look: https://github.com/tore-espressif/esp-idf/tree/feature/usb_host/cdc_ch340/examples/peripherals/usb/host/cdc/cdc_acm_vcp
> 
> It covers only the basics: RX/TX, setting of baudrate, linecoding and DTR/RTS control.
> 
> _Note:_ Please note that there are two commits in the linked branch:
> 
> 1. CDC-ACM driver bug fix [tore-espressif@438234a](https://github.com/tore-espressif/esp-idf/commit/438234ad702fcd8ddc9781dcb2782f15a014d235)
> 2. CH340 driver [tore-espressif@93e9c72](https://github.com/tore-espressif/esp-idf/commit/93e9c72534975243e0b7b4e09a1e5e818ba3a3e7)
> 
> Please let me know whether this solution works for you

Works on my side with CH340. Looking to changing this to a serial port over Wifi solution.

sramrajkar, 2022-12-27:
@tore-espressif Would this driver work for all the 4 channels of CH344 with RS485 support?

ChanSeongKim, 2023-05-06:
> Hello @sramrajkar @AristarchosOfSamos
>
> I uploaded an early version of CH340 driver into my esp-idf fork, feel free to have a look: [...]

I am trying to make PL2303 driver for host mode. 
Because I am trying to legacy device connectable through wireless using ESP32S3.  Legacy device has usb port connected with PL2303 IC.

Can you guide how to make driver PL2303 ?  
I started from cp210 driver.  But  datasheet of pl2303 is not enough to code. 
So I am stuck in how to modify this function. 

Following is part of my code.  
//pl2303_usb.hpp
#define PL230X_CMD_SET_LINE_CTL    (0x03) // Set the line control
...
#define PL230X_CMD_SET_BAUDRATE    (0x1E) // Set the baud rate

// my working on for PL2303.cpp , modifed from other driver source.  
esp_err_t PL230x::line_coding_set(cdc_acm_line_coding_t *line_coding)
{
    assert(line_coding);

    if (line_coding->dwDTERate != 0) {
        ESP_RETURN_ON_ERROR(this->send_custom_request(PL230X_WRITE_REQ, PL230X_CMD_SET_BAUDRATE, 0, this->intf, sizeof(line_coding->dwDTERate), (uint8_t *)&line_coding->dwDTERate), "PL230X",);
    }

    if (line_coding->bDataBits != 0) {
        const uint16_t wValue = line_coding->bCharFormat | (line_coding->bParityType << 4) | (line_coding->bDataBits << 8);
        return this->send_custom_request(PL230X_WRITE_REQ, PL230X_CMD_SET_LINE_CTL, wValue, this->intf, 0, NULL);
    }
    return ESP_OK;
}

// reference code of cp210x_usb.cpp 
esp_err_t CP210x::line_coding_set(cdc_acm_line_coding_t *line_coding)
{
    assert(line_coding);

    if (line_coding->dwDTERate != 0) {
        ESP_RETURN_ON_ERROR(this->send_custom_request(CP210X_WRITE_REQ, CP210X_CMD_SET_BAUDRATE, 0, this->intf, sizeof(line_coding->dwDTERate), (uint8_t *)&line_coding->dwDTERate), "CP210X",);
    }

    if (line_coding->bDataBits != 0) {
        const uint16_t wValue = line_coding->bCharFormat | (line_coding->bParityType << 4) | (line_coding->bDataBits << 8);
        return this->send_custom_request(CP210X_WRITE_REQ, CP210X_CMD_SET_LINE_CTL, wValue, this->intf, 0, NULL);
    }
    return ESP_OK;
}

Thank you in advance.

tore-espressif, 2023-05-12:
Hello @ChanSeongKim , the vendor claims that PL-2303 is CDC compatible, so you can try your luck with the base [CDC-ACM driver](https://components.espressif.com/components/espressif/usb_host_cdc_acm).

If it does not work, you willhave to reverse engineer from linux driver https://components.espressif.com/components/espressif/usb_host_cdc_acm

ChanSeongKim, 2023-05-14:
Thank you for your email.
I will check your information.

Since I didn't succeed with PL2303 until now, So I am trying with ch340 USB
IC.
I succeeded in connection with ch34x_usb.cpp  which is for Arduino
Nano(built with CH340 USB IC).
Following is the connection message with ch34x.
(Hello:0 is the message sent from arduino nano with 57600bps.)

I (307) VCP example: Enabling USB_HOST interface through USB_SEL---

I (307) VCP example: Example configure_usb_sel ====  !

I (317) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1|
Pulldown: 0| Intr:0

I (1327) VCP example: Installing USB Host

I (1357) VCP example: Installing CDC-ACM driver

I (1357) VCP example: Opening CH340 device

I (1757) VCP example: Setting up line coding

I (1757) VCP example: baudrate:57600(d)

rx=>Hello: 0

rx=>Hello: 1

rx=>Hello: 2

rx=>Hello: 3

Thank you for your interest.

tore-espressif, 2023-05-17:
looks good! is your code publicly available? would you be interested in creating an ESP-IDF component for PL2303?

ChanSeongKim, 2023-05-17:
I think it will be avaliable soon.  I think it would be ready nextweek. Because it is now some spaghetti code. I need  time to arrange it.  Let me know if you can introduce process of creating ESP-IDF component for PL2303HX.

tore-espressif, 2023-05-19:
An esp-idf component should follow our standard conventions such as containing CMakeLists.txt and README.md (more details in our [Build System docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html).

For the PL2303 component, it would be good to follow the design that is outlined in similar components such as [CH34x](https://github.com/espressif/idf-extra-components/tree/master/usb/usb_host_ch34x_vcp) or [CP210x](https://github.com/espressif/idf-extra-components/tree/master/usb/usb_host_cp210x_vcp). If you wanted to contribute the PL2303 component to our esp-idf registry (such as [here](https://components.espressif.com/components/espressif/usb_host_ch34x_vcp)), you 'd have to add also an `idf_compnent.yaml` file, described in [Component Manager docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html)

Please, let me know, if you needed any help

sxs777, 2023-07-13:
Hi,

I've tried running cdc_acm_vcp example to communicate between esp32s3 usb host and CH340, however I could not build the code.
I ve run 2 commands to add dependencies:
1-  idf.py add-dependency "espressif/usb_host_cdc_acm^2.0.0"
2 - idf.py add-dependency "espressif/usb_host_ch34x_vcp^2.0.0"

When building the code I get the following errors.
In file included from C:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/main/usb_cdc_acm_vcp.c:10:
C:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/managed_components/espressif__usb_host_ch34x_vcp/include/usb/vcp_ch34x.hpp:10:10: fatal error: array: No such file or directory
   10 | #include <array>
      |          ^~~~~~~
compilation terminated.
[3/10] ccache C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe  -IC:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/build/config -IC:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/managed_components/espressif__usb_host_ch34x_vcp/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/esp_additions/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/esp_additions/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_hw_support/include/soc/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_hw_support/port/esp32s3/. -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_hw_support/port/esp32s3/private_include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/heap/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/log/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/soc/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/soc/esp32s3/. -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/soc/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/hal/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/hal/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_rom/include/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_rom/esp32s3 -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/port/include/private -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/xtensa/esp32s3/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/port/esp32/include/arch -IC:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/managed_components/espressif__usb_host_cdc_acm/include -IC:/Espressif/frameworks/esp-idf-v5.0.1/components/usb/include -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=C:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v5.0.1=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -Wno-format-nonliteral -Wno-format-security -Wformat=0 -std=gnu++20 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v5.0.1-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/espressif__usb_host_ch34x_vcp/CMakeFiles/__idf_espressif__usb_host_ch34x_vcp.dir/usb_host_ch34x_vcp.cpp.obj -MF esp-idf\espressif__usb_host_ch34x_vcp\CMakeFiles\__idf_espressif__usb_host_ch34x_vcp.dir\usb_host_ch34x_vcp.cpp.obj.d -o esp-idf/espressif__usb_host_ch34x_vcp/CMakeFiles/__idf_espressif__usb_host_ch34x_vcp.dir/usb_host_ch34x_vcp.cpp.obj -c C:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/managed_components/espressif__usb_host_ch34x_vcp/usb_host_ch34x_vcp.cpp
FAILED: esp-idf/espressif__usb_host_ch34x_vcp/CMakeFiles/__idf_espressif__usb_host_ch34x_vcp.dir/usb_host_ch34x_vcp.cpp.obj 
ccache C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe  -IC:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/build/config -IC:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/managed_components/espressif__usb_host_ch34x_vcp/include -IC:/Espressif/ [... cut ...]  uint8_t)':
C:/Users/serge/OneDrive/Desktop/Firmware_Projects/ESP32_Projects/blinx/managed_components/espressif__usb_host_ch34x_vcp/usb_host_ch34x_vcp.cpp:73:19: error: exception handling disabled, use '-fexceptions' to enable
   73 |         throw (err);
      |                   ^
ninja: build stopped: subcommand failed.

My understanding is that array.h is missing but how to add it?
Also not sure about how solve C++ exception related  errors. It says use -fexceptions to enable them but where?

ChanSeongKim, 2023-07-13:
Hi~
Please have a look at this following  site.

https://github.com/tore-espressif/esp-idf/tree/feature/usb_host/cdc_ch340/examples/peripherals/usb/host/cdc/cdc_acm_vcp/main

sxs777, 2023-07-14:
Thank you very much for the provided link

tore-espressif, 2023-07-24:
1. This code requires C++ exceptions. You can enable them in menuconfig-> compiler settings
2. This code requires c++14, which should be enabled in IDF v5.0.1

Can you enable the exceptions and try again?

Berserker2229, 2024-02-09:
[image]