espressif/esp-idf #8952: 4.4.1 installation doesn't result in working build

mzimmersfluidra, 2022-05-14:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it): 4.4.1
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: ESP Command Prompt].
- Using an IDE?: [No] (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery] N/A

## Problem Description

Installed IDF and toolchain using esp-idf-tools-setup-online-2.15. Installation appears good, but when I try to run "idf.py build," I get this error:

`C:\Users\Michael.Zimmers\GitHub\iQPump-ESP32\fw>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\michael.zimmers\github\iqpump-esp32\fw\build
Executing "ninja all"...
[0/1] Re-running CMake...
FAILED: build.ninja
C:\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\tools\cmake\3.20.3\bin\cmake.exe --regenerate-during-build -SC:\Users\Michael.Zimmers\GitHub\iQPump-ESP32\fw -BC:\Users\Michael.Zimmers\GitHub\iQPump-ESP32\fw\build
CreateProcess failed: The system cannot find the file specified.
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1`

Jason2866, 2022-05-15:
Can you try without the dot in the path?
`Michael.Zimmers`

mzimmersfluidra, 2022-05-15:
Hi Jason -

That's the username IT assigned me. I can ask to have it changed, but I'm not sure that can happen.

Jason2866, 2022-05-15:
Do you have only in this path rights to place stuff?

mzimmersfluidra, 2022-05-15:
Yes, but IT will install in other locations on my behalf. Would you prefer that the IDF and the toolchain directly under the C: drive?

Jason2866, 2022-05-15:
C: is a good choice, since it avoids the next problem of windows (max path len...)

dobairoland, 2022-05-16:
Hi @mzimmersfluidra. Please let us know if the installation to a different path succeeded? Thanks!

mzimmersfluidra, 2022-05-16:
Hi Roland - of course. My IT department isn't available until 8:00 AM PDT, and it'll take a few minutes to make sure everything's working, but I should have an answer for you by noon PDT today.

Thanks...

mzimmersfluidra, 2022-05-16:
OK, I got it working. Evidently, the "." in the working folder is also a no-no. I moved my repo to a top-level folder, and I can now build.

Thanks for the help with this.

dobairoland, 2022-05-17:
Thank you for the feedback and sorry for the inconvenience. We need to keep this open and find a proper fix.

georgik, 2022-05-17:
The dot is red herring in my opinion. The error message contains several strange things:
- ESP-IDF tools are deployed in `C:\.espressif`, the installer deploys tools by default to `C:\Espressif`, this looks like remains of some older installation, please check environment variable, whether it points to correct location of tools
- the error message contains upper case and lower case path, this is strange, the case should be same and it can cause a problem when Windows filesystem is configured as Case sensitive

Check Windows Long path configuration by running the installer. On the second screen you should see check for Windows long path. If it's disabled, click `Apply Fixes` to fix the system config.

I would recommend whole new installation. Clean environment variables IDF_PATH and IDF_TOOLS_PATH from environment variables. Run the installer and deploy tools to `C:\Espressif` and ESP-IDF to `C:\Espressif\frameworks\esp-idf-v4.4` The installer should display these values.

Test the setup by launching PowerShell from icon deployed on Desktop and type:
```
cd examples\get-started\blink
idf.py build
```

When build passes, change to directory with your project and enter:
```
idf.py fullclean
idf.py build
```