espressif/esp-idf #8953: WiFi stops transmitting data in long-term operation test

tomoyuki-nakabayashi, 2022-05-15:
## Environment

- Development Kit: N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

In our long-term operation test, WiFi stack seems to stop transmitting data while keeping the connection to the access point..
Our application uses esp-tls to connect the server and UDP to communicate with devices in local area network.
Both of them stops the operation.

### Expected Behavior

WiFi stack continue to transmit data.

### Actual Behavior

WiFi stack perhaps fails to flush its buffer.

### Steps to reproduce

This problem occurs 2 or 3 weeks after when we started long-term operation test, it's hard to reproduce and debug.
Our application uses both BLE and WiFi.

## Debug Logs
### getaddrinfo in esp-tls failed

log from esp-tls
```
2022-05-14 14:00:14,922 : E (1989386792) esp-tls: couldn't get hostname for :xxx.yyy.zzz: getaddrinfo() returns 202, addrinfo=0x0
```

additional log from our app showing esp-tls internal error and errno (EAGAIN).
```
2022-05-14 14:00:39,913 : W (1989411792) esp_util: esp error: 32769, ESP_ERR_ESP_TLS_CANNOT_RESOLVE_HOSTNAME
2022-05-14 14:00:39,913 : W (1989411802) esp_util: errno: 11
```

We just call `esp_tls_conn_new_async()`.

```c
        r = esp_tls_conn_new_async(url + u->field_data[UF_HOST].off, u->field_data[UF_HOST].len,
                                   get_port(url, u), &cfg, tls);
```

### UDP send failed

Our app send UDP multicast to local area network.

```
2022-05-14 14:00:51,500 : E (1989423402) el-udp: sendto failed: 12
2022-05-14 14:00:51,595 : W (1989423502) wifi:m f null
2022-05-14 14:00:51,643 : W (1989423562) wifi:m f null
```

Implementation is just like

```c
    r = sendto(s_udp->fd, buf, req_len, 0, (struct sockaddr *)&ip4_multi_addr,
                sizeof(ip4_multi_addr));
    if (-1 == r) {
        ESP_LOGE(TAG, "sendto failed: %d", errno);
    }
```

## Other items if possible
### sdkconfig.defaults

Related configurations are like this:

```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_TASK_STACK_SIZE=8192

#
# Wi-Fi
#
CONFIG_ESP32_WIFI_NVS_ENABLED=n

#
# PHY
#
CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n
CONFIG_ESP_PHY_INIT_DATA_IN_PARTITION=y

#
# DHCP server
#
CONFIG_LWIP_IPV6_AUTOCONFIG=y

#
# TCP
#
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_LWIP_TCP_MSS=1436
CONFIG_LWIP_TCP_RTO_TIME=3000
# end of TCP
```

thanks in advance,

nishanth-radja, 2022-05-16:
@tomoyuki-nakabayashi can we have the application which you are using,to try and repro the issue.Also how is the BLE used,any data is being transmitted in BLE or is it for scanning alone.

jack0c, 2022-05-20:
@tomoyuki-nakabayashi two more test you can do:
1. not use BLE
2. not use BLE + disable Wi-Fi sleep

tomoyuki-nakabayashi, 2022-05-24:
@jack0c 

Currently, I also test BLE modem sleep disable configuration.

```
BTDM_CTRL_MODEM_SLEEP=n
```

Could you tell me which configuration I'll test to disable Wi-Fi sleep?

nishanth-radja, 2022-05-24:
@tomoyuki-nakabayashi can you try this [Link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv414wifi_ps_type_t)

tomoyuki-nakabayashi, 2022-05-24:
thanks! I'll try it.

AxelLin, 2022-06-26:
> thanks! I'll try it.

@tomoyuki-nakabayashi Any update?

tomoyuki-nakabayashi, 2022-06-26:
I'm continuing to test the operation.

Currently, it keeps operation for 4 weeks with `BTDM_CTRL_MODEM_SLEEP=n` configuration.
Also I found ESP-IDF v3.3.4 has the same stuck problem (without `BTDM_CTRL_MODEM_SLEEP=n` configuration).

AxelLin, 2022-06-30:
@nishanth-radja @jack0c 
 Looks like BTDM_CTRL_MODEM_SLEEP=y causes the issue.
Any additional information you need for this issue?

AxelLin, 2022-08-04:
@jack0c @nishanth-radja 

So would you fix the issue when BTDM_CTRL_MODEM_SLEEP=y?

AxelLin, 2023-05-03:
@tomoyuki-nakabayashi 
Do you still hit this issue with recent esp-idf releases?
Does it work with BTDM_CTRL_MODEM_SLEEP=y using recent esp-idf releases?

landaming, 2024-01-04:
I had this problem too

AxelLin, 2024-01-04:
> I had this problem too

@landaming 
Could you share more details?
which esp-idf version? Your sdkconfig and which module is being used.

landaming, 2024-01-04:
> > I had this problem too
> 
> @landaming Could you share more details? which esp-idf version? Your sdkconfig and which module is being used.

ESP-IDF V4.4.5
wifi and ble were used
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13827409/sdkconfig.txt)

Espressif-liuuuu, 2024-01-10:
Hi @landaming , could you share log for the issue? 
Was there any log like `m f null` or `beacon timeout` ?

landaming, 2024-01-10:
Hello, this is the log when the problem occurred  @Espressif-liuuuu 
[SaveWindows2023_12_1_11-20-44_112113.txt](https://github.com/espressif/esp-idf/files/13886175/SaveWindows2023_12_1_11-20-44_112113.txt)
[SaveWindows2023_12_4_17-31-48_053240.txt](https://github.com/espressif/esp-idf/files/13886176/SaveWindows2023_12_4_17-31-48_053240.txt)

landaming, 2024-01-11:
Thanks for the tip, I will try this method once, and then feedback the result here @Espressif-liuuuu

landaming, 2024-01-17:
Hi, @Espressif-liuuuu
I have used 25cd984 points to build the program to test this issue over the next 1-3 weeks,Thanks.
[image]

KonssnoK, 2024-02-14:
this recalls quite closely issues that we are experiencing on the field with 25000+ devices.
we are on top of latest 4.4.6 branch of esp-idf.

What i can add:

    it happens both on LTE and Wifi (we have 2 netif interfaces)
    it is not the netif interface that stops working ( the drivers receive data from mqtt/higher layers when the timeouts occur )
    sometimes the downtime is for few seconds/minutes
    sometimes the downtime is persistent until a reboot of the device is triggered
    we do use MQTT

Espressif-liuuuu, 2024-02-19:
> this recalls quite closely issues that we are experiencing on the field with 25000+ devices. we are on top of latest 4.4.6 branch of esp-idf.
> 
> What i can add:
> 
> * we do not use ble
> * it happens both on LTE and Wifi (we have 2 netif interfaces)
> * it is not the netif interface that stops working ( the drivers receive data from mqtt/higher layers when the timeouts occur )
> * sometimes the downtime is for few seconds/minutes
> * sometimes the downtime is persistent until a reboot of the device is triggered
> * we do use MQTT

Hi @KonssnoK , this could be a different issue, please consider to open a new ticket.
Could you pls share the information below
1) exact commit id
2) if the issue exists when power save is disabled
3) if the issue exists when no LTE interface
4) call `esp_wifi_statis_dump(0xffff)` each some minutes and share the log when the issue happen.

KonssnoK, 2024-02-19:
> 4\. `esp_wifi_statis_dump`

hello @Espressif-liuuuu  ,
1. 27ec26d2d3f44bbde5da14c7fdfc82226d567874
2. we use wifi mesh, so PS_NONE is our standard
3. yes
4. ok we'll add this

AxelLin, 2024-02-19:
> Hi, @Espressif-liuuuu I have used [25cd984](https://github.com/espressif/esp-idf/commit/25cd984ad9cbd622c759670e72340e4ab3f41fb0) points to build the program to test this issue over the next 1-3 weeks,Thanks.

Hi @landaming How is your test result, is this fixed or still has issue?

KonssnoK, 2024-02-19:
> > 4. `esp_wifi_statis_dump`
> 
> hello @Espressif-liuuuu ,
> 
>     1. [27ec26d](https://github.com/espressif/esp-idf/commit/27ec26d2d3f44bbde5da14c7fdfc82226d567874)
> 
>     2. we use wifi mesh, so PS_NONE is our standard
> 
>     3. yes
> 
>     4. ok we'll add this

@Espressif-liuuuu 
here is a log where you receive data on the wifi interface but the ping goes into timeout and you cannot connect to MQTT, with dumps on the wifi.

[espressif_wifi_dump.txt](https://github.com/espressif/esp-idf/files/14327990/espressif_wifi_dump.txt)
```

 (384649) wifi: BUFFER:
I (384649) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (384649) wifi: 1  16  16    496         496        0        0
I (384653) wifi: 2  64  0     13          13         0        0
I (384659) wifi: 3  128 0     9           9          0        0       
I (384665) wifi: 4  64  2     3           1          0        0
I (384672) wifi: 5  0   0     4           4          0        0
I (384678) wifi: 6  0   0     0           0          0        0       
W (08:48:12.632) main: From 8.8.8.8 icmp_seq=297 timeout
I (384690) wifi: 7  32  0     387         387        0        0
I (08:48:12.641) esp_netif_lwip: SENDW        sta  106 D6927DC3021E AC1754000AB0 0800
I (384696) wifi: 8  5   5     45618       45618      0        9       E (08:48:12.655) wifi_netif: TXW

I (384715) wifi: 9  0   0     480         512        0        0
I (384721) wifi: 10 2   2     2           2          0        0
I (384727) wifi:
I (384729) wifi: LMAC TX:
I (384731) wifi: tx_all   lrc src age to
I (384735) wifi: 1293     0   0   0   0
I (384739) wifi: 0        0   0   0   0
I (384742) wifi: 463      0   0   0   0  
I (384746) wifi: 0        0   0   0   0
I (384750) wifi: 0        0   0   0   0
I (384754) wifi:
I (384755) wifi: LMAC RX:
I (384758) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (384763) wifi: 357      21       45656    0        0
I (384768) wifi:
I (384769) wifi: HW RX:
I (384772) wifi: rx_suc   rx_end   full#1   full#2
I (384776) wifi: b3d5     206d     1a       0       
I (384781) wifi:
I (384782) wifi: Hardware:
I (384785) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (384793) wifi: 0        b9db     b9da     0        df57     8f28     237c     0       
I (384801) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (384808) wifi: 0        7198     0        9        1d       b        0        0
I (384816) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic
I (384825) wifi: 0        0        0        0        1f       b        31       0        0       
I (384833) wifi:
I (384835) wifi: DIAG:
I (384837) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6
I (384844) wifi: e00001   0        fa00a07  0        80748001 10000844 0       
I (384851) wifi: diag7    diag8    diag9    diag10   diag12   diagsel
I (384857) wifi: 240      0        82e000   40       0        0
I (384864) wifi:
I (384865) wifi: Power save:
I (384868) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
I (384882) wifi: 3626     3628     0        0        0        0        0        0        0        2322     0        0        0        0        0        nan   %
I (384896) wifi:
I (384897) wifi: HMAC RX:
I (384900) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (384905) wifi: sta 42896    377      0      42518  42508
I (384910) wifi: ap  3133     0        0      3133   3133
I (384916) wifi:
I (384917) wifi: ifx auth     assoc    deauth action amsdu
I (384922) wifi: sta 1        1        1      7      0
I (384928) wifi: ap  0        0        0      0      0
I (384933) wifi:
I (384934) wifi: HMAC TX:
I (384937) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (384944) wifi: 481      0        0      0      0      0       0
```

the only thing that could be worrying is the number of aborted RX:
```
I (384785) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (384793) wifi: 0        b9db     b9da     0        df57     8f28     237c     0       
```

Behind quite hard to reproduce, i'll try again.

landaming, 2024-02-19:
> > Hi, @Espressif-liuuuu I have used [25cd984](https://github.com/espressif/esp-idf/commit/25cd984ad9cbd622c759670e72340e4ab3f41fb0) points to build the program to test this issue over the next 1-3 weeks,Thanks.
> 
> Hi @landaming How is your test result, is this fixed or still has issue?

I am currently conducting tests on two devices, and while one of them is experiencing issues, I am unable to export the logs due to its installation location.  As a result, I must await the test results of the other device. @AxelLin

zhangyanjiaoesp, 2024-02-19:
@KonssnoK your issue seems not related to the original issue, please create a new ticket to record it. 
By the way, does the issue happened on the root node or other node? Can you provide the capture packets for your issue?

AxelLin, 2024-04-23:
> I am currently conducting tests on two devices, and while one of them is experiencing issues, I am unable to export the logs due to its installation location. As a result, I must await the test results of the other device. @AxelLin

@landaming 
Could you update your test results?

landaming, 2024-04-23:
> > I am currently conducting tests on two devices, and while one of them is experiencing issues, I am unable to export the logs due to its installation location. As a result, I must await the test results of the other device. @AxelLin
> 
> @landaming Could you update your test results?

Hello, later this problem did not appear again, should be solved @AxelLin

Sherry616, 2024-10-09:
Thanks for reporting, we will close this issue as IDF v4.4 is end of life now. 
Thanks for using our Espressif product!

closed 2024-10-09 by Sherry616