espressif/esp-idf #8956: Escape characters are not cleaned from IDF_PATH

TTauriainenQt, 2022-05-16:
Escape characters are not cleaned from `IDF_PATH`  which causes issues with CMake / FreeRTOS component.
`\` needs to be converted to `\\` or `/`.

```
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\sdk\esp-idf-v4.4.1

...

C:\sdk\esp-idf-v4.4.1>cd C:\src\<PROJECT_NAME>
C:\src\<PROJECT_NAME>>idf.py build

...

CMake Error at C:/sdk/esp-idf-v4.4.1/tools/cmake/component.cmake:298 (foreach):
  Syntax error in cmake code at

    C:/sdk/esp-idf-v4.4.1/tools/cmake/component.cmake:298

  when parsing string

    C:\sdk\esp-idf-v4.4.1/components/freertos/include/freertos;include;../../include

  Invalid character escape '\s'.
Call Stack (most recent call first):
  C:/sdk/esp-idf-v4.4.1/tools/cmake/component.cmake:472 (__component_add_include_dirs)
  components/QulPlatform/CMakeLists.txt:1 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "C:/src/<PROJECT_NAME>/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

C:\src\<PROJECT_NAME>>
```

Variable is set in:
https://github.com/espressif/esp-idf/blob/ad747b237ade56c506675a4bfa2274a5918cffd5/export.bat#L17-L18
https://github.com/espressif/esp-idf/blob/ad747b237ade56c506675a4bfa2274a5918cffd5/export.bat#L27

dobairoland, 2022-05-16:
Hello. Thank you for your report. I assume using a different directory instead of `C:\sdk` is a suitable workaround for you. 

BTW, have you used the [Windows tool installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer) to install everything?

igrr, 2023-02-27:
@TTauriainenQt Do you see this error with one of the IDF examples or only with your custom project? If you are seeing it with a custom project, could you check if any of the component CMakeLists.txt files in your project have anything like
```cmake
idf_component_register(INCLUDE_DIRS ${IDF_PATH}/components/freertos/FreeRTOS-Kernel/include/freertos ...)
```

igrr, 2023-02-27:
In this particular case, I am guessing that you had to add `${IDF_PATH}/components/freertos/FreeRTOS-Kernel/include/freertos` as an include directory because IDF exposes FreeRTOS headers as "freertos/xxx.h" instead of simply "xxx.h" as it is common with standard FreeRTOS.

There is actually a convenience shortcut provided by the build system for this case, but seems like we haven't documented it anywhere:
```cmake
idf_component_get_property(freertos_include_dir freertos ORIG_INCLUDE_PATH)
```
Then you can use `${freertos_include_dir}` variable in the include directories list of your component.